MarlinFirmware/Marlin #17346: Skew_compensation Marlin 2.0 with bug fixes

byemi, 2020-03-31:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

I've installed Arduino 1.8.10 and trying to flash the firmware "Marlin 2.0 with bug fixes", from which I made the needed changes in Configuration.h file. I've noticed that after enabling Skew_compesation option, I'm getting a long message error in the log window ( please check attached files ).
I had a look over the internet about this error, even in the Marlin forum, but none of them fixed the problem. I've started by changing the u8glib ( which I suppose that it doesn't have anything to do ) and changing the decimal points of the "Input all length measurements". 

Looking forward for a way of removing the error.
Thank you in advance.

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-03-31:
If the error message referred to a stray \342 in the code it is most likely to have been caused by copying the code from a Web page which has Unicode characters in it.
In your Configuration.h 
In #define XY_DIAG_AC 282.70  there are invisible characters between the 7 and the 0
in #define XY_DIAG_BD 283.10  there are invisible characters between the 1 and the 0

byemi, 2020-03-31:
Excellent. Thank you very much for your hint.
Can't understand why it happened.  The only thing I did was to remove the comment from skew_compensation and correct the values. It can be my fault.
Wish you a beautiful day.

closed 2020-03-31 by byemi