MarlinFirmware/Marlin #17349: [BUG] Auto bed levelling issue

VertexityStudio, 2020-03-31:
### Bug Description

The bed leveling does not work, I have no correction during printing, the Z axis motors do not move at all to make this correction. I use AUTO_BED_LEVELING_BILINEAR, I have my grid created by G29 and saved by M500. I activate it in my GCode start with M420 S1. Below are my configuration files, my M503 status and my start GCode.
Thank you in advance.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4408458/Configuration.zip)

M503 status:

> SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo: M92 X160.00 Y160.00 Z800.00 E202.06
echo:Maximum feedrates (units/s):
echo:  M203 X200.00 Y200.00 Z50.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X500.00 Y500.00 Z250.00 E3000.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P2000.00 R2000.00 T3000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Auto Bed Leveling:
echo:  M420 S1 Z0.00
echo:  G29 W I0 J0 Z0.01125
echo:  G29 W I1 J0 Z-0.03958
echo:  G29 W I2 J0 Z-0.02917
echo:  G29 W I3 J0 Z0.00208
echo:  G29 W I0 J1 Z0.03833
echo:  G29 W I1 J1 Z-0.01125
echo:  G29 W I2 J1 Z0.03042
echo:  G29 W I3 J1 Z0.03875
echo:  G29 W I0 J2 Z0.11917
echo:  G29 W I1 J2 Z0.04250
echo:  G29 W I2 J2 Z0.06125
echo:  G29 W I3 J2 Z0.10167
echo:  G29 W I0 J3 Z0.10208
echo:  G29 W I1 J3 Z0.05083
echo:  G29 W I2 J3 Z0.10000
echo:  G29 W I3 J3 Z0.16333
echo:Material heatup parameters:
echo:  M145 S0 H210 B50 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P17.58 I1.03 D75.28
echo:  M304 P146.25 I28.21 D505.53
echo:Z-Probe Offset (mm):
echo:  M851 X-35.00 Y-5.50 Z-1.07
echo:Stepper driver current:
echo:  M906 X800 Y800 Z800
echo:  M906 T0 E800
echo:Driver stepping mode:
echo:  M569 S1 X Y Z
echo:  M569 S1 T0 E
Disconnected.

Start GCode:

> ; -- START GCODE --
G21                     ;set units to millimetres
G90                     ;set to absolute positioning
M106 S0                 ;set fan speed to zero (turned off)
M190 S50
M109 S210
G28 ; home all axes
M420 S1 ; Activate auto levelling
G1 X0 Y0 F3000 ; move to prime
G1 Z1 F500;
G92 E0;
G1 F400 E5;
;G1 Z5 F3000 ; lift
G1 X0 Y10 F1500 ; move to prime
G1 Z0.3 F3000 ; get ready to prime
G92 E0 ; reset extrusion distance
G1 Y80 E7 F600 ; prime nozzle
G1 Y100 F5000 ; quick wipe
; -- end of START GCODE --

thierryzoller, 2020-03-31:
I have the same issue, but thought it must be me? Your Z-offset is correct?

VertexityStudio, 2020-04-01:
Yes my Z-offset is correct. Many test prints and problems are always at same places. With older version of Marlin no problem, Z axis is moving to make corrections. With lastest, no movement :/

thierryzoller, 2020-04-01:
Do you have the same issue than I have additionally  ? After a print it sometimes forget it is on Safe_Homing. When asking to Home Z it crashes into the bed not even extending the BLTouch probe. Something appears really wrong at the moment

VertexityStudio, 2020-04-02:
No everything works fine for me except the bed leveling.
I just tried AUTO_BED_LEVELING_UBL for unified bed leveling and there it works. If I submit AUTO_BED_LEVELING_BILINEAR, no more correction!

marty2525, 2020-04-03:
Hi I think your problem is that your not saving the map to the eeprom and loading before printing M500 to save M501 to load M502 clear
Before your M401 S1 you need M502 
Also G28 to disable bedleveling should be there this may be why your z is playing up

thierryzoller, 2020-04-03:
Well I see the Z-Offsets listed in his 503 so it's in the Eprom already.
I had to add G29 to every print for it to work..

marty2525, 2020-04-03:
In my list there is m503 but if yours says offsets that is the offset of
the probe in relation to the nozzle

marty2525, 2020-04-03:
Sorry it should have read no m503 code

adawalli, 2020-04-08:
Just to clarify, because I believe I have the same issue...after tuning mesh bed leveling and saving to EEPROM - do you find that the "Bed Z" adjustment (at least from LCD) has no impact at all? I can go wildly positive or negative and I see no adjustment from the extruder. Baby steps work just fine though.

boelle, 2020-06-20:
@SoftiGest is issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle