MarlinFirmware/Marlin #17353: [BUG] Power Loss Recovery doesn't work on Geeetech a20

borsegbr, 2020-03-31:
### Bug Description

Power Loss Recovery doesn't work on Geeetech a20 
Board GT2560v3_a20

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4411134/Configuration.zip)

### Steps to Reproduce

1. Start Printing from SD
2. Cut off power 
3. Power ON

**Expected behavior:** Resume Printing Procedure

**Actual behavior:** No Resume printing procedure : normal status screen

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

boelle, 2020-06-20:
@borsegbr is the issue gone?

borsegbr, 2020-06-20:
Nope I'm waiting motherboard replacement to test if is hardware problem

raphaelbruno, 2020-06-25:
I have the same problem, I'm using:
- Marlin 2.0.5.3
- MKS Base 1.4

The feature is enabled...
```
>>> m413
SENDING:M413
echo:Power-loss recovery ON
```

**Same steps:**
1. Start Printing from SD
2. Cut off power
3. Power ON

The printer don't ask to me if I want to resume the print, I got a normal status screen...

sjasonsmith, 2020-06-30:
Duplicate of #17378

Your issue is a little older than the one I am leaving open, but that issue has attracted more users to add to it, which should gain it more attention.

Please continue any discussion over on that issue, while ignoring that "Ender 3" is in the title. The symptoms and likely the root cause are the same, and not specific to one printer.

closed 2020-06-30 by sjasonsmith

thinkyhead, 2020-07-03:
Duplicate of #17378