MarlinFirmware/Marlin #17357: [BUG] K-factor Calibration Pattern Generator not working

granthworth, 2020-04-01:
[image]
[image]
[image]

### Bug Description

No G-code is produced when clicking "Generate G-code" and no error explains why. Attempted the same set of settings in Chrome and Edge with the same result.

### My Configurations

See screenshots for configuration from website. 

### Steps to Reproduce

Configure settings as normal for a new filament. Even changing many settings affecting the overall size of the pattern do not fix the issue.

**Expected behavior:** gcode is produced after clicking "Generate G-code" OR an error is given explaining why no gcode can be generated.

**Actual behavior:** No response, no error, and gcode is not produced

https://marlinfw.org/tools/lin_advance/k-factor.html

thisiskeithb, 2020-04-01:
Something was likely broken during code refactoring in https://github.com/MarlinFirmware/MarlinDocumentation/commit/5762155216bb8669ba274bd7fdaab946ea1f94ba.

polato94, 2020-04-01:
I think i found the bug. Something happens on the "Use Bed Leveling" Option. 
[image]

if you change line 60 to this:
[image]

it'll work. I don't know that much about JS, but maybe I could help a little. :)

jpmath54, 2020-04-01:
Same issue here.
I'm currently printing visors for COVID19 protection of my local hospital, and switched to direct drive and larger nozzle.

Please do something to fix this, so I can produce even more of them!

Regards!

thinkyhead, 2020-04-01:
I've been playing around with the code, and last I checked it was working. I'll re-test and see where the oops fell in…

thinkyhead, 2020-04-01:
Pushing the patch now…

closed 2020-04-01 by thinkyhead