MarlinFirmware/Marlin #17360: [BUG] G29 7x7 Grid Fail

inventabuild, 2020-04-01:
Firmware: 1.1.9.34

Running G29 AUTO_BED_LEVELING_LINEAR with 7x3 grid works fine in the start code of my prints.  Upon completion of probing my printer successfully executes the next lines of g-code.
Running G29 AUTO_BED_LEVELING_LINEAR with 7x7 grid FAILS in the start code of my prints.  Upon completion of probing the LCD screen just says Probing Successful and the printer does not execute the next line of g-code..

Is there an upper limit setting somewhere so I can increase the probing grid size or is this a bug?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle

inventabuild, 2020-06-24:
Thanks for following up.  My best guess is that the issue was caused by modifications to the firmware by Lulzbot for their 3d printers.  When I updated to latest Marlin 2.0.5.3 I was able to probe much larger grids and the problem went away.