MarlinFirmware/Marlin #17362: [BUG] SKR 1.3 - Hotend fan not turning on at indicated temp

amaschas, 2020-04-01:
### Bug Description

I'm working on installing a new BTT SKR 1.3 board on my Ender 3, and I'm using the latest Marlin 2.0-bugfix. I can turn the fans on with the touchscreen and with gcodes, and they appear to be plugged in correct. M106 S0 turns on the hotend fan, and M106 S1 turns on the part cooling fan. However, the hotend fan does not automatically turn on when the temperature of the hotend gets above 50C, which is the temp indicated in the config.

### My Configurations

[configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4412879/configurations.zip)

### Steps to Reproduce

1. With the printer on, heat the hotend past 50C with either the touchscreen or gcodes.

**Expected behavior:** The hotend fan should turn on when the temperature of the hotend exceeds 50C.

**Actual behavior:** The hotend fan does not turn on.

I used the configuration files for the BTT SKR 1.3 from the zipfile that is linked in the config directory readme and modified them slightly, but didn't change the fan configuration.

randellhodges, 2020-04-01:
You'll need to define the pin for `#define E0_AUTO_FAN_PIN -1` inside Configuration_adv.h.

amaschas, 2020-04-01:
Hrm, I thought that line indicated that the default pin for that board should be used, as defined in pins_BTT_SKR_common.h (as discussed in #15199). Is that not that case?

randellhodges, 2020-04-01:
No. You are defining the pin to be -1, not telling it to use the default. Set it to either `FAN1_PIN` or `P2_04` assuming you are plugged into E1 heater.

When I used the 1.3 that is what I did.

amaschas, 2020-04-01:
OK, I set `#define E0_AUTO_FAN_PIN FAN1_PIN` and `#define CHAMBER_AUTO_FAN_PIN FAN_PIN` (since I believe the latter is the part fan config?) and it started working. I'm thinking the documentation above this section might need to be updated, since it states `Your board's pins file specifies the recommended pins. Override those here or set to -1 to disable completely.` which appears to be incorrect.

randellhodges, 2020-04-01:
`CHAMBER_AUTO_FAN_PIN` is for a heated chamber, not for the part cooling fan. By default, you don't need to do anything for part cooling fan. You can leave `CHAMBER_AUTO_FAN_PIN` set at -1.

`set to -1 to disable completely` that is exactly what happened when it was set to -1. It disabled the automatic turn on of the fan.

amaschas, 2020-04-01:
Ah, ok, I'll update that. Regarding the documentation, I just meant that it doesn't specify that you actually have to define the pins at this point. The two options seem to be override or disable, but I guess I'm not sure what behavior I'm overriding, since if you comment out those lines the fan still doesn't work. Perhaps that line could read `Refer to your board's pins file for the recommended pins and set the value here, or set -1 to disable completely.` There seemed to be a lot of confusion about this bit of config as I googled around for a solution, so perhaps updating the wording to clarify might be worth it? I'm happy to generate a PR, I'm just not sure I'm capturing the behavior accurately.

Anyway, thanks for your help, I will definitely jump on the discord!

kkosa1, 2020-04-02:
Thank you randellhodges and amaschas for the dialogue.
I know the comments of lines 416, 417 were not clear to my simple mind.  It seemed (the comments) to imply the pins file would specify the defaults and as such I also commented out line 422 thinking in order not to disable( via the -1  present in the line).  Again since I thought the defaults were used as declared in the pins file, else no need to "override them here."

Maybe something as simple as:

 * Your board's pins file specifies the recommended pins. Enter your definition here
 * or set to -1 to disable completely.

Thank you for the community.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle