MarlinFirmware/Marlin #17365: [BUG] the second extruder no working after G29

lyfbuaa, 2020-04-01:
mega 2560+ramps1.4+marlin2.0.5.3(or 2.0.5.2)
### Steps to Reproduce

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4414059/Configuration.zip)

1. print start code: G28 G29 .......
2. slice the model with "only“ setting the second extruder to print
3.extecute G28 and G29
4.wait for the heating
5.begin to print,but the extruder don‘t work

if slicing the model with setting the first extruder to print,but the printer is work normally.

boelle, 2020-06-20:
@lyfbuaa is issue gone?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle