MarlinFirmware/Marlin #17366: [ONLINE BUG] - K factor calibration tool “generate g-code” button not working

gaskins1992, 2020-04-01:
### Bug Description

 When trying to use the online “K Factor Calibration Tool”, all of the required fields are filled out and no errors are shown, but when the “Generate G-Code” button is pressed, nothing visible happens. When clicking “Download as file”, a message box appears and says “Generate G-code first”. This occurs with regardless of which version, pattern, or printing direction options [are](url) selected.

See attached pdf for values i entered into the generator.

[Linear Advance Calibration Pattern | Marlin Firmware.pdf](https://github.com/MarlinFirmware/Marlin/files/4414105/Linear.Advance.Calibration.Pattern.Marlin.Firmware.pdf)

@Sebastianv650, within the Marlin firmware is says to tag you if there are issues. Not sure if you need to be tagged for an online problem, but maybe you know who would need to be notified or can help. 

Looks like the same issue here.
https://github.com/MarlinFirmware/Marlin/issues/17357

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Go to https://marlinfw.org/tools/lin_advance/k-factor.html
2. Fill out required information to generate g-code
3. Click “generate g-code” button at the bottom of the form.

**Expected behavior:** G-code will fill the preview window to the right

**Actual behavior:** G-code preview window remains blank

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-04-01:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/17357.

Sebastianv650, 2020-04-01:
@gaskins1992 thanks for the heads up, but the generator is not written be me and I honestly don't know how to code/repair it.

gaskins1992, 2020-04-01:
@Sebastianv650 gotcha. Well it was worth a shot. Thanks for your work on the firmware side! Hopefully I can get it working soon. May have to manually do some testing to get it dialed in.

thisiskeithb, 2020-04-01:
@gaskins1992: Since this is a duplicate bug report, mind closing it and discussing in https://github.com/MarlinFirmware/Marlin/issues/17357?

gaskins1992, 2020-04-01:
@thisiskeithb no problem

closed 2020-04-01 by gaskins1992

gaskins1992, 2020-04-01:
A solution can be found here https://github.com/MarlinFirmware/MarlinDocumentation/issues/288

thisiskeithb, 2020-04-01:
> A solution can be found here [MarlinFirmware/MarlinDocumentation#288](https://github.com/MarlinFirmware/MarlinDocumentation/issues/288)

It’s already been patched: https://github.com/MarlinFirmware/MarlinDocumentation/commit/71d94f837026362f13307e74f14ec92e80ac0dc1