MarlinFirmware/Marlin #17369: [BUG] Err: EEPROM Version

Gamesnic, 2020-04-01:
Hey everyone,

I can't find any solution for this. My Ender 3, with the latest stable Marlin Firmware (not Nightly) has its fan not running (The extruder fan, which always spins) and in the Status line stands "Err: EEPROM Version". I already tried M502 and then M500. No luck.
I hope I didn't post this into the wrong forum and if I did, please tell me.
I also attached the configs.

[Configs(Github won't accept .h).zip](https://github.com/MarlinFirmware/Marlin/files/4416493/Configs.Github.won.t.accept.h.zip)

thisiskeithb, 2020-04-01:
You need to send `M502`, followed by `M500` to reset your EEPROM.

You can also do this through the LCD by going to Configuration -> Advanced Settings -> Initialize EEPROM.

Gamesnic, 2020-04-01:
This is exactly what I've done. Initializing the EEPROM didn't do any change, when I tried to click "Init" it didn't even do anything.

Gamesnic, 2020-04-01:
Nevermind, restarting the whole printer seemed to fix it. Thanks for helping.

closed 2020-04-01 by Gamesnic