MarlinFirmware/Marlin #17371: [BUG] Accel, A-retract and A-travel edit upper limits are not respected after changing the Amax X and Amax Y values.

ktand, 2020-04-01:
### Bug Description

When editing via LCD:

Accel, A-retract and A-travel edit upper limits are not respected after changning the Amax X and Amax Y values. Must save and reboot to make the new limits valid. Just saving/loading settings doesn't help.

`LIMITED_MAX_ACCEL_EDITING` is NOT defined.

I'm using Marlin bugfix-2.0.x commit e7a9f17 from March 22nd.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4416755/Marlin.zip)

### Steps to Reproduce

1. Set the Amax X and Amax Y values to 1250. Store settings and **reset** the controller.
2. Modify Accel, A-retract or A-travel. Notice that the upper limit is 1250 (**correct**)
3. Set the Amax X and Amax Y values to 2000. Store settings but do **NOT** reset the controller.
4. Modify Accel, A-retract or A-travel. Notice that the upper limit is still 1250 (**incorrect**).
5. Reset the controller
6. Modify Accel, A-retract or A-travel. Notice that the upper limit is now 2000 (**correct**)

**Expected behavior:** 

In step 4 the limit should be 2000.

**Actual behavior:** 

The limit is 1250.

MarkMan0, 2020-04-01:
I've published a pull request #17280, so this should be fixed in the latest bugfix-2.0.x

ktand, 2020-04-01:
Fixed in latest bugfix-2.0.x. Closing.

closed 2020-04-01 by ktand