MarlinFirmware/Marlin #17374: [BUG] Need to switch off and on after a print to launch a new

Exalik, 2020-04-01:
### Bug Description 

It is necessary to cold reboot (switch off and on) the printer to launch a print after another.

### My Configurations

ZIP file containing your `Configuration.h` and `Configuration_adv.h` files : http://dl.free.fr/bxQqVOOvG

### Steps to Reproduce

1. Print something / let's the printer finish it
2. Remove the print from the bed
3. Launch a new print

**Expected behavior:** The new print should be done as the previous

**Actual behavior:** The heating sequence and the bed leveling (with 3DTouch) are ok but after the nozzle go to the upper right corner and it extrudes a lot of filament. I have to switch off the printer to stop it.

#### Additional Information

Geeetech A10M (GT2560 v3.1) + 5 x TLC2208 v3.0 + 3DTouch

thinkyhead, 2020-04-01:
When you see a printer run off to the corner and extrude filament like that, it means that the G-code is telling it to go to some unreachable position like X1000 Y1000. The movement of XY is being constrained within the bed, but the extruder is not being stopped.

thinkyhead, 2020-04-01:
You can try `M502`, `M500` as a first step to make sure your coordinate space is not offset by some unusual amount.

Exalik, 2020-04-01:
Thank's for your reply thinkyhead.

I print from sd card so I do not send gcode through usb to my printer. I do like this to avoid any print issue du to a computer problem.

thinkyhead, 2020-05-01:
Me too! But my printer still accepts commands over USB. If you filled your USB port with hot glue then you can still use the LCD and click the "Reset EEPROM" option.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle