MarlinFirmware/Marlin #17375: [BUG] Error: Probing Failed after G29

albertmbullock, 2020-04-01:
### Bug Description

Using the latest release as of today (2.0.5.3) on an SKR 1.3 w/TMC_2208's in standalone, with an inductive probe. While trying to use G29 to level the bed, the machine will home, home Z in the center, then begin to probe the points but stops at point 2. I sent an M111 S32 command before running a G29, and got the following output:

SENDING:M111 S32
echo:DEBUG:LEVELING
>>> G29
SENDING:G29
  current_position= X95.00 Y151.00 Z0.00 : >>> G29
Machine Type: Cartesian
Probe: FIX_MOUNTED_PROBE
Probe Offset X50.00 Y-2.00 Z0.00 (Right-Front & Same Z as Nozzle)
Auto Bed Leveling: BILINEAR (disabled)
  current_position= X95.00 Y151.00 Z0.00 : Probe::set_deployed
deploy: 1
Probe::move_z(10.00)
>>> do_blocking_move_to X95.00 Y151.00 Z10.00
<<< do_blocking_move_to
>>> do_blocking_move_to X95.00 Y151.00 Z10.00
<<< do_blocking_move_to
reset_bed_level
>>> Probe::probe_at_point(50.00, 10.00, raise, 0, probe_relative)
  current_position= X95.00 Y151.00 Z10.00 :
>>> do_blocking_move_to X0.00 Y12.00 Z10.00
<<< do_blocking_move_to
echo:busy: processing
  current_position= X0.00 Y12.00 Z10.00 : Probe::set_deployed
deploy: 1
  current_position= X0.00 Y12.00 Z10.00 : >>> Probe::run_z_probe
  current_position= X0.00 Y12.00 Z10.00 : >>> Probe::probe_down_to_z
>>> do_blocking_move_to X0.00 Y12.00 Z-2.00
<<< do_blocking_move_to
echo:busy: processing
  current_position= X0.00 Y12.00 Z0.12 : sync_plan_position
  current_position= X0.00 Y12.00 Z0.12 : <<< Probe::probe_down_to_z
  current_position= X0.00 Y12.00 Z0.12 : <<< run_z_probe
>>> do_blocking_move_to X0.00 Y12.00 Z5.12
<<< do_blocking_move_to
<<< Probe::probe_at_point
>>> Probe::probe_at_point(165.00, 10.00, raise, 0, probe_relative)
  current_position= X0.00 Y12.00 Z5.12 :
>>> do_blocking_move_to X115.00 Y12.00 Z5.12
<<< do_blocking_move_to
echo:busy: processing
  current_position= X115.00 Y12.00 Z5.12 : Probe::set_deployed
deploy: 1
  current_position= X115.00 Y12.00 Z5.12 : >>> Probe::run_z_probe
  current_position= X115.00 Y12.00 Z5.12 : >>> Probe::probe_down_to_z
>>> do_blocking_move_to X115.00 Y12.00 Z-2.00
<<< do_blocking_move_to
  current_position= X115.00 Y12.00 Z2.06 : sync_plan_position
  current_position= X115.00 Y12.00 Z2.06 : <<< Probe::probe_down_to_z
SLOW Probe fail!
  current_position= X115.00 Y12.00 Z2.06 : <<< run_z_probe
  current_position= X115.00 Y12.00 Z2.06 : Probe::set_deployed
deploy: 0
>>> do_blocking_move_to X115.00 Y12.00 Z2.06
<<< do_blocking_move_to
Error:Probing Failed
[ERROR] Error:Probing Failed

<<< Probe::probe_at_point
  current_position= X115.00 Y12.00 Z2.06 : Probe::set_deployed
deploy: 0
  current_position= X115.00 Y12.00 Z2.06 : > probing complete
<<< G29

I have tried increasing the Z_PROBE_LOW_POINT to -10, as well as -15, and still can't get the printer to get passed the second point.

[Marlin 2.0.5.3.zip](https://github.com/MarlinFirmware/Marlin/files/4417491/Marlin.2.0.5.3.zip)

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. Power on and connect to printer via Pronterface
2. Home all axis
3. Once all axis are homed, send a G29 command

**Expected behavior:** The printer will begin it's automatic bed leveling routine

**Actual behavior:** The printer stops at the second probing location, with no visual indicators on the display

#### Additional Information

I'm using an optocoupler isolator board to use the my NPN NO LJ18A3-8-Z/BX inductive probe, and have dual Z stepper drivers enabled.

albertmbullock, 2020-04-01:
UPDATE: I have rolled back to the 2.0.x bugfix, and it seems to be working just fine. Will begin a print.

Talha909, 2020-04-02:
> UPDATE: I have rolled back to the 2.0.x bugfix, and it seems to be working just fine. Will begin a print.

did you install the sensor on 12 v or 5 v

albertmbullock, 2020-04-02:
> 
> did you install the sensor on 12 v or 5 v
>
It's a 12V sensor, wired to 12v, but I'm using an optocoupler which then uses the board's 5v, when triggered. My printer is still running fine, after having rolled back to the bugfix build.

marknotgeorge, 2020-04-02:
I am having a similar issue using release 2.0.5.2 on my Ender 3 fitted with a SKR E3 1.2 board. My printer supply is 24v, and I have an inductive sensor with an optocoupler fitted to reduce the voltage to 5v.

albertmbullock, 2020-04-02:
> 
> 
> I am having a similar issue using release 2.0.5.2 on my Ender 3 fitted with a SKR E3 1.2 board. My printer supply is 24v, and I have an inductive sensor with an optocoupler fitted to reduce the voltage to 5v.

Unless you need a feature specific to 2.05.2, I would recommend rolling back to the 2.0.x.x bugfix.

PhilMaddox, 2020-05-01:
Experienced exactly the same issue on an ender 3 / SKR1.3/ Inductive probe /Marlin 2.0.5.3 
I found by experimentation that changing #define Z_PROBE_LOW_POINT to -3 got it to work.
Im not 100% sure how or why just know it worked for me. I suspect there default -2 is too close to the typical offset for this kind of sensor. Would be interesting to see if it works for others too....

dturner, 2020-05-08:
I was having the same issue on my Ender 3 Pro with SKR mini e3 1.2 board and BLTouch v3.1. Running Marlin bugfix-2.0.x with SKR config ([changes here](https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...dturner:bugfix-2.0.x)). 

Here's what would happen: 

- Boot from cold
- Home X Y using limit switches
- Home Z using BLTouch, this works fine. 
- Execute `G29 T` via OctoPrint Visualizer, this works fine the first time. 
- Execute `G29 T` again. 

I get M112 machine halted and have to reboot. 

Output of `serial.log`: 

```
2020-05-08 17:30:28,510 - Send: G29 T
2020-05-08 17:30:38,209 - Recv: Error:Probing Failed
2020-05-08 17:30:38,220 - Changing monitoring state from "Operational" to "Error: Probing Failed"
2020-05-08 17:30:38,238 - Send: M112
2020-05-08 17:30:38,253 - Send: N2 M112*35
2020-05-08 17:30:38,263 - Send: N3 M104 T0 S0*34
2020-05-08 17:30:38,264 - Send: N4 M140 S0*97
2020-05-08 17:30:38,272 - Changing monitoring state from "Error: Probing Failed" to "Offline (Error: Probing Failed)"
```

Possible fix: I found that if I *always* execute `G28` before running `G29` or `G29 T` then I don't get the issue.

leberkaassemmel, 2020-05-15:
This problem also occured to me. I have both z-axis endstops wired up and was homing to the z-max endstop. I did not set the height of the z-axis as accurate as possible, since the probing should do that for me. I had problems with auto bed leveling on different positions of the bed with different settings (billinear probing with different points). After changing my z-homeing to be done by the probe, so z-min endstop, the auto bed leveling now works corretly.

PhilMaddox, 2020-05-17:
Went back to basics and first set up my bed mechanically to as level as possible . Problems solved 
Using 2.0.5.3 .
I think i was expecting too much from ABL !!  I've now run 8 prints and numerous G29s without problem

leberkaassemmel, 2020-05-17:
Seems like there is some fail-safe mechanism, which prevents damage to the nozzle, if the switch/probe is not triggered inside a certain interval. At least this would explain our problems and their solution. Sadly, I could not find any settings inside the configuration files.

PhilMaddox, 2020-05-17:
> Seems like there is some fail-safe mechanism, which prevents damage to the nozzle, if the switch/probe is not triggered inside a certain interval. At least this would explain our problems and their solution. Sadly, I could not find any settings inside the configuration files.

That  would make sense in the case that a probe fails while probing bed . It is just frustrating that the halt message " probing failed" doesn't really tell us much and even debug isn't much better to the average user.

sms043, 2020-06-04:
#define Z_PROBE_LOW_POINT          -20 
This fixed the z probe error after G29.
Probe = M18 PNP, Bed = spring steel, Printer = Ender 3, Board = SKR mini 3 1.2, Marlin = 2.0.5.3

hugocurran, 2020-06-10:
I have exactly the same problem on 2.0.5.3. Worked OK on 2.0.4.4 but I have rebuilt the bed since then - I assumed that was my problem, but the bed looks OK.

hugocurran, 2020-06-10:
Oh - and I have tried setting the Z_PROBE_LOW_POINT to -20 with no difference.

iamumr, 2020-06-12:
I have the same problem。

sylwesterr, 2020-06-14:
I have the same problem, having SKR Pro 1.1 with 3DTouch from MKS (BLTouch clone). The probe works well but when I had min Z set to -1 it failed randomly on 4-5 point, then I triggered bed leveling update from Bed Level Visualizer in Octoprint once it failed at last point (I have 25 since have 310mm square bed) and once on 12th, then I started reading this thread and started playing with decreasing Z min further down and since -4 (now I am on -5) the proving fails right on first probing point. Here’s my debug:

Send: G29
Recv:   current_position= X129.00 Y179.00 Z12.47 : >>> G29
Recv: Machine Type: Cartesian
Recv: Probe: BLTOUCH
Recv: Probe Offset X31.00 Y-19.00 Z-2.47 (Right-Front & Below Nozzle)
Recv: Auto Bed Leveling: BILINEAR (disabled)
Recv: 
Recv: >>> do_blocking_move_to X129.00 Y179.00 Z10.00
Recv: <<< do_blocking_move_to
Recv: >>> Probe::probe_at_point(21.00, 10.00, raise, 0, probe_relative)
Recv:   current_position= X129.00 Y179.00 Z10.00 :
Recv: >>> do_blocking_move_to X-10.00 Y29.00 Z10.00
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv:   current_position= X-10.00 Y29.00 Z10.00 : Probe::set_deployed
Recv: deploy: 1
Recv: Probe::move_z(10.00)
Recv: >>> do_blocking_move_to X-10.00 Y29.00 Z12.47
Recv: <<< do_blocking_move_to
Recv: BLTouch DEPLOY requested
Recv: BLTouch Command :10
Recv: bltouch.deploy_proc() end
Recv: >>> do_blocking_move_to X-10.00 Y29.00 Z12.47
Recv: <<< do_blocking_move_to
Recv:   current_position= X-10.00 Y29.00 Z12.47 : >>> Probe::run_z_probe
Recv:   current_position= X-10.00 Y29.00 Z12.47 : >>> Probe::probe_down_to_z
Recv: BLTouch DEPLOY requested
Recv: BLTouch Command :10
Recv: bltouch.deploy_proc() end
Recv: >>> do_blocking_move_to X-10.00 Y29.00 Z0.47
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:   current_position= X-10.00 Y29.00 Z0.47 : sync_plan_position
Recv:   current_position= X-10.00 Y29.00 Z0.47 : <<< Probe::probe_down_to_z
Recv: FAST Probe fail! - No trigger.
Recv:   current_position= X-10.00 Y29.00 Z0.47 : <<< run_z_probe
Recv:   current_position= X-10.00 Y29.00 Z0.47 : Probe::set_deployed
Recv: deploy: 0
Recv: Probe::move_z(10.00)
Recv: >>> do_blocking_move_to X-10.00 Y29.00 Z12.47
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: BLTouch STOW requested
Recv: BLTouch Command :90
Recv: bltouch.stow_proc() end
Recv: >>> do_blocking_move_to X-10.00 Y29.00 Z12.47
Recv: <<< do_blocking_move_to
Recv: Error:Probing Failed
Changing monitoring state from "Operational" to "Error: Probing Failed"
Send: M112
Send: N3 M112*34
Send: N4 M104 T0 S0*37
Send: N5 M140 S0*96
Changing monitoring state from "Error: Probing Failed" to "Offline (Error: Probing Failed)"
Connection closed, closing down monitor

sylwesterr, 2020-06-15:
Ok, I think I figured out the problem, in my case at least. I have Z offset, relative to probe's deplyed end, set to -2.47. I had set #define Z_PROBE_LOW_POINT set to default value of -2, so when the probe was going down it was not reaching the bed before passing that -2 limit. When I changed the value to -5 the problem dissapeared. After probing the values of my bed are around -4.

boelle, 2020-06-21:
@albertmbullock still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle

themacboy, 2020-07-26:
I have the same problem after install Makebase Robin Nano V.2. Marlin version  2.5.0.3, while doing the autobed level the board crash with a M112 message

javierpal, 2020-07-26:
I have the same error, I only activated Hybrid Treshhold

McSquid, 2020-07-26:
I have the same issue. I just upgraded to an SKR 1.4 from a previously working MKS 1.4 setup to make use of 2209 drivers.  came from Marlin 1.1x to 2.0.x using IR fixed probe. as soon as the probe is triggered I get probing failed. tried setting the Z_PROBE_LOW_POINT to -20 with no difference. I'm homing to a max endstop which works.

themacboy, 2020-07-26:
My problem was solved. MY bed was too down in one site and G29 can't test that point. Then leave with a error.

I dint noticed that in this site I set the bed too low and my arm kick with the servo, then was imposible to test that point and next ones.

I simple move the bed 2 milimeters upwards and now all is working fine

javierpal, 2020-07-26:
I just updated my marlin to bugfix2.x and started working

McSquid, 2020-07-26:
I just solved my issue. My Z bed size was set to 380mm, my z probe was triggering at 5mm, my offset is -0.13mm. I changed my Z bed size to 376.7 and probing started working. I guess It will fail if it is too far from supposed 0. This was not a problem in 1.1x.

vanhau018, 2020-09-10:
be sure your mechanic, because your bed is deviated too much,
Z_PROBE_LOW_POINT -2 (your transducer only allows a -2 point difference, that difference results in a better print product)

vanhau018, 2020-09-10:
be sure your bed, because your bed was skewed too much,
Z_PROBE_LOW_POINT -2 (your probe only allows a -2 difference between points in the bed)

ICEbAIR82, 2020-11-08:
Same problem here.
Z_PROBE_LOW_POINT is set to -20

procedure: G28, then G29
result: Home position, Nozzle lifts up to probe position, lowers a bit (very shortly!) and error. I doubt that it went -20 before error occured.