MarlinFirmware/Marlin #17376: MKS Robin nano 1.2 Marlin  2.0.5.3 - without SD card infinite booting loop, with card freeze at beginning.

porlock, 2020-04-01:
### Bug Description
In Marlin 2.0.5.3, MKS, Robin nano 1.2 without SD card inserted goes to infinite loop after Marlin splash screen, with SD card freeze at very beginning and not launch at all. On Marlin 2.0.5 everything was fine. Mks Robin nano has build in flash but it was not supported and all configuration was stored on SD card i saw some commits that change that in 2.0.5.3 and utilize build in flash instead SD card, but it seams that there is a bug in this implementation.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4417907/Marlin.zip)

### Steps to Reproduce

1. Compile Marlin 2.0.5.3  for MKS RobinNano 
2. Try boot without SC card inserted
3. Try boot with SC card inserted

**Expected behavior:** 
Marlin boots successfully, configuration is stored in build in flash.

**Actual behavior:** [What actually happens]
Without SD card inserted goes to infinite loop after Marlin splash screen, with SD card inserted it freeze at very beginning.

ConceitedCode, 2020-04-03:
I also have the same issue occuring after updating my printer to the latest 2.0.x build.

Miraculix200, 2020-04-04:
Did you try compiling it with a "vanilla" configuration file, instead of using your previous configuration file? I was at least able to get it running without freezing or getting stuck at the boot screen after the splash screen. Just made 3-4 necessary changes so it compiles properly for MKS Robin Nano.

Wasn't able to get it running with my previous configuration files however, and was too lazy to figure out what exactly causes this behaviour. So I went back to 2.0.5

porlock, 2020-04-04:
> My problem may be caused by adding some code to upscale the display to 3.5" resolution (480xsomething) however. Maybe it's not the same problem as the one you have.

So your vanilla config version works with or without upscale fix ?

This is my version of upscale fix with 2.0.5.3
https://github.com/porlock/SapphireProMarlin2.0stable/tree/SapphirePro-dev

If you can please test your working configuration, just change this line 

//#define FSMC_GRAPHICAL_TFT
#define SAPPHIRE_GRAPHICAL_TFT

That we let us know is upscale fix make this problem or this is something else. 
I bet it is about sdcard and flash memory handling

porlock, 2020-04-04:
Bye default thi is comment out

//#define EEPROM_SETTINGS  
//#define SDSUPPORT

That is why your  vanilla config compilation works, but you can't store any changes to your configuration, that is not a solution.

porlock, 2020-04-04:
I just test it and can 100% confirm that this two options commented, 

//#define EEPROM_SETTINGS
//#define SDSUPPORT

robin nano boots correctly, but like i mentioned it is not a solution.

porlock, 2020-04-04:
adding 
#define SDCARD_CONNECTION              ONBOARD
 #define SDIO_SUPPORT
 #define SD_DETECT_PIN                     PD12

to pin config for robin nano does not solve the issue :(

porlock, 2020-04-04:
Ok, like i suspected, Marlin implementation of storing data on flash memory doesn't support robin nano (this board has flash build in but implementation if faulty), so it has to be used in old way that is store EPROM data on SD card. To make this possible pins config should have this options enabled, and then it will works like previous.  (i have to setup everything from scratch and store settings again)

//
// SD Card
//
#define SDCARD_CONNECTION ONBOARD
#define SDIO_SUPPORT
#define SD_DETECT_PIN      PD12

//
// EEPROM
//
//#define FLASH_EEPROM_EMULATION
#define SDCARD_EEPROM_EMULATION

Miraculix200, 2020-04-05:
I've tried your version of the Marlin source code and can confirm it works on my Sapphire Pro after making these changes to pins_MKS_ROBIN_NANO.h and uncommenting #define EEPROM_SETTINGS in my Configuration.h

closed 2020-05-01 by thinkyhead