MarlinFirmware/Marlin #17377: [BUG] (USE_CONTROLLER_FAN)

dzieva, 2020-04-01:
### Bug Description
The fan is always on, regardless of the configuration parameters

### My Configurations
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4417944/Configurations.zip)

### Steps to Reproduce
After updating the firmware from version 2.01 to version 2.0.5.3 the fan is always on and at the same speed not obeying the settings:
#define CONTROLLERFAN_SPEED_IDLE 90
#define CONTROLLERFAN_IDLE_TIME 50

**Expected behavior:** [What you expect to happen]
I expected the above settings to work

**Actual behavior:** [What actually happens]
Nothing happens when the parameters are modified

dzieva, 2020-04-02:
Something is wrong here!

[image]

dzieva, 2020-04-03:
Before
[image]

After
[image]

It may not have been the most elegant way to do it, but it worked for me.

Please fix your code.

boelle, 2020-06-20:
you can fix it yourself and submit a PR

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle