MarlinFirmware/Marlin #17378: [BUG] Ender 3 - POWERLOSS recovery does not work as expected

GSlzr, 2020-04-01:
### Bug Description
Powerloss recovery does not work as expected, even with a PLR file present in the SDCard the system does not boot to the recovery screen

### My Configurations
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4418001/Configuration.zip)
Also set the SAVE_INFO_INTERVAL_MS to 1000 on src/feature/powerloss.h

### Steps to Reproduce
Using an Ender-3 with the firmware bugfix-2.0.x downloaded on 01/04/2020
1. Start a print and wait for some layers to be printed
2. Turn off the power using the power switch on the PSU
3. Turn the printer back on

**Expected behavior:** An option to continue the print should show up on the screen

**Actual behavior:** The system boots to the initial screen without presenting a recovery option

systemerror71, 2020-04-06:
Regards,
I also have the same problem, with Marlin 2.0.5.3 the power loss recovery enabled but in practice it does not work .... the printer stops and does not recover the work.
This problem occurs from version 2.0.x while in version 1.1.9 it worked perfectly.

systemerror71, 2020-04-06:
Steps to Reproduce

Using an Ender-3 with the firmware bugfix-2.0.x downloaded on 01/04/2020

    Start a print and wait for some layers to be printed
    Turn off the power using the power switch on the PSU
    Turn the printer back on

Expected behavior: An option to continue the print should show up on the screen

Actual behavior: The system boots to the initial screen without presenting a recovery option

TheMiguelBi, 2020-05-01:
I also have the same problem.
It does not resume automatically, but it **is possible** to do it **manually**:
Starting a new print from the same SD, a pop-up window appears asking me if I want to resume printing or delete it. So my printing can resume (actually I have other problems with offset Z axis, but recovery works).

Does this also happen to you?

systemerror71, 2020-05-01:
Hi .... no it doesn't happen to me ... the printer stays still .... and if I try to start a new print ... no popup window appears.  Regards

TheMiguelBi, 2020-05-01:
> Hi .... no it doesn't happen to me ... the printer stays still .... and if I try to start a new print ... no popup window appears. Regards

Out of curiosity, which display do you use ??
I have done many other tests today. 
Here the detail: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/644#issue-610793577

systemerror71, 2020-05-01:
Hi I have the lcd stock monitor ... but I ordered exactly what you indicated. Greetings and thanks

Talha909, 2020-06-03:
did you find the solution.

systemerror71, 2020-06-03:
Hi, I changed the motherboard and I bought the skr mini e3 with
Bigtreetech touch lcd monitor.
So I solved it from the root :-)))))
Hi

-- 

Francesco Pinna
Microsoft Beta ID 1625034
Microsoft Windows Specialist
-----------------------
Profile Linkedin:
http://it.linkedin.com/pub/francesco-pinna/11/aa3/8b3
my blog:
http://betatester71.blogspot.com/

*****************************************************
Il contenuto della presente comunicazione e'strettamente riservato,
essendo indirizzato esclusivamente al "destinatario" sopra individuato e
potendo contenere informazioni strettamente personali e/o confidenziali.
Qualora fosse pervenuto a soggetto diverso dal "destinatario" questi deve
intendersi sin d'ora avvisato che qualsiasi forma di diffusione dei
dati,dei fatti e delle notizie apprese e' assolutamente vietata.
Vi chiediamo in tal caso, di cancellarla dal Vostro sistema dopo aver
notificato al mittente l'errore commesso. (Rif. D.Lgs. 196/2003).Grazie

*****************************************************

Talha909, 2020-06-05:
> Hi, I changed the motherboard and I bought the skr mini e3 with Bigtreetech touch lcd monitor. So I solved it from the root :-))))) Hi Il giorno mer 3 giu 2020 alle ore 15:19 Talha909 <notifications@github.com> ha scritto:
> […](#)
> -- Francesco Pinna Microsoft Beta ID 1625034 Microsoft Windows Specialist ----------------------- Profile Linkedin: http://it.linkedin.com/pub/francesco-pinna/11/aa3/8b3 my blog: http://betatester71.blogspot.com/ ***************************************************** Il contenuto della presente comunicazione e'strettamente riservato, essendo indirizzato esclusivamente al "destinatario" sopra individuato e potendo contenere informazioni strettamente personali e/o confidenziali. Qualora fosse pervenuto a soggetto diverso dal "destinatario" questi deve intendersi sin d'ora avvisato che qualsiasi forma di diffusione dei dati,dei fatti e delle notizie apprese e' assolutamente vietata. Vi chiediamo in tal caso, di cancellarla dal Vostro sistema dopo aver notificato al mittente l'errore commesso. (Rif. D.Lgs. 196/2003).Grazie *****************************************************

Thanks for the answer. Can you share the experience of power loss. Is is working properly. Can you share the config files. And please tell me in which files you have changed the setting.

sdaitzman, 2020-06-14:
This issue is occurring for me as well. https://github.com/sdaitzman/Marlin/

boelle, 2020-06-21:
@GSlzr still an issue?

sdaitzman, 2020-06-21:
Just merged latest bugfix-2.0.0 (fun 5am merge) and will go downstairs and flash and see ☺️

https://github.com/sdaitzman/Marlin

sdaitzman, 2020-06-21:
Just retested, definitely still not working. I tried a default SD card test print (the 2.5 hour dog) and waited for it to finish a couple layers, pulled the plug, plugged it back in. Nothing appeared on reboot--no resume dialog as is normally expected.

sdaitzman, 2020-06-30:
Does the SD card have to be formatted in a particular way? Is there a way I can debug/check if the files are being created at all?

sjasonsmith, 2020-06-30:
@borsegbr reported the same issue on a Geetech a20, you can find their configs in issue #17353.

I'm not personally familiar with power-loss recovery to know what exactly is expected, but based on the number of reports I will mark this as a confirmed bug.

sjasonsmith, 2020-06-30:
In the most recent duplicate issue I linked, @astro-smart pointed out some concerns with the behavior of Power Loss Recovery if you don't have a mini-UPS on the system. I am hoping someone experienced with Power Loss Recovery will be able to review his comment, and help us understand what are appropriate expectations for this feature.

From @astro-smart:
_Ive determined the only way to get this to work without the sd card being written to every 2 seconds is to use the power pin method instead. You need to get a mini ups for this to work as without it, the board powers off far to quickly for the states to be written the the sd card only with loss of power and power pin.. I ordered a mini ups for about 3 bucks each. When I get it, I will try the option and let you know. Currently, the software SD method does work but its too many writes to the sd card to use for a viable solution. After about a month, Im sure the 100k writes will kill the card. If fact I noticed even when not printing the SD card is written too. That part is a bug I would expect._

spooky79, 2020-06-30:
in Marlin 2.0.3 the function works, so this is a software problem

GSlzr, 2020-06-30:
> @GSlzr still an issue?

Sorry, for the late reply. Currently I cannot test it, I don't have the printer with me.

astro-smart, 2020-07-03:
Ok...With 2.053 I can confirm the hardware pin method with mini ups from BTT works great. Without this method, writing to the SD card is not viable as it write even though no printing is active. You will run out the 100K read/writes on SD card rather quickly. So, buy the 10 dollar part and use the power loss pin option... Works great....

sjasonsmith, 2020-07-05:
I have reproduced this and found a workaround to get it working on my machine. I am currently working on understanding that code a little better to determine whether or not it is the correct fix.

sjasonsmith, 2020-07-06:
I have posted a small change in a PR, which fixes this on my printer. Please try it out and report back whether it works in your printer.
https://github.com/MarlinFirmware/Marlin/pull/18557

Tuebel, 2020-07-06:
I installed the lates bugfix on my Ender 3 with an SKR mini E3 without using the powerloss PIN. Now the screen is displayed as expected and the test print has been continued successfully.

@sjasonsmith Thank you!

closed 2020-07-07 by thinkyhead

sdaitzman, 2020-07-08:
Confirming that the message reappears successfully for me. Unfortunately, once I select the "Resume Print" option, it now does nothing... it moves to the park location, displays "Printing" on the screen, and then sits there doing nothing. Will continue trying to debug and raise another issue if this continue.

thinkyhead, 2020-07-08:
> once I select the "Resume Print" option, it now does nothing... it moves to the park location, displays "Printing" on the screen, and then sits there doing nothing

If you can get a terminal attached before this point, it would be interesting to see if it's putting out any console messages.

sdaitzman, 2020-07-09:
Weirdly, I'm no longer able to reproduce the bug. Will update if I can!