MarlinFirmware/Marlin #17382: Compile error with delta calibrate activated

Kowy54, 2020-04-02:
When i compiling code for skr 1.4 for my custom delta, i get this error when i have uncommented delta calibrate
Marlin\src\lcd\menu\menu_delta_calibrate.cpp: In function 'void _man_probe_pt(const xy_pos_t&)':
Marlin\src\lcd\menu\menu_delta_calibrate.cpp:49:79: error: 'DEFAULT_XYZ_STEPS_PER_UNIT' was not declared in this scope; did you mean 'DEFAULT_AXIS_STEPS_PER_UNIT'?
   49 |     move_menu_scale = _MAX(PROBE_MANUALLY_STEP, MIN_STEPS_PER_SEGMENT / float(DEFAULT_XYZ_STEPS_PER_UNIT));
      |                                                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                               DEFAULT_AXIS_STEPS_PER_UNIT
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_main.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_media.cpp.o
*** [.pio\build\LPC1768\src\src\lcd\menu\menu_delta_calibrate.cpp.o] Error 1

ellensp, 2020-04-02:
Please provide Configuration files, as was requested.

Kowy54, 2020-04-02:
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/4419658/Config.zip)

ellensp, 2020-04-02:
You have to start with a delta example configuration., It add extra defines such as 
// delta speeds must be the same on xyz
`#define DEFAULT_XYZ_STEPS_PER_UNIT ((XYZ_FULL_STEPS_PER_ROTATION) * (XYZ_MICROSTEPS) / double(XYZ_BELT_PITCH) / double(XYZ_PULLEY_TEETH)) // 80
`

Kowy54, 2020-04-02:
Now its working. Thanks ;)

closed 2020-04-02 by Kowy54