MarlinFirmware/Marlin #17384: [BUG] SKR Pro 1.1 BLTouch is deploying during print

randellhodges, 2020-04-02:
### Bug Description

BLTouch probe is dropped during prints, causing the pin to be bent and the probe to go into alarm and start blinking.

The linked issue sums it up. I had this problem back then, but went back to my SKR 1.4 board. I thought I'd give the SKR PRO 1.1 a shot again and I'm still getting the issue.

Original issue with the bltouch randomly deploying during a print: 
https://github.com/MarlinFirmware/Marlin/issues/16318

I might have also seen this issue, but I'm not 100% sure:
https://github.com/MarlinFirmware/Marlin/issues/17337

### My Configurations

https://github.com/randellhodges/Marlin/tree/dc21d2e2b764e99fda49cab45e343630efbce3bd/Marlin

### Steps to Reproduce

1. Use configuration
2. Start Print
3. Wait between 1 and 1.5 hours. (I did a few shorter prints, < hour and didn't get the behavior).

**Expected behavior:** Pin doesn't drop/BLTouch doesn't alarm

**Actual behavior:** Pin drops, maybe gets bent and then pulled back up, BLTouch starts blinking because it is now in an alarm state.

randellhodges, 2020-04-02:
Small update. I am about 4.5 hours into the print and it hasn't failed yet after some modifications. Per discussion on discord, I commented out the lines for PA_1 and PA_2 inside the PinMap_PWM block of the PeripheralPins.c file and also added a 1000uF 35v cap to the 5v line on the board.

I don't think I needed to comment out the PA_2 line since I'm not using it for the trigger port.

So, the question is, did commenting out PA_1 fix it, or did adding the cap to the 5v line, or have I just gotten lucky and haven't hit the condition that caused my first 2 long(er) prints to fail.

I had 3 prints succeed (all less than an hour) and 2 prints fail about 1.5 into a 5 hour print with the SKR PRO 1.1 before I made any changes.

randellhodges, 2020-04-03:
I noticed in the history of the PeripheralPins.c file that PA_1 and PA_2 had some changes made that also included TIM2 being assigned to them. So both alternate functions was defined.

I commented those line out and left only PA_1=TIM5_CH2, PA_2=TIM5_CH3 and I still had problems so that also rule out the capacitor helping.

I have commented out *all* PA_1/PA_2 lines inside the PinMap_PWM block and so far no blinking bltouch.

On a side note I would think having 2 PWM alternate functions assigned to a single pin is an error, regardless.

randellhodges, 2020-04-08:
I think I saw a drop even with those pins commented out. I think I might actually have a hardware issue, however, I still think that the code changes with GTR that having 2 PWM functions assigned to those pins is probably an error.

I'm closing this for now.

closed 2020-04-08 by randellhodges

thinkyhead, 2020-04-08:
> I still think that the code changes with GTR that having 2 PWM functions assigned to those pins is probably an error

The most useful person to @ and ask about it would be the GTR maintainer who made the change. I would hate to fiddle with it on my limited knowledge and make it worse.