MarlinFirmware/Marlin #17385: UBL Probing problem in Marlin-2.0.5.2 & Marlin-2.0.5.3

hamed-ta, 2020-04-02:
I have ubl bed leveling problem on ( Marlin-2.0.5.2 and Marlin-bugfix-2.0.5.2 & 2.0.5.3)  but it is ok in Marlin-bugfix-2.0.3

```
#define Z_CLEARANCE_DEPLOY_PROBE   1 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  1 // Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE     1 // Z Clearance between multiple probes
#define Z_AFTER_PROBING           5 // Z position after probing is done

#define Z_PROBE_LOW_POINT          -5 // Farthest distance below the trigger-point to go before stopping

// For M851 give a range for adjusting the Z probe offset
#define Z_PROBE_OFFSET_RANGE_MIN -30
#define Z_PROBE_OFFSET_RANGE_MAX 30

// Enable the M48 repeatability test to test probe accuracy
#define Z_MIN_PROBE_REPEATABILITY_TEST

// Before deploy/stow pause for user confirmation
#define PAUSE_BEFORE_DEPLOY_STOW
#if ENABLED(PAUSE_BEFORE_DEPLOY_STOW)
  #define PAUSE_PROBE_DEPLOY_WHEN_TRIGGERED // For Manual Deploy Allenkey Probe
#endif

```
```
Error:Probing Failed
Probing mesh point 31/36.
Error:Probing Failed
Probing mesh point 32/36.
Error:Probing Failed
Probing mesh point 33/36.
Error:Probing Failed
Probing mesh point 34/36.
Error:Probing Failed
Probing mesh point 35/36.
Error:Probing Failed
Probing mesh point 36/36.
Error:Probing Failed
X:8.00 Y:7.00 Z:5.00 E:0.00 Count X:640 Y:560 Z:2000
ok
M104 S70 T0
M140 S70
ok T:48.17 /70.00 B:46.58 /0.00 @:127 B@:0
ok T:48.14 /70.00 B:46.51 /70.00 @:127 B@:0
G28
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X:0 Y:0 Z:-1200
ok
X:0.00 Y:0.00 Z:5.00 E:0.00 Count X:0 Y:0 Z:2000
ok
M502
echo:Hardcoded Default Settings Loaded
M500
echo:Settings Stored (635 bytes; crc 5797)
M501
echo:V75 stored settings retrieved (635 bytes; crc 5797)
Unified Bed Leveling System v1.01 inactive

Unified Bed Leveling initialized.

UBL reset
G29 P1 V4
Default storage slot 0 selected.
Mesh invalidated. Probing mesh.
Probing around (0.00,0.00).

Probing mesh point 1/36.

Bed X: 8.000 Y: 7.000 Z: 2.020
Probing mesh point 2/36.
Bed X: 8.000 Y: 47.600 Z: 2.063
[... 1128 lines cut ...]
Probe::move_z(1.00)
>>> do_blocking_move_to X49.40 Y7.00 Z1.90
<<< do_blocking_move_to
  current_position= X49.40 Y7.00 Z1.90 : >>> Probe::run_z_probe
  current_position= X49.40 Y7.00 Z1.90 : >>> Probe::probe_down_to_z
>>> do_blocking_move_to X49.40 Y7.00 Z-10.20
<<< do_blocking_move_to
  current_position= X49.40 Y7.00 Z1.90 : sync_plan_position
  current_position= X49.40 Y7.00 Z1.90 : <<< Probe::probe_down_to_z
SLOW Probe fail!
  current_position= X49.40 Y7.00 Z1.90 : <<< run_z_probe
  current_position= X49.40 Y7.00 Z1.90 : Probe::set_deployed
deploy: 0
>>> do_blocking_move_to X49.40 Y7.00 Z1.90
<<< do_blocking_move_to
Error:Probing Failed
<<< Probe::probe_at_point
  current_position= X49.40 Y7.00 Z1.90 : Probe::set_deployed
deploy: 0
>>> do_blocking_move_to X49.40 Y7.00 Z5.00
<<< do_blocking_move_to
>>> do_blocking_move_to X8.00 Y7.00 Z5.00
<<< do_blocking_move_to
X:8.00 Y:7.00 Z:5.00 E:0.00 Count X:640 Y:560 Z:2000
```

thinkyhead, 2020-04-02:
Thanks for the log. It shows that probing does indeed fail.

- In order to determine whether it's only a problem with the board we need to know what kind of board you have.

- In order to determine whether it's a conflict with some other feature we need to know every feature you have enabled.

Please ZIP up your `Configuration.h` and `Configuration_adv.h` files and include them in your next reply by dropping the ZIP file onto the text box.

hamed-ta, 2020-04-03:
Here you are

[image]

[image]

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4425710/configs.zip)

Grogyan, 2020-04-04:
@thinkyhead occurs on the SKR1.3 as well.

Seems to be Safe_homing not actually probing the actual centre of the bed.  Such that when the last line of probes for UBL are executed, they cannot be reached

Grogyan, 2020-04-04:
Probe offsets not being applied?

hamed-ta, 2020-04-05:
> Probe offsets not being applied

yes, I have applied the probe offset in the settings.

> #define NOZZLE_TO_PROBE_OFFSET { 0, 0, 0.2 }

hamed-ta, 2020-04-05:
> Safe_homing

well, I don't think so as with the same setting, Marlin 2.0.3 works well, but in new version, the probing even doesn't start.

thinkyhead, 2020-04-27:
Recently patches were made to deal with bad constraints on bed leveling. So 
please test the latest [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands.

hamed-ta, 2020-04-28:
I'm going to test it today.

hamed-ta, 2020-04-28:
I tested it, still it has the problem and after starting the probing it goes to the first point and by hitting the prob in first point, the errors appear in the log, you can see in the video below

Configurations
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4544292/Configs.zip)

[video file 
](https://alphabit.ir/dev/levpro.mp4)

leewilko, 2020-05-29:
> I tested it, still it has the problem and after starting the probing it goes to the first point and by hitting the prob in first point, the errors appear in the log, you can see in the video below
> 
> [Configs.zip](https://github.com/MarlinFirmware/Marlin/files/4544292/Configs.zip)
> 
> [Video file ](https://alphabit.ir/dev/levpro.mp4)
> 
> Log Output

Exactly the same problem for me with SKR 1.4 board using UBL with a microswitch probe. Probe triggers in endstop probe section fine with a M119 and UBL runs but fails with the following everytime I probe no matter how I probe:

>>>G28
SENDING:G28
>>>G29 P1
SENDING:G29 P1
Default storage slot 0 selected.
Mesh invalidated. Probing mesh.
Probing mesh point 1/100.
Error:Probing Failed
[ERROR] Error:Probing Failed

Probing mesh point 2/100.
Error:Probing Failed
[ERROR] Error:Probing Failed

Probing mesh point 3/100.
Error:Probing Failed
[ERROR] Error:Probing Failed

I am using the latest Marlin bugfix branch downloaded around a 1 week ago as well so it seems the updates have not resolved, and chance of an update or a workaround?

Happy to help test any fix or troubleshoot?

boelle, 2020-06-21:
@hamed-ta still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle

hamed-ta, 2020-06-24:
> @hamed-ta still an issue?

I have tried to solved the issue by changing the settings but it still exists and I'm waiting for the next path to be able to use my printer.

boelle, 2020-06-24:
you can just use [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x)

hamed-ta, 2020-06-27:
The last commit is for two months ago I have tested it and it’s not working.