MarlinFirmware/Marlin #17386: BLTouch no longer working on 2.0

pietrobaricco, 2020-04-02:
Have been running 2.0.x-bugfix for quite a while without issues on a MKS SBASE, today I tried the latest 2.0.5.4 and BLTouch does not deploy nor self test, no errors on terminal either.
While homing, the head just crashes onto the bed.
Same thing with 2.0.4.4, then reverted back to the latest bugfix commit and it started working again.

thinkyhead, 2020-04-27:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

pietrobaricco, 2020-05-01:
Flashed today the latest bugfix branch, and the issue seems to be gone!

closed 2020-06-20 by boelle