MarlinFirmware/Marlin #17388: [BUG] BLTouch problem. Random deploy and stow during printing! Ender 5

kaeltaz, 2020-04-02:
Hello
I have a problem with my Ender 5. Motherboard SKR E3 mini 1.2, Marlin 2.0.4.4
The problem is that during printing (less often during heating before printing) the BLtouch leveling sensor deploy and stow. It's like a momentary trip. It is so dangerous and alarming that it happens during printing which may cause a collision with the printed element.

Has anyone encountered such a phenomenon and be able to indicate what may be wrong or broken. At first I thought that it was the fault of the motors and the wiring harness where all the wires connect and go to the top of the printer, but after replacing the cable connecting the sensor with the motherboard to a decent shielded symmetrical cable (BITNER) the problem did not stop - it made no difference

I am asking for help because I am currently grounded with printing. I am sending a link to YouTube with a recording of the problem:

[youtube](https://www.youtube.com/watch?v=YI3vZZAyzyw&feature=youtu.be)

thinkyhead, 2020-04-02:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-04-02 by thinkyhead

kaeltaz, 2020-04-05:
https://youtu.be/Ka5lb3ms6AA

https://youtu.be/LIVWAOxidYk

the problem also occurs when editing the color value Ws2812 neopixels