MarlinFirmware/Marlin #17391: [BUG] BLTouch does not deploy on Home Z (including Auto Home) with Marlin 2.0.5.3

Exalik, 2020-04-03:
### Bug Description

On Auto Home, the Home Y and Home X work fine but the BLTouch does not deploy. It is also on Home Z command.

### My Configurations
[Marlin 2.0.5.3 config.zip](https://github.com/MarlinFirmware/Marlin/files/4427148/Marlin.2.0.5.3.config.zip)

### Steps to Reproduce

1. Power up the printer
2. Launch an auto home
3. Look the printer doing the Home Y then Home Y
4. Look the nozzle to go under the center of the bed.

**Expected behavior:**  The BLTouch should deploy and the Z axis should go down.

**Actual behavior:** The BLTouch does not deploy and the Z axis does not move.

#### Additional Information

This issue is not on Marlin 2.0.5.2
Printer Geeetech A10M (GT2560 v3.1) with TMC2208 v3.0 and BLTouch

Grogyan, 2020-04-04:
#define BLTOUCH not defined

Exalik, 2020-04-05:
Hello Grogyan, thank's for your help

The BLTOUCH directive is defined through the TOUCHPROBE directive which is in Start_here.h file.

Grogyan, 2020-04-05:
On further look into it yesterday, your configs do not seem to match stock marlin.

Are you sure you downloaded it from the main branch, or from another repository?

Exalik, 2020-04-06:
You are right Grogyan, it is a fork of Marlin but just a pre-configured build for Geeetech printers.

WDuck, 2020-04-08:
Same problem here. I Used the original branch and configured myyself. Board is SKR Pro V 1.1.
Controlling the BL Touch by M280 works and also the triggering is recognized according to M119.
BL Touch is a genuine one.
Lucky that i'm not the only one with this issue.
When I find time  tonight I will roll back to an earlier release and let you know the result.

Kind regards,
Philipp

[WDuck_Marlin_2_0_5_3_config.zip](https://github.com/MarlinFirmware/Marlin/files/4451881/WDuck_Marlin_2_0_5_3_config.zip)

Exalik, 2020-04-08:
Thank's WDuck for your test.

I use a previous version for the moment but I wish this to be solved.

WDuck, 2020-04-08:
Hm, even with 2.0.3 it doesn't work.
What previous version are you using?

WDuck, 2020-04-08:
This is the output from "DEBUG_LEVELING"
<22:28:17.701 : N15 M111 S32*100
22:28:17.701 : echo:DEBUG:LEVELING
22:28:26.405 : N16 G28*36
22:28:26.405 : >>> G28
22:28:26.405 : Machine Type: Cartesian
22:28:26.406 : Probe: BLTOUCH
22:28:26.406 : Probe Offset X-34.00 Y-25.00 Z0.00 (Left-Front & Same Z as Nozzle)
22:28:26.406 : Auto Bed Leveling: UBL (disabled)
22:28:26.408 : Raise Z (before homing) to 4.00
22:28:26.408 : >>> do_blocking_move_to X0.00 Y-14.00 Z4.00
22:28:26.408 : <<< do_blocking_move_to
22:28:27.569 : >>> homeaxis(X)
22:28:27.569 : Home 1 Fast:
22:28:27.570 : >>> do_homing_move(X, -750.00, [50.00])
22:28:32.785 : <<< do_homing_move(X)
22:28:32.786 : Move Away:
22:28:32.786 : >>> do_homing_move(X, 5.00, [50.00])
22:28:33.125 : <<< do_homing_move(X)
22:28:33.125 : Home 2 Slow:
22:28:33.125 : >>> do_homing_move(X, -10.00, 25.00)
22:28:33.469 : <<< do_homing_move(X)
22:28:33.469 : >>> set_axis_is_at_home(X)
22:28:33.470 : Axis 88 home_offset = 0.00 position_shift = 0.00
22:28:33.470 : > home_offset[X] = 0.00
22:28:33.470 : current_position= X0.00 Y-14.00 Z4.00 :
22:28:33.470 : <<< set_axis_is_at_home(X)
22:28:33.472 : current_position= X0.00 Y-14.00 Z4.00 : sync_plan_position
22:28:33.472 : current_position= X0.00 Y-14.00 Z4.00 : > AFTER set_axis_is_at_home
22:28:33.472 : <<< homeaxis(X)
22:28:33.472 : >>> homeaxis(Y)
22:28:33.472 : Home 1 Fast:
22:28:33.473 : >>> do_homing_move(Y, -775.50, [50.00])
22:28:39.050 : <<< do_homing_move(Y)
22:28:39.050 : Move Away:
22:28:39.050 : >>> do_homing_move(Y, 5.00, [50.00])
22:28:39.391 : <<< do_homing_move(Y)
22:28:39.391 : Home 2 Slow:
22:28:39.392 : >>> do_homing_move(Y, -10.00, 25.00)
22:28:39.732 : <<< do_homing_move(Y)
22:28:39.732 : >>> set_axis_is_at_home(Y)
22:28:39.733 : Axis 89 home_offset = 0.00 position_shift = 0.00
22:28:39.733 : > home_offset[Y] = 0.00
22:28:39.733 : current_position= X0.00 Y-14.00 Z4.00 :
22:28:39.733 : <<< set_axis_is_at_home(Y)
22:28:39.734 : current_position= X0.00 Y-14.00 Z4.00 : sync_plan_position
22:28:39.734 : current_position= X0.00 Y-14.00 Z4.00 : > AFTER set_axis_is_at_home
22:28:39.734 : <<< homeaxis(Y)
22:28:39.734 : BLTouch Command :160
22:28:40.283 : BLTouch Command :90
22:28:41.083 : last_written_mode - 0
22:28:41.083 : config mode - OD
22:28:41.083 : home_z_safely >>>
22:28:41.087 : current_position= X0.00 Y-14.00 Z4.00 : sync_plan_position
22:28:41.087 : destination= X250.00 Y255.00 Z4.00 : home_z_safely
22:28:41.088 : >>> do_blocking_move_to X250.00 Y255.00 Z4.00
22:28:41.088 : <<< do_blocking_move_to
22:28:46.832 : >>> homeaxis(Z)
22:28:46.832 : Home 1 Fast:
22:28:46.832 : >>> do_homing_move(Z, -727.50, [4.00])>

Exalik, 2020-04-09:
I am using the 2.0.5.2 at the moment.

WDuck, 2020-04-09:
Also with 2.0.5.2 I dont get M401/M402 to work.

kolbenhans, 2020-04-10:
https://github.com/MarlinFirmware/Marlin/issues/17440#issuecomment-611727891

minosg, 2020-04-20:
@WDuck  I think you are following the same path as me. 

I had the Bilinear leveling crash the FW and the M401 M502 commands not working and freezing the printer and it had to do with EXTUI.

Do you by any chance have a custom LCD?

Please look at this bug report and the workaround for MalyanLCD

https://github.com/MarlinFirmware/Marlin/issues/17429

On another test can you verify if the probe manually deploys to begin with? 

`M280 P0 S10
M280 P0 S90`

There have been some patches moving around the timers for BL touch and you may have some PWM or Buzzer signal interfering with the probe.

closed 2020-05-01 by Exalik