MarlinFirmware/Marlin #17394: [BUG] Autoleveling does not compensate correctly if axes are inverted

gjmndfklgjretoi, 2020-04-03:
### Bug Description
First off, big thank you to the devs for this project!

Marlin 1.1.x.

Autoleveling apparently does not compensate correctly when X or Y axes are inverted. I suspect that either the probe offset needs to be inverted, the mesh itself needs to be inverted, or both. 

I think I personally have hit this rare corner case where:

1. All three of my axes are inverted.
2. I use bed leveling.

It seems like from asking around, people either have one or the other, but not both.

My printer is a Folger2020, all three axes are inverted with this printer. My probe is both "to the right" and "behind" the nozzle.

Symptoms: Parts of a 50x50x1 test cube will print correctly, but one corner will always print "too high". I will try readjusting the bed and redoing G29, but always the same symptom.

Also: I can confirm my printer does indeed print with the correct handedness. The output from my printer matches the orientation in slic3r exactly, with origin matching origin, and +200 matching +200 on all three axes.

I print with ABS but this is *not* an issue of warping. I get that also unless I put my printer in a cardboard box. This is a case of printing "too high" in one area.

### My Configurations

```
Send: M503
Recv: echo:  G21    ; (mm)
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D3.00
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X80.00 Y80.00 Z4016.00 E700.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X250.00 Y250.00 Z2.00 E22.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X1000 Y1000 Z5 E1000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P500.00 R500.00 T500.00
Recv: echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 B20000 S0.00 T0.00 X20.00 Y20.00 Z0.40 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S1 Z0.00
Recv: echo:  G29 W I0 J0 Z-2.35793
Recv: echo:  G29 W I1 J0 Z-1.86590
Recv: echo:  G29 W I2 J0 Z-1.46741
Recv: echo:  G29 W I3 J0 Z-1.12395
Recv: echo:  G29 W I4 J0 Z-0.86067
Recv: echo:  G29 W I0 J1 Z-1.57897
Recv: echo:  G29 W I1 J1 Z-1.08519
Recv: echo:  G29 W I2 J1 Z-0.70164
Recv: echo:  G29 W I3 J1 Z-0.32540
Recv: echo:  G29 W I4 J1 Z-0.07872
Recv: echo:  G29 W I0 J2 Z-0.78572
Recv: echo:  G29 W I1 J2 Z-0.32340
Recv: echo:  G29 W I2 J2 Z0.07558
Recv: echo:  G29 W I3 J2 Z0.42460
Recv: echo:  G29 W I4 J2 Z0.66647
Recv: echo:  G29 W I0 J3 Z-0.03863
Recv: echo:  G29 W I1 J3 Z0.41373
Recv: echo:  G29 W I2 J3 Z0.76665
Recv: echo:  G29 W I3 J3 Z1.12256
Recv: echo:  G29 W I4 J3 Z1.34434
Recv: echo:  G29 W I0 J4 Z0.68215
Recv: echo:  G29 W I1 J4 Z1.16954
Recv: echo:  G29 W I2 J4 Z1.55168
Recv: echo:  G29 W I3 J4 Z1.89763
Recv: echo:  G29 W I4 J4 Z2.11185

Recv: echo:PID settings:
Recv: echo:  M301 P22.20 I1.08 D114.00
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z-16.75
Recv: ok
```
### Steps to Reproduce

1. Create bilinear probed grid with: G28 G29.
2. Save mesh with M500.
3. My start gcode issues G28, then M420 S1, then the actual printing starts, so bed compensation is enabled for sure. I can confirm also that I can see the Z motors turning during the print, so *something* is happening in terms of compensation.

**Expected behavior:**

Bed leveling should work correctly despite axes inversion. I would expect to be able to print a 180x180 mm piece with the compensation being correct across the entire area.

**Actual behavior:**

The print is flat except for a "problem area". This "problem area" is almost always just one corner. I can readjust the bed, for example, set the bed to something random, then G28, G29, M500, G28, M420 S1, <actual printing code>, and it's always one corner that is printing too high or too low, usually too high.

#### Additional Information

I've replaced my servo three times now because I kept thinking it was that.

Output from probe test:
```
Recv: Mean: 1.341169 Min: 1.330 Max: 1.349 Range: 0.019
Recv: Standard Deviation: 0.006216
```
Always just one corner isn't compensated for correctly: 

[image]

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4429791/config.zip)

Thank you again devs!

nfons, 2020-04-14:
what leads you to believe its the inversion?

thinkyhead, 2020-04-20:
This is usually an issue where the probe is not staying true to the bed during the probing procedure, causing readings at one end of the axis to get thrown off. The only things you can do are ensure that the probe is getting readings that match up with the readings you get when you do manual probing. If you find that the meshes differ then you know for sure that the probe is not staying in perfect alignment.

thierryzoller, 2020-04-22:
https://github.com/MarlinFirmware/Marlin/issues/17399#issuecomment-616210265

boelle, 2020-06-20:
@gjmndfklgjretoi is the issue still there?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle