MarlinFirmware/Marlin #17396: ADAPTIVE AUTO BED LEVELING

inventabuild, 2020-04-04:
1.  CAN YOU PLEASE MAKE AUTO BED LEVELING ADAPTIVE SO IF I KNOW THERE IS A CORNER ON MY BED THAT PERFORMS POORLY WITH AUTO BED LEVELING I CAN CREATE A SEPARATE FRONT,RIGHT,BACK,LEFT BED LEVELING ZONE  AND ADD EXTRA PROBES FOR THAT AREA?

2.  IT WOULD BE NICE IF MARLIN COULD AUTO-DETECT AREAS W/ HIGH VARIATION AND ADD EXTRA POINTS TO PROBE IN THAT AREA AND CREATE A SEPARATE PLANE FOR THAT AREA WHEN THE HOT END ENTERS IT TO PRINT.

Grogyan, 2020-04-04:
Tried Unified Bed Leveling (UBL)?

inventabuild, 2020-04-04:
Thanks Grogyan, perfect timing, I actually just started watching Chris Riley's UBL How To video on YouTube when I got your message.

It's a little bit more manual than what I had in mind, but it looks like a good solution for problem areas on the bed.

Does UBL work good for Buildtak,Prusa and other removable flexible beds or does taking these beds off and on after every print change the actual position of the probed points?

inventabuild, 2020-06-24:
UBL was not the panacea I was hoping for.  As room and bed temps change the matrix loses accuracy so multiple matrices need to be juggled and it gets to be a little cumbersome for my liking.  Of course other people's mileage may vary.

I prefer bilinear bed leveling before every print and I think "Adaptive" Auto Bed Leveling is still a worthwhile feature if it can be implemented.  Spending more time probing on the larger z high-to-low delta areas and less time on more planar areas seems like better accuracy can be accomplished in shorter probe time.

thinkyhead, 2020-06-25:
You don't need to remove it. There is now a feature that will adapt. It is covered in this Chris's Basement video. https://www.youtube.com/watch?v=AjB0ZucPf9s

closed 2020-06-25 by thinkyhead