MarlinFirmware/Marlin #17397: [BUG] Mix at 0%;0% on power up (Marlin 2.0.5.2)

Exalik, 2020-04-04:
### Bug Description

At power up, the mix is at 0%;0%

If a print is launching without change the mix ratio, the printer freeze just after the bed leveling.
[Marlin 2.0.5.2 config.zip]

### My Configurations

(https://github.com/MarlinFirmware/Marlin/files/4430894/Marlin.2.0.5.2.config.zip)

### Steps to Reproduce

1. Power up the printer
2. Look the Auto Home and the nozzle to go up for about 5mm
3. Look the beautiful screen... freezed (lol)

**Expected behavior:** The print should start and the screen should not freeze

**Actual behavior:** The print does not start and the screen is frozen

#### Additional Information

Geeetech A10M (GT2560 v3.1) with TMC2208 and 3DTouch

pitercal99, 2020-04-12:
I'm also having the same problem
I tried to save on EEPROM but V-Tools settings are not included into the EEPROM data

thinkyhead, 2020-04-27:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

closed 2020-04-27 by thinkyhead

Exalik, 2020-04-27:
This issue has been recently solved in the bugfix-2.0.x