MarlinFirmware/Marlin #17398: [BUG] Marlin 2.1

roel8032, 2020-04-04:
After printing a second product, the leveling speed is much higher. (G29). Probably G28 also. Pritner: Infitary M508.

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle

roel8032, 2020-07-17:
> Lack of Activity
> This issue is being closed due to lack of activity. If you have solved the
> issue, please let us know how you solved it. If you haven't, please tell us
> what else you've tried in the meantime, and possibly this issue will be
> reopened.

The big survived until today. It is not to be found in the config files.
I think the software reads the: DEFAULT_MAX_FEEDRATE from configuration.h and when you repeat the print it does not. Or something like that.
#define SD_REPRINT_LAST_SELECTED_FILE is set in configuratio_adv.h

The bug appears for Anet A6 printer and the Infitary M508, at least.
The bug is to be found in code i think.

thinkyhead, 2020-07-17:
The "max feedrate" can be changed by G-code and some slicers might decide to insert commands which do that. If it's a concern, send `M501` in between prints.

roel8032, 2020-07-18:
You are right. Now i have M501 before G28 and G29 in the Gcode files and the problem is gone.