MarlinFirmware/Marlin #17399: [BUG] Marlin 2.0.5 - Auto Bed leveling not work - no compensation

ashinhood, 2020-04-04:
### Bug Description
I have the Marlin 1.1.8 version working correctly with Bltouch and Auto bed levelling linear
Then i decide to upgrade to Marlin 2.0.5 but there is a problem about the bed leveling, i tried all type of bed leveling, all do a good result M48 have 0.003 difference beetween the probing, G29 T give me a good matrix of result of the probing but during the printing process the printer not compensate the gap beetween the part of the bed lower.

For test i put back the Marlin 1.1.8 and all work correctly again

### My Configurations

This zip contain my configuration for 1.1.8 working and 2.0.5 with problem
[marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/4430943/marlin_config.zip)

### Steps to Reproduce

1- G28
1- G29 
1- Start printing process

**Expected behavior:**    First layer printed correctly

**Actual behavior:**   There is a part of the first layer too high not stick to bed

#### Additional Information
I flash again with old version, and now i'm printing correctly

hille78, 2020-04-04:
Im having the same exact problems with 2.0.5 if it actually does probe all points.
But most of the time it probes once or 3 times and just halts with an error

scaery, 2020-04-04:
Try to enable RESTORE_LEVELING_AFTER_G28

```
/**
 * Normally G28 leaves leveling disabled on completion. Enable
 * this option to have G28 restore the prior leveling state.
 */
#define RESTORE_LEVELING_AFTER_G28 // always do leveling before print
```

You did already specified Z in NOZZLE_TO_PROBE_OFFSET. Are you sure it is always correct like that?

Still, Im not familiar with the Bltouch, but that would be my first search to correct the mentioned behaviour.

ashinhood, 2020-04-04:
I specified the Z nozzle offset because is the value tested, but the distance is not correct in every part of the bed, but only in 1 portion of the bed, that let me think about is not an OFFSET problem but something about auto bed leveling compensation that not work correctly.

I tried to enable RESTORE_LEVELING_AFTER_G28 same problem

scaery, 2020-04-04:
Okay, BlTouch has a lot of troubles if you look in the issue list. Im not sure if your problem was not already been mentioned, else just stick with the old firmware which works currently or try the bugfix release. Maybe a duplicate of #17361

Grogyan, 2020-04-05:
First, when you change firmware, you must do a M50, followed by a M500

I noticed that you are using safe homing.

Have you by chance noticed if the G28 probe position is actually at the centre of the bed with the probe offsets?
For myself, with UBL, probe offsets aren't being applied, due to the requirement to use Safe Homing, which on my machine I3 Mk2.5 RepRap SKR1.3.  
Ie bed centre is exactly (Probe_Offset/2) short from it is supposed to be

thinkyhead, 2020-04-05:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

Grogyan, 2020-04-05:
[Safe_home_Mesh.txt](https://github.com/MarlinFirmware/Marlin/files/4432834/Safe_home_Mesh.txt)
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4432836/Configuration.zip)

I understand that my config is UBL, however, the output mesh data is present in the log.
Take note of G28 position, this is safe_homing which is offset from it's true location by 50% of the Probe_offsets

thestephenmarshall, 2020-04-07:
I seem to be experiencing the same issues as OP. Lack of leveling adjustments is consistent with the grid produced running:

- `G29 P1`
- `G29 P3`
- `G29 J`

Grid looks good to go, but regardless I end up with incorrect Z offset (too low or too high). Before I disable my printer for an undetermined amount of time to produce the `.txt` file required, I am trying out [2.0.4.4](https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.4.4) which would be the approximate version at which I can recall ABL to be working as expected. Will post back with results after a few prints.

Talha909, 2020-04-08:
> I seem to be experiencing the same issues as OP. Lack of leveling adjustments is consistent with the grid produced running:
>
> * `G29 P1`
> * `G29 P3`
> * `G29 J`
>
> Grid looks good to go, but regardless I [...]

I will wait for your result. I also have the same problem. One time very high and time very low. Yesterday it was working fine with 2.5.3

thierryzoller, 2020-04-12:
The Grid I print shows that it doesn't compensate. The only reason I can print is because I actually manually level it before ABL. Which defeats the purpose of having it

delskev, 2020-04-12:
I have the Same problem with the BLTouch v3.1.
I tried different firmware based on Marlin .
And I end up like @thierryzoller ... manually level the bed...

thestephenmarshall, 2020-04-13:
Update here after almost a full week running `2.0.4.4` I am no longer experiencing these issues with no compensation. I am still not 100% sure of anything at the moment as I did end up adding `mean` compensation steps to the end of my leveling procedure:

- `G29 P5` - get the mean
- `G29 P6 C[mean]` - apply the mean to the mesh

That being said, I am using a BLTouch v3.1 and right after moving back to `2.0.4.4` I did indeed see improvements. After I finish my project (sometime this week) and can fool around with it, I will grab the data requested above from both versions for comparison.

nfons, 2020-04-14:
@thestephenmarshall  i will have to try that out as well. I have a very similar issue in https://github.com/MarlinFirmware/Marlin/issues/17262

thestephenmarshall, 2020-04-16:
Unfortunately, my Y axis bearings need replacement so I will not be able to proceed until I receive the Misumi replacements in a few days.

nfons, 2020-04-16:
@thestephenmarshall  did you try this with bilnear previously? or was it only UBL

thestephenmarshall, 2020-04-16:
@nfons I did not try with bilinear - only UBL.

jimschaefer, 2020-04-16:
I recently upgraded my cheap Tronxy X8 Printer (Melzy V2_5 controller board) with marlin 2.0.X bug fix and got a BLtouch clone (3dTouch from Geetech) working and also a heated bed with a glass plate.  I was expecting the 3 point bed leveling to work better than my previous manual leveling process but it seems to compensate incorrectly or very poorly.  I manually leveled bed and one layer brim printed OK but with bed leveling one side is higher than the other so much that adhesion is not good enough to print.  I am not sure  if it is hardware or software related and would like some feedback.  

I ran some M48 V  probe tests and the standard deviation comes back  with a standard deviation anywhere from 0.026 to 0.056.  Z speed around 60 mm/sec but slower speeds actually did worse for some reason.   I put the system into a debug mode while probing (see attached file) and i was getting s second probe discrepancy anywhere from 0.09 to -0.40 with heaters on or off, didn't really matter. Note... my numbers above are from multiple tests, I just attached one of them to this post.

My initial thought is there is just too much variation in the values the probe is reporting but I have no experience to know if this is true or not.  
I think the errors could be due to the following, but I am not sure how to rule things out:
- backlash in Z, mechanical capabilities (prints OK when manually leveled though)
- inferior quality BLtouch clone triggers at different points
- stepper tuning or step loss. I noticed as soon as the probe triggers the z axis jerks to a stop. Acceleration parameters have no impact on how smoothly probing moves stop
- noise picking up the 10 ms trigger from the BL touch clone.  I removed the capacitor from the board Z endstop per BLtouch technical note on their website. turned off heaters (not servos though)Wires are not shielded or twisted to BLtouch.
- software problem in Marlin 2.0.X bugfix or configuration issue.

Any suggestions on how to narrow this down ?  Files attached. 
Thanks !

[ConfigFilesAndLog.zip](https://github.com/MarlinFirmware/Marlin/files/4488935/ConfigFilesAndLog.zip)

Other comments:  The 2.0.x bug fix really improved the BLtouch logic.. on the mainline 2.0.5.3 the probe extend retract is VERY inconsistent. I almost cracked my glass bed on homing.    However, sometimes the BLtouch servo movements (pin up/pin down) get really flaky. Pin won't go down/won't come up, alarms, etc.  Rebooting board seems to always fix the problem.  I don't know how the PWM signal is generated for the device but wonder whatever caused the performance improvement since 2.0.5.3 doesn't always fix the issue, maybe changes the PWM signal ?  Just a thought.  I was reading how challenging it has been to get stepper pulses the correct width on the low end boards like mine.

GotoHack, 2020-04-19:
I have the same problem. If you check out fc74c6a you should have a working version with BLTouch.
I do not have time some far to dissect and find where the bug was introduced.

lordnibbler22, 2020-04-22:
Hello All,

i am currently running 2.0.5.3 on 2 printers with a SKR 1.4 turbo board. and i got some really strange first layer issues with UBL. I switched to ABL and found that the first probe after power ON or reset produces absolutely wrong results. when i run a G29 again the mesh is good. I can verify it on 2 different machines.

Probe is a PINDA v2 on a Prusa Bear with SKR 1.4 Turbo.

Printer 1 G29 after Power ON/reset

[image]

Printer 1 G29

[image]

Printer 1 G29 again

[image]

Printer 2 G29 after Power ON/reset

[image]

Printer 2 G29

[image]

Printer 2 G29 again

[image]

sergiofagundes, 2020-04-24:
Is the Bed leveling compensation to be used only during print of gcode files or it will work even if using a direct G0/G1 Zxx on pronterface or similar software ?

I can't get to use compensation on pronterface for example:

G28
M420 S1
G0 Z1
G42 I0 J0

So after this I move in lcd menu the Z down (-0.025 steps) till the nozzle hits the bed, but when it happens it's not a 0 (Zero), Z is much high and doesn't match the leveling mesh data.

That's why I'm questioning if this is meant to work only during prints (I can't test a print right now, since I'm out of filament :)

sergiofagundes, 2020-04-25:
Oh well, just installed today (25/04/20) the bugfix and now the compensation is working. Answering my own question ;-). Yes the compensation is meant to use in all movements (while M420 S1 = bed leveling ON). I am out of filament but have tested every point in the leveling points and all are good (with the new bugfix downloaded today).

thierryzoller, 2020-04-26:
Consindering many reports that ABL works if the printer is restarted but not after cancelling a print or starting a second one. Can you confirm that's the case once you got filament please. Thank you.

thinkyhead, 2020-05-03:
> I switched to ABL and found that the first probe after power ON or reset produces absolutely wrong results. when i run a G29 again the mesh is good. I can verify it on 2 different machines.

@lordnibbler22 — It would be interesting to have a look at the logs.

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.

---
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.

Repeat this procedure as needed to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on.

---
- Copy the log output into a **.TXT file** and attach it to your next reply.

Azrael321, 2020-05-03:
Today i started a print with the latest Bugfix (03.05.2020). I can notice, that with ABL_Bilinear, there is no movement in Z during the first layer. After the first layer, ABL seems to work as it should (i can see compensation movements in Z)

thestephenmarshall, 2020-05-07:
Very interesting! In fact, I am seeing the same thing that @Azrael321 is mentioning. After the first layer, the z-axis motors are adjusting according to the mesh data. Here is the 5x5 UBL mesh data I used in this example:

```
Recv:     ( 20,180)                      (175,180)
Recv:         0       1       2       3       4
Recv:  4 | -0.235  -0.068  -0.004  -0.040  -0.063
Recv:    |
Recv:  3 | -0.226  -0.082  +0.005  -0.031  -0.054
Recv:    |
Recv:  2 | -0.217  -0.095  +0.014  -0.022  -0.045
Recv:    |
Recv:  1 | -0.228  -0.096  +0.015  -0.033  -0.056
Recv:    |
Recv:  0 | -0.236  -0.100  +0.014  -0.016  -0.040
Recv:         0       1       2       3       4
Recv:     ( 20, 20)                      (175, 20)
```

## First Layer

[image]

## Second Layer

[image]

thestephenmarshall, 2020-05-08:
> Update here after almost a full week running `2.0.4.4` I am no longer experiencing these issues with no compensation. I am still not 100% sure of anything at the moment as I did end up adding `mean` [...]

Just need to update/correction to this bit. So the "improvements" I was seeing here in `2.0.4.4` were actually a false positive as the prints I ran to test were all done right after powering up the machine and not subsequent prints or retrys. This means I did not power cycle the machine either. Seems this bug exists in `2.0.4.4` as well.

thierryzoller, 2020-05-09:
Sounds similar/same then https://github.com/MarlinFirmware/Marlin/issues/17872#issuecomment-624836618

thinkyhead, 2020-05-10:
> …the z-axis motors are adjusting according to the mesh data…

I found something similar in my testing, and did a bunch of logging, but I was never able to track it down to the exact cause. But that seems to be the right approach, so let's try that. I'll get some extra logging in tomorrow and play around with it, and then I'll post for you all to try, and hopefully we'll find the reason for the weirdness.

As a quick experiment, does the effect go away if you add `M420 S0` to the end of the G-code file?

lordnibbler22, 2020-05-11:
@thinkyhead 
some logs with the latest bugfix from today. looks like the wired first run mesh is gone.

[1_G29.txt](https://github.com/MarlinFirmware/Marlin/files/4612482/1_G29.txt)
[2_G29.txt](https://github.com/MarlinFirmware/Marlin/files/4612483/2_G29.txt)
[3_G29.txt](https://github.com/MarlinFirmware/Marlin/files/4612484/3_G29.txt)

thestephenmarshall, 2020-05-15:
One more set of updates from testing various versions of Marlin 2.0.x:

- Issues exists as far back as `2.0.3` (maybe further back?)
- bugfix branch doesn't resolve anything for me (I did try the [SHA listed here](https://github.com/MarlinFirmware/Marlin/issues/17399#issuecomment-616210265) as well)
- After printing about 20 different varying calibration shapes, one pattern seems to stand out: within a 5x5 mesh, incorrect mesh data is being consistently generated in I1 and I3 columns. From the looks of it, the data seems to be inverted in these columns. I verified this by manually editing the columns e.g. `I2 = (I1 + I3) / 2` which yielded the expected bed adhesion and extrusion width.

What I've done as of now that seems to work fine:

1. Always start as first print (reset button or power cycle) before printing
2. The above doesn't work alone now (for whatever reason), so I've stopped using `G29 P1` until the issues listed here and in many other similar issues are resolved
3. To get a workable mesh, I've returned to manual bed leveling. This isn't good enough alone, so:
4. finally, using `G29 J` (tilt mesh) after `G34` (align Z axis) **on a zeroed mesh** gives me the results I need to continue printing

Figured I would share since I haven't had much luck with any other methods.

thierryzoller, 2020-05-15:
What do you mean by "zeroed mesh" in #4

thestephenmarshall, 2020-05-15:
I did try [Bilinear ABL](https://marlinfw.org/docs/gcode/G029-abl.html) as well, but no dice there as well. I am very used to UBL, so I'm going to stick with that unless someone finds something interesting.

thinkyhead, 2020-05-16:
@thestephenmarshall — If your bed is very flat and only tilted, you can use LINEAR instead, and you might get nicer results.

I'm not sure about the hypothesis that every other column is negated. There is no logic in Marlin that cares about rows and columns with respect to the results returned by the probe. The Z value is based solely on the Z position at the time the probe triggers. If the triggered Z position in the planner wasn't making it back to the current position, the error would accumulate, not oscillate.

You may want to try running a variety of `M48` tests to make sure that the probe is behaving repeatably in all areas of the bed, just to rule out unusual probe behavior.

thinkyhead, 2020-05-16:
Thanks for the logs, @lordnibbler22.

The first log seems to show that the Catmull-Rom subdivision isn't quite working. I'll have to mess around with that.

The second log seems copacetic.

The third log also looks good. The Z position differs throughout by only 20µm.

I will run several bed leveling tests today and do tiny print jobs in between them.

> "ABL works if the printer is restarted but not after cancelling … or starting a second print

@thestephenmarshall — Others have mentioned that the Z discrepancy issue occurs between prints but does it really require a print? Or, will it manifest if any XYZ moves are done? If it requires a print to be started and finished (or canceled) it makes me wonder: Why should a print make a difference?

With that in mind, please send along a small G-code file that exhibits the problem, so that I have the exact START and END G-code from your slicer. This might be an important factor.

nfons, 2020-05-17:
@thestephenmarshall  not sure if you mentioned...what board are you using?

thestephenmarshall, 2020-05-17:
@nfons I am using SKR v1.3

thestephenmarshall, 2020-05-17:
> @thestephenmarshall — If your bed is very flat and only tilted, you can use LINEAR instead, and you might get nicer results.

Good advice. I will have to try this out next. I do think  something might be up in the BLTouch area of the hardware on my setup. Perhaps some sort of magnetic interference. It’s really strange tbh as the print surface is 4mm borosilicate glass.

thestephenmarshall, 2020-05-17:
This is a highly modified MPMS v2.1. 

Only thing left is the frame, rods, z assembly and Y motor. BTT SKR v1.3, TMC 2209 v1.2 drivers on X&Y, TMC2208 v3 dual drivers on Z. 

Flexion HT extruder (not that it makes a difference here), but that demanded that I either decrease microsteps or upgrade the extruder motor to a Quimat > 2A capable motor (which I did) for more torque which ate up and extra 5mm on the X axis. 

Bed is `200x195mm` with `8mm` swiss clip gutters. 

`M851` - `Probe Offset X0.00 Y-42.00 Z-2.40` Probe is aligned right in front of the direct drive assembly. 

Results from the Probe Accuracy Test 

Center - `M48 P5 L5 X100 Y100`
```
Mean: -0.008500 Min: -0.013 Max: -0.005 Range: 0.007
Standard Deviation: 0.002549
```

Front Left - `M48 P5 L5 X8 Y8`
```
Mean: -0.268500 Min: -0.273 Max: -0.263 Range: 0.010
Standard Deviation: 0.003391
``` 

Front Right - `M48 P5 L5 X187 Y8`
```
Mean: 0.016500 Min: 0.010 Max: 0.022 Range: 0.013
Standard Deviation: 0.004062
```

Rear Left - `M48 P5 L5 X8 Y158`
```
Mean: -0.135000 Min: -0.138 Max: -0.130 Range: 0.007
Standard Deviation: 0.002739
```

Rear Right - `M48 P5 L5 X187 Y158`
```
Mean: -0.063000 Min: -0.068 Max: -0.058 Range: 0.010
Standard Deviation: 0.003674
```

The deviations seem within tolerance.

I switched to Linear and here are the `M420 V` results

```
Send: M420 V
Recv: Bed Level Correction Matrix:
Recv: +1.000000 +0.000000 +0.000789
Recv: -0.000000 +1.000000 +0.000335
Recv: -0.000789 -0.000335 +1.000000
```

How could `X8 Y158`, `X100 Y100` and `X187 Y8` all be `+1.0mm`? To me, this matrix looks odd. `G29 J` on UBL doesn't produce these anomalies.

wakille, 2020-05-17:
Hey guys,
so I am also suffering from first layer problems with the combination of a SKR 1.3 and 3DTouch Probe (newest gen). I tried UBL and ABL, and results always in the same outcome. For x<100 the bed is too high or for x>100 too low (depending on set Z-Offset). I now did some testing with ABL. First I ran G29 and afterwards also probed the same points with M48. Here are the results:

G29
    0      1      2
0 +0.350 -0.051 +0.224
1 +0.096 +0.001 -0.057
2 +0.269 -0.029 +0.301

M48

    0      1      2
0 +0.336 -0.056 +0.209
1 +0.083 -0.013 -0.050
2 +0.234 -0.040 +0.299

Exact results with M48 P4 X*** Y*** P4 L2 E V2

X|Y : Mean(Standard Deviation)
10|10: 0.336190(0.002428)
100|10: -0.056023(0.001864)
190|10: 0.209440(0.000675)
10|100: 0.083323(0.003135)
100|100: -0.013386(0.002079)
190|100: -0.050436(0.001037)
10|190: 0.234158(0.001353)
100|190: -0.040897(0.001938)
190|190: 0.299456(0.002369)

Difference between G29 and M48:
    0               1         2
0 +0.014 +0.005 +0.015
1 +0.013 +0.014 +0.007
2 +0.035 +0.011 +0.002

Bed size: 200x200

I had some issues logging the process so I cant provide it at the moment. But I will do the same test with UBL and provide also logs. Also I have this issue all the time, not only after aborting a print.

Mils24, 2020-05-18:
Hi guys, 

Ive commented on a few of these similar threads as Im not sure which one is most active.

Ive followed your request @thinkyhead please see debug file for me attached.

Printer: Homemade delta
Motherboard: MKS SBASE 1.3
Firmware: Marlin Bugfix 2.0.x (currently) also tried Marlin 2.0.5 before moving to Bugfix as I had read somewhere that Bugfix worked with ABL.
Sensor: Triangle Labs 3DTouch (BL Touch clone)
ABL type: Bilinear ABL

[Debug.txt](https://github.com/MarlinFirmware/Marlin/files/4643058/Debug.txt)

Mils24, 2020-05-18:
Ive been doing a bit more thinking and more reading of other threads on here with similar issues.

Seems a few people have linked odd ABL and BLTouch issues to Z_SAFE_HOMING.

Question - With Z_SAFE_HOME ON on a Delta should the endeffector move to the centre of the bed before homing up to the top? (assuming you have the safe homing set to (delta_printable_radius)/2??? 

This is something that has puzzled me also. I have Z_SAFE_HOMING on with the safe point set as per above. However when my prints finish or I'm I'm testing points of the bed and then home, my printer just moves straight up. My expected behaviour was move to centre of the bed and then up but it's not doing that.

If my Z_SAFE_HOMING isn't working properly and some people think its linked to ABL issues could this be my problem?

Is it crazy to tun it off?

thinkyhead, 2020-05-18:
If you set your `Z_SAFE_HOMING` point to the opposite of your probe offset, then the effector will be in the center of the bed for safe homing.

thinkyhead, 2020-05-20:
> For x<100 the bed is too low or for x>100 too high

@wakille — This almost certainly indicates that you have a twist in your X gantry, which is a common problem. It causes probe readings to become distorted, and there is nothing unfortunately that the firmware can do to fix the issue. You must repair the twist in your gantry.

thinkyhead, 2020-05-20:
> How could X8 Y158, X100 Y100 and X187 Y8 all be +1.0mm? To me, this matrix looks odd. G29 J on UBL doesn't produce these anomalies.

@thestephenmarshall — You are confusing a MATRIX with a probe grid. A matrix is used to tilt all points in a 3D space. Those numbers are not the measured Z heights, but are calculated by the Least-Squares method. See https://marlinfw.org/docs/gcode/G029-abl-linear.html

wakille, 2020-05-22:
> @wakille — This almost certainly indicates that you have a twist in your X gantry, which is a common problem. It causes probe readings to become distorted, and there is nothing unfortunately that [...]

Ok, thanks for the tipp. But I dont unterstand clearly what´s wrong with my gantry. Do you think my rails are bent or that there is a height difference between the left and right side of my x-axis ? And is there a way to be sure, that this is my issue, like can I meassure something to confirm this theory ? Because e.g. I meassured the height difference and is at maximum 0.4mm. Thanks a lot for you help!

thestephenmarshall, 2020-05-26:
Back to provide some more updates. I have tried these things:

1. `G29 J2` - probe mesh at the four corners. Grid looks good, but compensation is still off when printing. Still seems like too much compensation (same areas as my previous posts). 

2. Found that `G29 S` should probably always be run declaring the slot ala `G29 S0` as an example. During tests, I learned that simply running the command without the slot parameter can potentially save the data in some strange undefined place.. Who knows where it went (and I never explicitly selected any other slot than 0) ¯\_(ツ)_/¯

3. `G29 J5` (or higher number of points) produces same strange I1 and 13 grid points as `G29 P1`

4. Since the meshes I have provided above sure do look like bend X and/or Y rods, I did these things:

  - Completely disassembled the printer to check if the rods are bent. Rods not bent! (though unless I am mistaken even ABL should be able to compensate for this)
  - Ran `G29 P0` followed by `G29 S0` to ensure a zero mesh start (machine true)
  - Started my next print which spans the length and width of the bed
  - Upon noticing very similar results to the tests above, adjusted the 8,8 and 192,192 bed knobs slightly (while printing) until the line widths were consistent. Now the printer is laying down perfect layers including the first.
  - If the BLTouch was picking up bent rods and my rods were truly bent, shouldn't my line widths and layers be reflecting this? e.g. nozzle dragging, line width too narrow (gaps between lines)?

Attached is the current 9-10 hour print sliced with Cura 4.6.1. Not using any experimental features or flow equalization.

[WDI3_main-corners-mid-Corners-Mid.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/4684230/WDI3_main-corners-mid-Corners-Mid.gcode.zip)

Rage01, 2020-06-13:
Did we ever find out a fix for this? Or what is causing the inconsistency of the bed levels? Sorry I am not much of any help.

bitshiftr, 2020-06-14:
I too have this issue with my Ender 5 (SKR mini E3 v1.2 and bltouch v3.1) running bugfix branch. I can clearly see during my first layer that compensation is not being applied in Bilinear ABL. Since my bed is fairly level to start with, I am still able to manage, but it's clear that the layer is too thin at some places vs. others.

Mils24, 2020-06-15:
I have given up on trying to get Bilinear to work. I had actually given up on Marlin and moved to Smoothieware but unfortunately I couldn't get me MKSTFT to allow me to do baby stepping with smoothie so Ive came back to Marlin. This time however I've opted for UBL instead of bilinear and it seems to work.

wakille, 2020-06-15:
Disabled Bed Leveling and now printing without better than with bed leveling enabled. Checked also my mechanical setup and found no issue. And I also still don't understand, why ABL should not compensate for bent rails. And if the measurement would be distorted, shouldn't be the repeatability be bad ? For me this still looks like a bug, but maybe I am just too inexperiecend. Thought about using the RepRapFirmware Fork for the SKR boards.

Mils24, 2020-06-15:
I defintly think there is some sort of bug in Marlin. Its just odd that it happens for some and not all. Makes me wonder if there is another setting that some of us have enabled that is cause it not to work.
My z_safe_homing is still not working either but I have just gave up on it. I tried as per thinkyheads instruction but setting it at 0,0 or the opposite to my ZProbe offset doesn't work. My head always just goes straight up from the last known XY position instead of moving to 0,0 and then homing.
I also thought about trying the RepRap fork but I don't have the WiFi ESP board and I don't really want to buy any more parts for this machine. I already bought Raspberry pi and have it hooked up to Octoprint therefore adding the wifi module would feel like the Pi was a waste of money. 

For now I'm just going to stick with UBL enabled and hope that sometime in a future release the Bilinear gets fixed. I would prefer bilinear as with my MKSTFT I don't have the functionality to easily adjust the mesh and therefore bilinear seems like it would be easier.

AnHardt, 2020-06-15:
ABL (Auto BED Leveling) is strictly only to compensate for tilted beds. Nowadays we also can improve printing on bumpy beds (mesh based and UBL). ABL is NOT meant to solve any other geometry problem of the printer. 
Any probe based leveling system is particularly sensible to errors changing the difference between nozzle-tip and trigger-point of the probe with the position of the nozzle. [For example](https://github.com/AnHardt/Marlin/issues/84)
To avoid that probing problem you could use good old manual-mesh-bed-leveling (using the nozzle as a probe) or could overrule the probe by editing (overruling the probe) the mesh-points with UBL. 

Always when you see a **systematic error** (left/right/front/back/diagonal too high/deep) it's very likely to be a distortion of the frame to make the probing fail - resulting in a leveling based on wrong data. Or leveling is simply not switched on.

boelle, 2020-06-21:
@ashinhood still an issue?

thinkyhead, 2020-06-21:
Please test the latest [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

shitcreek, 2020-06-22:
We've tested the latest build with ABL Bilinear and found no issue when we did an exaggerated test ie setting the bed leveling compensation mid point to 2mm (using `M421`). We saw that the print head moves up and down accordingly.
I ran this test gcode repeatedly and the logs produced are identical:
[leveltest-dr.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/4815924/leveltest-dr.gcode.zip)
[image]

Bed leveling results are within acceptance (biggest difference was 0.007)
```
Bilinear Leveling Grid:
measured_z = [
 [ -0.250, +0.032, +0.323 ],
 [ -0.388, -0.039, +0.099 ],
 [ -0.414, -0.055, +0.275 ]
];

Bilinear Leveling Grid:
measured_z = [
 [ -0.258, +0.036, +0.323 ],
 [ -0.389, -0.039, +0.100 ],
 [ -0.423, -0.053, +0.268 ]
];

Bilinear Leveling Grid:
measured_z = [
 [ -0.270, +0.026, +0.320 ],
 [ -0.405, -0.044, +0.098 ],
 [ -0.439, -0.062, +0.256 ]
];
```
However when I did some test prints, line adhesion to each other on 2nd prints appear to not be a great as the first prints after power cycling (first prints weren't perfect either). But this could have been caused by something else perhaps.
Nevertheless by simply squishing the first layer a bit more fixes this completely:
[image]

I am pretty confident that ABL is working as it should.

boelle, 2020-06-23:
@shitcreek i will risk my neck and close this one then

it also works for me

closed 2020-06-23 by boelle

aSocialMenace, 2020-08-21:
I am having this exact same problem.. Im using the btt skr mini e3 in my ender 3 pro and I'm running octoprint on a pi 4 (2gb) .. I've followed ever tut online and I can't get it to compensate in the first layer. I've complied this firmware a dozen times or so over the past couple weeks.. I've used "know good" configuration.h files and still she won't adjust in the first layer.. I can generate and view the mesh in octoprint but the printer does not correct the z on the first layer.... I'm super frustrated here.. anyone have a fix for this? Anyone know that last good version of Marlin that bl touched worked with??? I am considering reverting back to 1.1.8 as mentioned above but I'm unfamiliar with how to compile that version with the btt skr mini e3 board..

sognatore99, 2020-09-04:
I was (maybe still am) having the same issue with Marlin bugfix 2.0.x.  I switched out my 3d Touch with a newer one and still had same issues.  I googled the heck out of it and only came up with a note in the BL touch manual.  I chose the second option and had to put it also in the Marlin starup section in the "Configuration_adv" section. Note use /n to separate commands.  Once I did this the G28 command worked correctly, I however leveled the bed manually so I really can not say if it fixed the overall issue.  Here is the snippet from the BL Touch (works with generic also) manual:

Depending on your board, you can choose between the following two.
1. Logic voltage free(default) mode (Recommended) ← Both 3.3V /5V Logic are available
M280 P0 S160 ; BLTouch alarm release
G4 P100 ; delay for BLTouch 
G28 ; home
G29 ; auto bed leveling

2. If the nozzle is in contact with the bed after missing the trigger signal(A board with large capacity capacitor in end-stop input circuit, such as the Melzi).
M280 P0 S140; ← Only 5V Logic mode(Do not activate 5V logic on 3.3V logic system without 3.3V logic conversion)
G4 P2000 ; delay for BLTouch
M280 P0 S160 ; BLTouch alarm release
G4 P100 ; delay for BLTouch 
G28 ; home 
G29 ; auto bed leveling

Boards with large capacity capacitor in end-stop input circuit: Melzi and some of the Creality3D, ANET board, etc. (Select 1 if you have already removed the capacitor from your board)

jermbe, 2020-09-22:
Hi, something I read triggered me.

When people use TMC2208/TMC2209 drivers, they need to reverse their axis in the firmware (or reverse the connectors on the motherboard).
Can it be that if the axis are reversed in the firmware, that the ABL is correcting this on a non reversed bed?
My BL touch sees that my bed is low left, a high right. But when I print left the filament is almost scraped off, and right it is almost not fixed on the bed. => It is compensating the wrong place of the bed.

gbkwiatt, 2020-10-20:
I am having same issue, and I had same idea as above, that it might be somehow reversed, I can't make BLtouch work. I used to have BLtouch on ender 3 with marlin 1.8 and never had an issue. But just got ender 3 with board 4.2.7, and can't make it work properly, I can see on octopi and bed mesh visualisation that it does detect unevenness, but seems not compensating for it during print, I am giving up

[31 more replies]