MarlinFirmware/Marlin #17401: [BUG] (SD_FINISHED_RELEASECOMMAND issue in 2.0.5.3)

BaBBa71, 2020-04-04:
### Bug Description
In my custom Delta printer with firmware 2.0.5.3 an unknown "disable all stepper" occur even I have set only to disable E in _Configuration_adv.h_ with this line:

 ```
  #define SD_FINISHED_STEPPERRELEASE true         
  #define SD_FINISHED_RELEASECOMMAND "M84 E" 
```
even if I have defined also this line:

```
#define DEFAULT_STEPPER_DEACTIVE_TIME 60
#define DISABLE_INACTIVE_X false
#define DISABLE_INACTIVE_Y false
#define DISABLE_INACTIVE_Z false 
#define DISABLE_INACTIVE_E true
```

This occours only when I print from SD CARD. Same model (same GCODE with no disable command obviously at the end !!!) printed by serial USB connection don't disable the steppers.
 
I tried to change the value in `#define SD_FINISHED_RELEASECOMMAND` with a simple sound like `"M300 S4400 P3000"` but at the end of SD printing, after last command in gcode (homing with G28) no sound is played and the delta effector fall down.

### My Configurations

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4431171/config.zip)

### Steps to Reproduce

1. In a delta printer with graphical display 12864 and SD support enabled, modify this line in _Configuration_adv.h_ with this:

1st example config

```
  #define SD_FINISHED_STEPPERRELEASE true         
  #define SD_FINISHED_RELEASECOMMAND "M84 E" 
```

or 2nd example config

```
  #define SD_FINISHED_STEPPERRELEASE true         
  #define SD_FINISHED_RELEASECOMMAND "M300 S4400 P3000"
``` 

2. Select "Print from media" and select a simple gcode file with no disable steppers command (M18 or M84) at the end of printing code. 
3. Start print.
4. Wait the end of print
5. Observe the status of XYZ steppers.

**Expected behavior:**
No X Y Z steppers disabled at the end of print in both example. For the second configuration line a 3 second sound should be heard.

**Actual behavior:**
All steppers disabled in first conf.
All steppers disabled and no sound reproduced in second config.

#### Additional Information
My previous marlin firmware works perfectly with this parameters but have some issue (during printing) with usability of display,
Actually I made a workaround by setting "false" in SD_FINISHED_STEPPERRELEASE but I want to disable only the extruders steppers.

studiodyne, 2020-04-05:
[17393](https://github.com/MarlinFirmware/Marlin/issues/17393)

BaBBa71, 2020-04-05:
> I think all issues have the same source
> [17393](https://github.com/MarlinFirmware/Marlin/issues/17393)
> On 2020 03 19 , no bug , and after this date , sd printing freeze on any command in the buffer sent by lcd
> Please can you help me to confirm this , by trying all i explain in my issue page
> 
> Thks

I tried this and get [same issue](https://github.com/MarlinFirmware/Marlin/issues/17393#issuecomment-609389019)

BaBBa71, 2020-04-10:
> I think all issues have the same source
> [17393](https://github.com/MarlinFirmware/Marlin/issues/17393)
> On 2020 03 19 , no bug , and after this date , sd printing freeze on any command in the buffer sent by lcd
> Please can you help me to confirm this , by trying all i explain in my issue page
> 
> Thks

Sorry but this error does not have the same source. I have compiled today bugfix and the issue  [17393](https://github.com/MarlinFirmware/Marlin/issues/17393) was solved. But my issue still remain. The delta carriage falling down with:

`  #define SD_FINISHED_STEPPERRELEASE true `
 and
`#define SD_FINISHED_RELEASECOMMAND "M84 E"`

thinkyhead, 2020-04-12:
It seems this bug was introduced between 1.1.8 and 1.1.9. https://github.com/MarlinFirmware/Marlin/commit/8f26c3a6d37ca7284f11112c9f9c12595218176e#diff-93a564322f37b6240dd4384e982f9425

I'll patch it shortly.

thinkyhead, 2020-04-12:
The patch is now pushed. Grab the latest `bugfix-2.0.x` code to confirm that it works properly.

closed 2020-04-12 by thinkyhead

BaBBa71, 2020-04-12:
Wonderful. I installed the latest version and the error has been resolved. Thank you very much! @thinkyhead ... Marlin is the best 3D printer firmware ever!

commit 68fb0271 (2020-04-12): Fix SD_FINISHED_RELEASECOMMAND bug

Fixes #17401 bug that was introduced in 8f26c3a
commit 39fe6efb (2020-04-12): Fix SD_FINISHED_RELEASECOMMAND bug

Fixes #17401 bug that was introduced in 8f26c3a
commit 7aed32df (2020-04-12): Fix SD_FINISHED_RELEASECOMMAND bug

Fixes #17401 bug that was introduced in 8f26c3a