MarlinFirmware/Marlin #17409: Home axis from a custom M command

jgoo9410, 2020-04-05:
I have written a custom M command in the firmware. It all works fine but when my M command is entered and my function called, I’d like the X axis to home at the beginning of the function. I have tried using the        `homeaxis(X_AXIS)` function, but because I’m using tmc drivers, the necessary extra steps aren’t performed (switch to spreadcycle and monitor for a home trigger), the x axis crashes and never homes.

My question is; how do I call the G28 function from within a function inside marlin. And if possible, how would I go about only homing the X axis when the function has been called?

thinkyhead, 2020-04-05:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-04-05 by thinkyhead

jgoo9410, 2020-04-05:
:(

thinkyhead, 2020-04-27:
Please take advantage of the vast community resources available, and understand this is our development tool for tracking work that we need to do, and it is not a support forum.

```cpp
process_subcommands_now_P(PSTR("G28X"));
```

jgoo9410, 2020-04-27:
Thanks! Your colleagues were very helpful in a different thread, I’m over on Discord now. Il stay off here except for bug reports.