MarlinFirmware/Marlin #17412: ERROR COMPILING MARLIN 2.0.5.3

inventabuild, 2020-04-05:
Compiler throws 2 errors in types.h on the following lines of code.  The red squiggly occurs on float (bolded for reference)

// Conversion macros
#define feedRate_t(**float**(MM_M) / 60.0f)
#define (float(MM_S) * 60.0f)
#define MMS_SCALED(V)    ((V) * 0.01f * feedrate_percentage)

Why do I get these errors?
"(" may not appear in macro parameter list {76,25}
macro names must be identifiers {77,9}

ellensp, 2020-04-05:
You need to attach you config files. The issue is bound to be in there

inventabuild, 2020-04-05:
Thanks, configuration.h and configuration_adv.h attached
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4433630/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4433632/Configuration_adv.zip)

ellensp, 2020-04-05:
LULZBOT_DEFAULT_Kp ??  that is not standard marlin. What firmware are you really using?

inventabuild, 2020-04-05:
It's a true direct download of Marlin 2.0.5.3 from Github.

I copied over the Kp, Ki, Kd values for the E3D hotend from another version of Marlin and accidentally included the LULZBOT prefix.  I removed the LULZBOT prefix and still get the same errors.

inventabuild, 2020-04-05:
Do you still want me to download current marlin bugfix?

I see something I can download called "Patches to latest 2.0.x
Marlin 2.0 with bug fixes"

If I open that in Github desktop how will I incorporate the configuration.h and configuration_adv.h that I just spent hours setting up into the bug fix version?

ellensp, 2020-04-05:
can't replicate your error on 2.0.5.3 or bugfix. Your configs (apart from LULZBOT issues) seems fine.

inventabuild, 2020-04-05:
In other words you can compile my configuration.h and configuration_adv.h?

Are you using Visual Studio Code to compile?

ellensp, 2020-04-05:
Yes. they compile fine. Yes I'm using platformio under VSCode

inventabuild, 2020-04-05:
That's what I'm using.  Wow, what s/ I do if they won't compile for me?

inventabuild, 2020-04-05:
Also, just to make sure I didn't do something screwy, how do i verify the version of Marlin I'm attempting to compile.

inventabuild, 2020-04-05:
By the way, I'm using the environment env.mega2560 to compile, but I don't think that could have anything to do w/ my issue could it?

ellensp, 2020-04-05:
If did a git clone it tells the you the version lower left hand corner. If you downloaded a zip, the version number in configuration.h is a hint, but no definitive way.  default_envs = mega2560 is fine and what I used.

ellensp, 2020-04-05:
hit the clean icon and restart vscode (its easily confused) and try again.

inventabuild, 2020-04-05:
I hit clean icon which Succeeded, then hit Build and got the same errors.  I closed VSC, reopened it and reopened the Marlin firmware I was working on.  Tried to Build it and got the same errors.

inventabuild, 2020-04-05:
Back to my version question, I clicked download and Marlin asked me if I want to open it in Desktop and I clicked yes.  I still don't see the version #

inventabuild, 2020-04-05:
In Github desktop I clicked on the Current branch name (the name I gave it) and  dropdown says Default branch 2.0x from 4 days ago so I guess it's the latest 2.0.

inventabuild, 2020-04-05:
I'm not sure why I get an error and you don't.  I'm stumped.

inventabuild, 2020-04-05:
Scrolling through the errors in the terminal it gives a little more info:

In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\AVR\Servo.cpp:56:
Marlin\src\HAL\AVR\../../inc/../core/types.h:76:25: error: "(" may not appear in macro parameter list
 #define feedRate_t(float(MM_M) / 60.0f)
                         ^
Marlin\src\HAL\AVR\../../inc/../core/types.h:77:9: error: macro names must be identifiers
 #define (float(MM_S) * 60.0f)
         ^
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\AVR\HAL_SPI.cpp:34:
Marlin\src\HAL\AVR\../../inc/../core/types.h:76:25: error: "(" may not appear in macro parameter list
 #define feedRate_t(float(MM_M) / 60.0f)
                         ^
Marlin\src\HAL\AVR\../../inc/../core/types.h:77:9: error: macro names must be identifiers
 #define (float(MM_S) * 60.0f)
         ^
Compiling .pio\build\mega2560\src\src\HAL\shared\backtrace\unwarm.cpp.o
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\AVR\MarlinSerial.cpp:39:
Marlin\src\HAL\AVR\../../inc/../core/types.h:76:25: error: "(" may not appear in macro parameter list
 #define feedRate_t(float(MM_M) / 60.0f)
                         ^
Marlin\src\HAL\AVR\../../inc/../core/types.h:77:9: error: macro names must be identifiers
 #define (float(MM_S) * 60.0f)
         ^
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\AVR\watchdog.cpp:25:
Marlin\src\HAL\AVR\../../inc/../core/types.h:76:25: error: "(" may not appear in macro parameter list
 #define feedRate_t(float(MM_M) / 60.0f)
                         ^
Marlin\src\HAL\AVR\../../inc/../core/types.h:77:9: error: macro names must be identifiers
 #define (float(MM_S) * 60.0f)
         ^
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\AVR\eeprom.cpp:24:
Marlin\src\HAL\AVR\../../inc/../core/types.h:76:25: error: "(" may not appear in macro parameter list
 #define feedRate_t(float(MM_M) / 60.0f)
                         ^
Marlin\src\HAL\AVR\../../inc/../core/types.h:77:9: error: macro names must be identifiers
 #define (float(MM_S) * 60.0f)
         ^
*** [.pio\build\mega2560\src\src\HAL\AVR\Servo.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\MarlinSerial.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\HAL_SPI.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\watchdog.cpp.o] Error 1
In file included from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\shared\HAL_spi_L6470.cpp:28:
Marlin\src\HAL\shared\../../inc/../core/types.h:76:25: error: "(" may not appear in macro parameter list
 #define feedRate_t(float(MM_M) / 60.0f)
                         ^
Marlin\src\HAL\shared\../../inc/../core/types.h:77:9: error: macro names must be identifiers
 #define (float(MM_S) * 60.0f)
         ^
*** [.pio\build\mega2560\src\src\HAL\AVR\eeprom.cpp.o] Error 1
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\AVR\HAL.cpp:25:
Marlin\src\HAL\AVR\../../inc/../core/types.h:76:25: error: "(" may not appear in macro parameter list
 #define feedRate_t(float(MM_M) / 60.0f)
                         ^
Marlin\src\HAL\AVR\../../inc/../core/types.h:77:9: error: macro names must be identifiers
 #define (float(MM_S) * 60.0f)
         ^
*** [.pio\build\mega2560\src\src\HAL\shared\HAL_spi_L6470.cpp.o] Error 1
*** [.pio\build\mega2560\src\src\HAL\AVR\HAL.cpp.o] Error 1

GMagician, 2020-04-05:
> #define feedRate_t(float(MM_M) / 60.0f)

Is not a valid C line.

I can see  in types.h in line 76:
`#define MMM_TO_MMS(MM_M) feedRate_t(float(MM_M) / 60.0f)`

inventabuild, 2020-04-05:
Thanks, please let me know what should I do so my code will compile.

inventabuild, 2020-04-05:
Ah, I see what you mean.  I fixed that line and now only the next line errors out.  Please let me know what the next line s/b.

inventabuild, 2020-04-05:
Believe it or not Google found the correct code in a Prusa Mini webpage: https://prusa3d.github.io/Prusa-Firmware-Buddy-Doc/d9/d49/types_8h.html#a6fd191ff76c5e211e39560ff3b17a838

#define MMM_TO_MMS( MM_M ) feedRate_t(float(MM_M) / 60.0f)
#define MMS_TO_MMM	(MM_S) (float(MM_S) * 60.0f)

When I add both those lines -> no more errors there.  Why is my fresh download of Marlin missing those lines of code?

GMagician, 2020-04-05:
I think you have a corrupted Marlin copy on your PC. Line I mentioned is written in Marlin code as well

GMagician, 2020-04-05:
Save your config file, erase Marlin folder (and subfolders) on your pc and re-download them from Marlin repository

inventabuild, 2020-04-05:
I deleted the Marlin folder inside the Github Desktop folder, now I'm trying to redownload Marlin and open it in Github Desktop and I'm getting the error message Can't find "Configurations"'.

Github Desktop was working fine before I deleted the Marlin folder inside the Github folder.  Please advise.

inventabuild, 2020-04-05:
Everything is really hosed up now.  Are you sure I was supposed to delete everything Marlin folder or just certain Marlin folders?

GMagician, 2020-04-05:
If you start from "cloning" marlin repository or just downloading the zip from repository and then copying your config files in Marlin/Marlin (overwriting already present files) you should be able to compile everything

inventabuild, 2020-04-05:
Hmmm, I just uninstalled and reinstalled Github Desktop and redownloaded Marlin and now everything seems to be working ok.

I'm trying to install bugfix now and this is all it contains.  Where's the rest of the files?  How do I keep my personally modified config files and add all the fixes to the Marlin firmware?
[image]

GMagician, 2020-04-05:
Since this is not a Marlin bug I think you have to close this and move to Discord or similar to get this kind of help.
If your are not skilled with github, simple thing is to download zip from bugfix branch and overwrite cnf with yours, and repeat everytime you need fresh code.
Otherwise from github, create a new branhc starting from bugfix, overwrite with your cnf and commit "locally". To update just rebase your branch with bugfix

thinkyhead, 2020-04-05:
Careful when downloading new Marlin not to mix together files from older Marlin. File names may change even between minor versions.

closed 2020-04-05 by thinkyhead