MarlinFirmware/Marlin #17414: [BUG] BTT_SKR_PRO_V1.1 BLTOUCH do not deploy (when homing)

kolbenhans, 2020-04-05:
### Bug Description

BLTOUCH not deploy when homing Z. Everything work when testing by GCode, but homing Z does not deploy BLTouch.
>>> SENDING:M43 S
Servo probe test
. using index:  0, deploy angle: 10, stow angle:   90
. Probe Z_MIN_PROBE_PIN: 37
. Z_MIN_PROBE_ENDSTOP_INVERTING: false
. Check for BLTOUCH
= BLTouch Classic 1.2, 1.3, Smart 1.0, 2.0, 2.2, 3.0, 3.1 detected.
** Please trigger probe within 30 sec **
. Pulse width (+/- 4ms): 12
= BLTouch pre V3.1 (or compatible) detected.	'(i triggered the bltouch here) 

'======= OK =======

>>> SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: open
z_probe: TRIGGERED           

'======= OK =======

>>> M280 P0 S10
SENDING:M280 P0 S10       	' ==== BLTOUCH deployed successful =====
>>> M119
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: open
z_probe: open                      

'======= OK =======

but when Homing, BLTouch do not get deployed.

- Config attached.
- BOARD BTT_SKR_PRO_V1.1
- BLTOUCH connected to PA1 and PA2 (- PROBE pins - not Z_MIN_ENDSTOP_PIN)
- Changes done:
	//#define USE_ZMIN_PLUG (both tested)
	#define Z_MIN_ENDSTOP_INVERTING true
	//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN (PROBE PIN IS DEFINED IN PINS FILE)
	#define Z_PROBE_SERVO_NR 0 (both tested)
	#define Z_SERVO_ANGLES { 90, 0 } (both tested)
	#define BLTOUCH
	#define AUTO_BED_LEVELING_BILINEAR
	#define Z_SAFE_HOMING
	#define NUM_SERVOS 1 (both tested)
	#define SERVO_DELAY { 300 } (multiple values from 200 to 700 tested)

### My Configurations

attached. [Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/4434204/Configurations.zip)

### Steps to Reproduce

1. Testing M502, M500, M43 S, M119, M280 P0 S10 & M280 P0 S90
2. Homing all Axis

**Expected behavior:** [What you expect to happen]
BLTOUCH deploy when homing Z Axis

**Actual behavior:** [What actually happens]
BLTOUCH do not deploy when homing Z Axis

#### Additional Information

kolbenhans, 2020-04-07:
did a test with DEBUG_LEVELING_FEATURE enabled as well: result:
>>> M111 S247
SENDING:M111 S247
echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
echo:M105
echo:M105
>>> G28
SENDING:G28
echo:G28
>>> G28
Machine Type: Cartesian
Probe: BLTOUCH
Probe Offset X36.00 Y3.00 Z-0.92 (Right-Back & Below Nozzle)
Auto Bed Leveling: BILINEAR (disabled)
Raise Z (before homing) to 5.00
>>> do_blocking_move_to X0.00 Y0.00 Z5.00
<<< do_blocking_move_to
>>> homeaxis(X)
Home 1 Fast:
>>> do_homing_move(X, -450.00, [50.00])
echo:busy: processing
echo:busy: processing
<<< do_homing_move(X)
Move Away:
>>> do_homing_move(X, 5.00, [50.00])
<<< do_homing_move(X)
Home 2 Slow:
>>> do_homing_move(X, -10.00, 25.00)
<<< do_homing_move(X)
>>> set_axis_is_at_home(X)
Axis 88 home_offset = 0.00 position_shift = 0.00
> home_offset[X] = 0.00
  current_position= X0.00 Y0.00 Z5.00 :
<<< set_axis_is_at_home(X)
  current_position= X0.00 Y0.00 Z5.00 : sync_plan_position
  current_position= X0.00 Y0.00 Z5.00 : > AFTER set_axis_is_at_home
<<< homeaxis(X)
>>> homeaxis(Y)
Home 1 Fast:
>>> do_homing_move(Y, -450.00, [50.00])
echo:busy: processing
echo:busy: processing
<<< do_homing_move(Y)
Move Away:
>>> do_homing_move(Y, 5.00, [50.00])
<<< do_homing_move(Y)
Home 2 Slow:
>>> do_homing_move(Y, -10.00, 25.00)
<<< do_homing_move(Y)
>>> set_axis_is_at_home(Y)
Axis 89 home_offset = 0.00 position_shift = 0.00
> home_offset[Y] = 0.00
  current_position= X0.00 Y0.00 Z5.00 :
<<< set_axis_is_at_home(Y)
  current_position= X0.00 Y0.00 Z5.00 : sync_plan_position
  current_position= X0.00 Y0.00 Z5.00 : > AFTER set_axis_is_at_home
<<< homeaxis(Y)
BLTouch Command :160
BLTouch Command :90
last_written_mode - 0
config mode - OD
home_z_safely >>>
  current_position= X0.00 Y0.00 Z5.00 : sync_plan_position
  destination= X150.00 Y150.00 Z5.00 : home_z_safely
>>> do_blocking_move_to X150.00 Y150.00 Z5.00
<<< do_blocking_move_to
echo:busy: processing
>>> homeaxis(Z)
Home 1 Fast:
>>> do_homing_move(Z, -450.00, [4.00])
echo:busy: processing
[ERROR] Can't read from printer (disconnected?) (SerialException): call to ClearCommError failed
Disconnected. 

(BLTOUCH does not deploy so i turned off the printer)
as seen in the log:
BLTouch Command :160
BLTouch Command :90

The BLTOUCH command after reset is 90 (stow) but this should be 10 (deploy) right?
changing:
#define Z_SERVO_ANGLES { 10, 90 } // Z Servo Deploy and Stow angles
don't have any effekt either.

WDuck, 2020-04-09:
I have the same issue and can confirm your findings.
Also this seems to be a wider spread issue: #17391 #17440
Hopefully someone who knows about the implementation can fix this.
On my BTT SKR PRO V1.1 not even with latest bugfix, nor development or 2.0.1 (where it worked with the old RAMPS 1.4) I get it to work.
Really strange.

hills8, 2020-04-10:
this solved the problem for me:
https://github.com/MarlinFirmware/Marlin/issues/17440#issuecomment-612171479

kolbenhans, 2020-04-10:
Works fine for me too, thx!

closed 2020-04-10 by kolbenhans