MarlinFirmware/Marlin #17416: [BUG] 2.0.5.3 - Layer shifting after bed leveling

Galaxi93, 2020-04-05:
### Bug Description

After bed leveling, in the middle of printing, the printhead randomly shift on x, y or both axes like when a stepper motor lose steps. Oviously I tried with the same gcode whithout the G29 command and the print gone correctly.
I tried both versions 2.0.5.3 and bugfix and the problem remains.

### My Configurations

Mine is a diy dual extruder 3d printer  with a double X carriage controlled by a BiTreeTech SKR PRO v1.1. 
I'm using TMC2208 for X and Y axes in standalone mode, TMC5130 for Z axis and DRV8825 for E axis. 
For probing I installed a 3dtouch clone.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4434446/Configuration.zip)

### Steps to Reproduce

1. Simply start a print with G29 command and wait for disaster...

#### Additional Information

Here some pictures of failed prints:
[image]
[image]
[image]
[image]

Lord-Quake, 2020-04-06:
I noticed you declared stepper Y with TMC2209 instead of TMC2208

```
#define X_DRIVER_TYPE  TMC2208_STANDALONE
#define Y_DRIVER_TYPE  TMC2209_STANDALONE  <----------
#define Z_DRIVER_TYPE  TMC5160
#define X2_DRIVER_TYPE TMC2208_STANDALONE

```

Galaxi93, 2020-04-06:
I tried that driver and I forgot to change in the configuration file. Anyway I changed to the correct value and the problem remains

Lord-Quake, 2020-04-06:
I'm curious.... Could you post your gcode file. I would give it a try since I'm also using TMC2208_STANDALONE drivers.

Galaxi93, 2020-04-06:
Here the gcode: 
[SimplBox4 x6 g29.zip](https://github.com/MarlinFirmware/Marlin/files/4439785/SimplBox4.x6.g29.zip)

It's for 6 separate prints in one file, printed in vase mode (pay attention to your print head dimensions, because it prints one complete object and than moves away and goes down to print another). Remember also the size of bed (mine is 300x300mm).

Now I'm trying new configuration files with default values (I set only mandatory things like steps and axis directions). Later I will post the results.

Lord-Quake, 2020-04-06:
OK, unfortunately I can't print file. Can you post a one object version with the same splicer settings that replicates your problem?

Galaxi93, 2020-04-06:
[Surface_Calibration_Test.zip](https://github.com/MarlinFirmware/Marlin/files/4439898/Surface_Calibration_Test.zip)
This is a small object printed near the origin. With this I had a small shift near the end of print.

Lord-Quake, 2020-04-06:
Right now I cannot replicate your problem.

This is a off topic suggestion:
I would recommend you turn on your heated bed before you do auto bed leveling.

Galaxi93, 2020-04-06:
@Lord-Quake Thank You for your helping!

Anyway I tested the "blank" configuration and It works! Now I'm enabling step by step all settings I'd before in order to find out what causes the problem.

ladefoged94, 2020-04-06:
If you figure what started this issie please update in here:)

I got same issue when i use 2.0.5 :)

Galaxi93, 2020-04-07:
I found the problem!!!
The issue disappears when I comment these rows:
```
//#define X_DRIVER_TYPE  TMC2208_STANDALONE   <--- commented
//#define Y_DRIVER_TYPE  TMC2208_STANDALONE   <--- commented 
#define Z_DRIVER_TYPE  TMC5160
//#define X2_DRIVER_TYPE TMC2208_STANDALONE  <--- commented
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
//#define Z4_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE DRV8825
#define E1_DRIVER_TYPE DRV8825
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988
//#define E6_DRIVER_TYPE A4988
//#define E7_DRIVER_TYPE A4988
```

I don't know why these settings cause the problem, maybe there is a incompatibility between a TMC5160 driver and a standalone one.
For now I made too much tests, maybe tomorrow I will do more test to figure out what's the real problem. Maybe someone else can try to replicate the problem by setting both standalone driver and direct TMC driver.

@ladefoged94  Can your share your config files?

Galaxi93, 2020-04-07:
I commented this row
`#define ENDSTOPS_ALWAYS_ON_DEFAULT`
and the issue disappeared. So I can deduce it was a electrical issue. But why happens only with bed leveling? Well, my printer has a "big" nema 23 motor at Z axis and normaly is almost turned off, but during bed leveling it spins continuesly during printing causing electrical interference...

@ladefoged94 Maybe you have the same problem. I will wait a while before close this issue.

ladefoged94, 2020-04-07:
As wierd as this sound i resolved parts of my issue earlier
What i did was:

1.take my 2.0 bugfix upload to printer and init eprom with that 

2. Take new firmware upload eprom woth that

Now it works, but as soon as i init eprom it goes back to layershift and several other issues

I need to dig into this tomorrow to see if i missed some sort of configs or whats going on.

Hope this might can help you somehow too

jjwagner, 2020-04-07:
Is it possible one or more motors are overheating? I have had a persistent occasional problem on a BTT SKR Mini E3 (2209 drivers) where the X motor would become scalding hot after a few minutes. M906 showed the currents as programmed, and reducing the X motor current made it start losing steps (but not get any cooler!).

I first saw this with 2.0.4.4. I downloaded 2.0.x bugfix last evening, updated my configuration files and recompiled. 

Today I have not had X motor overheating, but I have a mysterious issue where the X axis scale seems to change after several minutes of printing, causing the X carriage to crash into the MAX side of its travel. M92 shows the correct steps/mm. I can fix it by pausing and doing an M502 to reset all the settings.

ladefoged94, 2020-04-07:
My motors is not really hot, my drivers could be tho, i will check tomorrow.

I use same board as you, with sensorless homeing and ABL

Its my only printer having those issues and my temporary fix makes me think its my own faulth 

Its just confusing since i went over both config and config adv many times to compare vaules etc and all seems equal

AFprinter, 2020-04-09:
How much current are your steppers rated for, you've got your y axis sitting at 2 amps.  You probably want to drop both your x and y axis to 1000 ish and start there.  It looks like your likely overheating your drivers.

closed 2020-04-10 by Galaxi93