MarlinFirmware/Marlin #17417: Bed Probing Sanity check tied to Z endstop height

Tailslide, 2020-04-05:
### Bug Description

Sanity check on bed probing is based on Z endstop height and can't be resolved by changing g92 zero height or settting zero in G55 etc workspace coordinates. 

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4434663/Configuration.zip)

### Steps to Reproduce

1. G28
2. Move to area with higher Z than Z endstop and reset coordinates with G92 X0 Y0 Z0
3. Probe for example with G29 L0 R100 F100 B0

**Expected behavior:** [What you expect to happen]

Probe successful

**Actual behavior:** [What actually happens]

Probing Failed

#### Additional Information

This is useful for CNC work.  For example my machine knows the height of the bed from the Z endstop but doesn't know the height of the stock I am cutting until it is probed.  Typically you set the Z height of the stock with G92 Z0. I was able to sucessfully probe by commenting out this line in probe.cpp:

//   || (sanity_check && current_position.z > -offset.z + _MAX(Z_CLEARANCE_MULTI_PROBE, 4) / 2)  // Probe triggered too high?
  
Here is the debug output it seems to all be based on absolute coordinates:

G29 L-300 R0 F300 B0
  current_position= X1199.13 Y391.52 Z16.13 : >>> G29
Machine Type: Cartesian
Probe: FIX_MOUNTED_PROBE
Probe Offset X0.00 Y0.00 Z0.00 (Aligned With & Same Z as Nozzle)
Auto Bed Leveling: BILINEAR (disabled)
  current_position= X1199.13 Y391.52 Z16.13 : Probe::set_deployed
deploy: 1
Probe::move_z(20.00)
>>> do_blocking_move_to X1199.13 Y391.52 Z20.00
<<< do_blocking_move_to
>>> do_blocking_move_to X1199.13 Y391.52 Z20.00
<<< do_blocking_move_to
>>> Probe::probe_at_point(-300.00, 300.00, raise, 0, probe_relative)
  current_position= X1199.13 Y391.52 Z20.00 :
>>> do_blocking_move_to X899.13 Y691.52 Z20.00
<<< do_blocking_move_to
echo:busy: processing
echo:busy: processing
  current_position= X899.13 Y691.52 Z20.00 : Probe::set_deployed
deploy: 1
  current_position= X899.13 Y691.52 Z20.00 : >>> Probe::run_z_probe
  current_position= X899.13 Y691.52 Z20.00 : >>> Probe::probe_down_to_z
>>> do_blocking_move_to X899.13 Y691.52 Z-2.00
<<< do_blocking_move_to
echo:busy: processing
echo:busy: processing		
  current_position= X899.13 Y691.52 Z14.94 : sync_plan_position
  current_position= X899.13 Y691.52 Z14.94 : <<< Probe::probe_down_to_z
SLOW Probe fail!
  current_position= X899.13 Y691.52 Z14.94 : <<< run_z_probe
  current_position= X899.13 Y691.52 Z14.94 : Probe::set_deployed
deploy: 0
>>> do_blocking_move_to X899.13 Y691.52 Z14.94
<<< do_blocking_move_to
Error:Probing Failed
<<< Probe::probe_at_point
  current_position= X899.13 Y691.52 Z14.94 : Probe::set_deployed
deploy: 0
  current_position= X899.13 Y691.52 Z14.94 : > probing complete
>>> do_blocking_move_to X899.13 Y691.52 Z20.00
<<< do_blocking_move_to
echo:busy: processing
X:-300.00 Y:300.00 Z:4.87 E:0.00 Count X:89913 Y:69152 Z:8000
<<< G29
ok

neilvangeffen, 2020-04-30:
same here. With the line commented out, does it still work with ABL?

I wonder what the thought process was for limiting the max height of the bed to be 2mm before stopping the probing and calling it a fail.
I got it to probe fine by sending a 0 as the sanity_check, but still wont work for ABL

Tailslide, 2020-04-30:
Yes ABL worked for me after commenting out that line. I find it pretty useful when I need to make accurate depth cuts on wood.

boelle, 2020-06-20:
@Tailslide is this still an issue?

Tailslide, 2020-06-20:
> @Tailslide is this still an issue?

As far as I know it’s still an issue.. still using it with that code commented out.

boelle, 2020-06-23:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

Tailslide, 2020-06-24:
Was there something changed to fix it?  It's not the easiest thing in the world to test and no one appeared to look at this at all.

thinkyhead, 2020-07-25:
Yes, it has been changed. If there are any issues with the new code let us know.

closed 2020-08-31 by github-actions[bot]