MarlinFirmware/Marlin #17420: [BUG] too accurate hotend heating due ADVANCED_PAUSE_FEATURE

MrStump, 2020-04-05:
### Bug Description

marlin 2.0.5.3
few diffirent printer - same problem
#define ADVANCED_PAUSE_FEATURE

when filament changed, and i press button to continue, even when hotend is heated (for ex.)240/240 it could "heating" it 3-5 minutes. looks like it measure too accurate, like 0.01 degree, i just see on display like few minutes printer try to reach (for ex.)240 and sometime temp shown 239-241, but mostly time 240 and it still "heating"

### Steps to Reproduce

1. when print choose "change filament"
2. wait for printer will ready for new filament "beeps"
3. press button

**Expected behavior:** if temp is (for ex.)240/240 it start to purge and print

**Actual behavior:** waiting few minutes and see "heating" when it already heated

#### Additional Information

#define DEFAULT_Kp 20.03
#define DEFAULT_Ki 1.73
#define DEFAULT_Kd 58.09

Thanks.

thinkyhead, 2020-04-05:
What do you see if you set your `TEMP_RESIDENCY_TIME` value to 2 or 3?

MrStump, 2020-04-06:
thanks! 
set `TEMP_RESIDENCY_TIME 2` and printing continues almost immediately! i did't know about this setting. 
when it was 10 i really waiting some time 3-5 minutes for continue printing, but on temp LOG there was (for ex) sometime 239.5-241.4, but no 238 or 242, so temp was in the indicated range `TEMP_WINDOW 1`
anyway Great Thanks again!

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle

MrStump, 2020-06-21:
that was not a bug. sorry, that was my mistake, i did't know about TEMP_RESIDENCY_TIME settings. when i set it to 2 all works just great! thanks again and sorry.

MrStump, 2020-06-21:
> Lack of Activity
> This issue is being closed due to lack of activity. If you have solved the
> issue, please let us know how you solved it. If you haven't, please tell us
> what else you've tried in the meantime, and possibly this issue will be
> reopened.

that was not a bug. sorry, that was my mistake, i did't know about TEMP_RESIDENCY_TIME settings. when i set it to 2 all works just great! thanks again and sorry.