MarlinFirmware/Marlin #17423: [2.0.5.3] TMC5160 and TMC2130 Arduino DUE Y stepper disables randomly

MrStump, 2020-04-06:
AAAAND similar to #12944 issue...
CORE_XY, arduino DUE+ramps `BOARD_RAMPS4DUE_EFB` with TMC5160 in SPI mode on XY
all was fine more than year, and few weeks ago (in FEB) when i change XY MICROSTEPS for some reason and reflash DUE - Y motor just starts randomly stopping
firstly there was marlin release 2.0.(something) and i now tried 2.0.5.3
i change wires to new one, change DUE to the new one, even change drivers to TMC2130 SPI - all with no luck, Y-motor just stops in 30-120 min
no temp warning, it's just "disabled"
M122 show me that 
`ENABLED X:true Y:false`
`StealthChop X:true Y:false`
and, the wierdest, it reset default 16 to `msteps X:16 Y:256` 
i tried in firmware `Y_MICROSTEPS` from 4 to 256 but with no change in situation, it just stop in some time.
i have no idea what could happen
here is LOG of M122 before and after issue
[image]

hexxter, 2020-04-27:
i think i have the same problem.

boelle, 2020-06-20:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

MrStump, 2020-06-21:
> Configurations, please
> Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
> requested in the Issue template) and drop them into your next reply.
> We'll check them over and see if anything is amiss.

really sorry for this delayed respond.
the problem was in my RAMPS board, one of the Y driver pins on the pad loosened and randomly lost contact with the driver. 
that was hardware problem, not firmware issue.
Sorry.

closed 2020-06-21 by MrStump