MarlinFirmware/Marlin #17424: [BUG] Safe_Homing not in center of bed

Grogyan, 2020-04-06:
### Bug Description

Safe_Homing is not going to the center of the bed.
This affects UBL and other bed leveling, including Bi-linear and Mesh, as it offsets significantly the mesh which the probing is applied on

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4437329/Configuration.zip)
[Safe_home_Mesh.txt](https://github.com/MarlinFirmware/Marlin/files/4437343/Safe_home_Mesh.txt)

### Steps to Reproduce

1. Ensure to set Probe type (BLTouch) in your config
2. Ensure to set Auto Bed Leveling type (UBL, Bilinear or Mesh)
3. Ensure Bed Leveling debug is enabled
4. After uploading the firmware open your Terminal of choice
5. Type M111 S247 to 
6. Type G28
7. Note these two lines
Recv:   current_position= X0.00 Y0.00 Z5.00 : sync_plan_position
Recv:   destination= X47.00 Y90.00 Z9.25 : home_z_safely

**Expected behavior:** [What you expect to happen]
Safe_Homing to position the probe in the center of the specified bed_size
210 / 2 is 105 on the X
200 / 2 is 100 on the Y

**Actual behavior:** [What actually happens]
Safe_Homing positions the probe on X Axis at (NOZZLE_TO_PROBE - 10) // NOZZLE_TO_PROBE X on my machine is 58, No idea why it is 47
Safe_Homing positions the probe on Y Axis at ((Y_Bed_Size / 2) -10), on my machine this is supposed to be 100, not 90

#### Additional Information

Issue appears to be in the G28.cpp file, around line 130
2 parameters are parsed, but only one is processed

Have tried 3 separate release versions, and the latest marlin2,0_bugfix

GMagician, 2020-04-06:
Let's see...

you have:
```c++
#define NOZZLE_TO_PROBE_OFFSET { 58, 10, -3.65 }
#define X_BED_SIZE 210
#define Y_BED_SIZE 200
#define Z_SAFE_HOMING_X_POINT ((X_BED_SIZE) / 2)
#define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2)
```

then:
```c++
Z_SAFE_HOMING_X_POINT = 105
Z_SAFE_HOMING_Y_POINT = 100
```

> Safe_Homing to position the ***probe in the center*** of the specified bed_size

X=105 and Y=100...the nozzle is in center but applying probe offset
X=105-58 = 47
Y=100-10 = 90

GMagician, 2020-04-06:
> the probe offset should be halved

That sound not correct. If nozzle is centered at X105 Y100 to center probe you have to move whole offset, not half of it

GMagician, 2020-04-06:
Are you sure that when you move to X0 Y0 and then to X210 Y200 your nozzle is in the same position of "opposite" corner?

Grogyan, 2020-04-06:
Yes, full movement on the X and Y axis have been verified

Manually jogging the printer after G28, proved that the half distance of the nozzle_to_probe, offset from where it should be for the probe

GMagician, 2020-04-07:
That's mathematic & physic. If nozzle is centered ad x105 y100 and probe has offset 58 and 10 you have to move nozzle away of such offset to center it.
Then:
1) your esteps are wrong
2) your probe offset is wrong
3) You forgot to execute M502 + M500 and are still using some old configuration

Grogyan, 2020-04-09:
Esteps are correct,
Probe offset is correct
And do a M502 and M500 frequently

What I might play with is M428

closed 2020-04-09 by Grogyan