MarlinFirmware/Marlin #17425: Home doesnt work first time after boot

Musicdad12, 2020-04-06:
### Bug Description

After turning on my printer and pressing the home from Octoprint and from display, the printer moves the z axis up and stops. When I press Home the second time, the printer homes as expected.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. turn on printer
2. press home
3. press home again
**Expected behavior:** all axis home

**Actual behavior:** z moves up 10mm and stops

#### Additional Information

I am running a SKR 1.4 turbo with TMC2209 sensorless homing. Raspi 3B+

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4438311/Marlin.zip)

boelle, 2020-06-20:
@Musicdad12 is issue gone?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle