MarlinFirmware/Marlin #17427: [BUG] "Marlin USB Device" disappears after a few seconds

JRHeaton, 2020-04-06:
#17416 # Bug Description

I am using an SKR 1.3, and have compiled Marlin 2.x (latest) from both the official repo and the BTT repo. Flashing to the board via the SD card has seemed to work, but ever since I first got the board, from smoothieboard to now Marlin, I cannot get it to stay mounted over USB so that I can attach a serial shell. (or so that any slicers can connect). After a few seconds, the serial port disappears by itself, and I can expedite this problem by trying to connect to the serial port manually first. It just hangs, then disappears. Is the board faulty? Or is this some issue on my part? FWIW I have the board set up for USB power, and have it disconnected from anything else. I've also tried on BOTH Windows (with LPC drivers etc) and Mac (no drivers required). Same issue. 

### My Configurations

First try, BTT config files here: https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/tree/master/BTT%20SKR%20V1.3/firmware/Marlin-bugfix-2.0.x/Marlin

Second try, default Marlin files, with `MOTHERBOARD` changed from here: https://github.com/MarlinFirmware/Configurations/tree/release-2.0.5/config/default

### Steps to Reproduce

1. Receive BTT SKR 1.3 in mail.
2. Notice that I cannot get a serial shell opened over USB, it keeps disappearing.
3. Investigate on both Mac and PC by reinstalling drivers, re-compiling and re-flashing Marlin 2.x firmware.
4. Still same issue, not sure at this point if I have a faulty board(which would seem weird as it shows up for a moment), or if it's a problem with me.

**Expected behavior:** A stable port to which I can connect via Cura/Pronterface/etc.

**Actual behavior:** Port/USB device mounts/appears for a few seconds, then disappears.

#### Additional Information

* Available on request!

thisiskeithb, 2020-04-06:
Do you have a screen hooked up? Without thermistors hooked up, it's probably halting due to thermal runaway protection, but won't be visible unless you have a screen.

hectori4502, 2020-04-06:
The same happens to me with my skr v1.1
windows 10 disconnects the port (in a marlin version of Nov 29, 2019 works fine without problem)

Bob-the-Kuhn, 2020-04-06:
Isn't windows "fun"?

The SKR V1.1 & 1.3 are LPC1768 based boards so the INF file in Marlin should be enough to get it working properly.  The file is at **Marlin\src\HAL\LPC1768\win_usb_driver\lpc176x_usb_driver.inf**.

If that doesn't help then try one of the compatible drivers.  On my Win10 PRO it they are at:
1. Device manager
2. COM & LPT ports
3. right click on port
4. select update driver
5. Browse for drivers on your computer
6. Let me pick from a list of available drivers

JRHeaton, 2020-04-07:
> Isn't windows "fun"?
>
> The SKR V1.1 & 1.3 are LPC1768 based boards so the INF file in Marlin should be enough to get it working properly. The file is at [...]

I appreciate you trying to help, however as mentioned in the original post, this has been ruled out as an issue of drivers as I have installed the drivers on windows and also tried the connection on a Mac which does not require drivers.

JRHeaton, 2020-04-07:
> Do you have a screen hooked up? Without thermistors hooked up, it's probably halting due to thermal runaway protection, but won't be visible unless you have a screen.

So I got back to my computer, commented out `CR10_STOCKDISPLAY` in `Configuration.h`, recompiled, plugged it into EXP3 on my Ender 5, powered it up over USB and tada! 🎉 just as you guessed, I am getting this "MAX TEMP E1" error. 

That being said, this is a single-extruder machine, and I have `#define EXTRUDERS 1` -- I would assume, as a programmer, and by the board silkscreen labels, that this would be referring to E1, or the _second_ extruder. Short of connecting up the whole machine to check, is there something else you think I've configured incorrectly for the build that would make the firmware think it's trying to heat E1 instead of E0? Thanks!

JRHeaton, 2020-04-07:
> Do you have a screen hooked up? Without thermistors hooked up, it's probably halting due to thermal runaway protection, but won't be visible unless you have a screen.

@thisiskeithb So I got back to my computer, commented out `CR10_STOCKDISPLAY` in `Configuration.h`, recompiled, plugged it into EXP3 on my Ender 5, powered it up over USB and tada! 🎉 just as you guessed, I am getting this "MAX TEMP E1" error. 

That being said, this is a single-extruder machine, and I have `#define EXTRUDERS 1` -- I would assume, as a programmer, and by the board silkscreen labels, that this would be referring to E1, or the _second_ extruder. Short of connecting up the whole machine to check, is there something else you think I've configured incorrectly for the build that would make the firmware think it's trying to heat E1 instead of E0? Thanks!

thisiskeithb, 2020-04-07:
Can you hook up a couple of thermistors? If you’re still getting a max temp error after that, your board is likely damaged.

JRHeaton, 2020-04-07:
@thisiskeithb I connected both the hotend and bed thermistors, and am getting E1 still. Is there _anything_ left to check, config-wise? [Res,therm]istor values, etc.? Or any other cross-diagnostic procedure to verify that the board is in fact dead before I spend the money on a new one? Really appreciate your help thus far.

JRHeaton, 2020-04-07:
Just thought of this, but perhaps it's worth hooking up to a hardware UART if present on the board? Would Marlin potentially print more info about the exact failure over that before halting?

ellensp, 2020-04-07:
The firmware incorrectly reports E1, it is really E0. 
Max temp is when there is least resistance on NTC thermistors. So your hot end thermistor legs are probably shorted.  Unfortunately when this happens on a skr1.3 it often takes out the analog port it was connected to...

First thing is unplug that thermistor from the controller. and power up Does the error on the LCD change? It should change to min temp error.  
If it does you may be in luck as the pin is still changing. If it doesn't. Its a dead analog input..

While thermistor is unplugged measure its resistance through its wires. Most are 100k at around 25C

JRHeaton, 2020-04-07:
@ellensp Thanks so much for your input. The error was first happening with nothing connected to the board, and plugging in bed & extruder thermistors doesn't seem to fix the error. So I suspect given what you're saying that one of the inputs is faulty. I guess the last thing I could try, since I only have the 2 thermistors, is to solder 3 100k resistors across the thermistor connector pins and see if the board behaves differently...?

closed 2020-04-08 by thinkyhead

hectori4502, 2020-04-14:
skr v1.1 Marlin bugfix 13April2020 works OK now.