MarlinFirmware/Marlin #17430: [BUG] compiling error in M425.cpp (2.0.5.3)

kaibs, 2020-04-07:
Due to the current situation I'm digging deeper into Marlin and it's more exotic features. Yesterday I wanted to test out [Backlash Compensation](https://marlinfw.org/docs/gcode/M425.html), but I've got the following compiling error: 

`Marlin\src\gcode\calibrate\M425.cpp:50:24: error: macro "CAN_CALIBRATE" requires 2 arguments, but only 1 given
...`

Following the same for the other uses of the `CAN_CALIBRATE()` ([full error message on pastebin](https://pastebin.com/6mmWRbgy)).

It seems like the function `CAN_CALIBRATE()` was added with [#17281](https://github.com/MarlinFirmware/Marlin/pull/17281) without replacing anything, so I just removed it again, resulting in a flawlessly compiling Marlin. Now Backlash Compensation seems to be working fine for me.

Here are my current configs and the modified M425.cpp:
[bug_report.zip](https://github.com/MarlinFirmware/Marlin/files/4442804/bug_report.zip)

also the modified (and compiling) Marlin in my repo:
https://github.com/kaibs/MarlinHevo/tree/2.0.5.3

Regarding my setup:

* VS Code (1.43.2) with platformio (4.3.1)
* HyperCube Evolution based COREXY-kinematics
* BTT SKR v1.4 with TMC2130 drivers

setar, 2020-04-17:
I confirm the compilation problem when the #define BACKLASH_COMPENSATION option is enabled.
And i see bug_report.zip by 	kibs.
The fix do repair compilation process, but disable check on AXIS_CAN_CALIBRATE : bad idea.
For example CoreXY kinematics must have only Z axis for BACKLASH_COMPENSATION.

thinkyhead, 2020-04-17:
This is now patched. Thanks for the notice!

commit 3a42b6c5 (2020-04-17): Patch M425 for CAN_CALIBRATE

Fixes #17430
closed 2020-04-17 by thinkyhead