MarlinFirmware/Marlin #17431: [BUG] G34 aborts on 'Decreasing accuracy' when total accuracy has improved

ogsv, 2020-04-07:
When running G34, first iteration shows a max difference between probe points of 0.16 mm, and calculates a stepper difference of 0.29 mm:
```
PROBEPT_49: 10.50
PROBEPT_50: 10.66
PROBEPT_51: 10.64
CALCULATED STEPPER POSITIONS: Z1=-0.01 Z2=0.27 Z3=0.28
DIFFERENCE Z1-Z2=0.28 Z2-Z3=0.02 Z3-Z1=0.29
```

Corrections are then applied correctly:
```
Z1 corrected by 0.00
Z2 corrected by 0.28
Z3 corrected by 0.29
```
On the second iteration, alignment is almost perfect, and with T=0.05, should be successful:
```
PROBEPT_49: 11.06
PROBEPT_50: 11.07
PROBEPT_51: 11.02
CALCULATED STEPPER POSITIONS: Z1=-0.04 Z2=-0.11 Z3=-0.06
DIFFERENCE Z1-Z2=0.08 Z2-Z3=0.06 Z3-Z1=0.02
```
However;
```
Decreasing accuracy detected.
G34 aborted.
```
I am using Z_STEPPER_ALIGN_KNOWN_STEPPER_POSITIONS, so abort on decreasing accuracy is expected. As can be seen, difference Z2-Z3 has increased from first to second iteration, and I guess this is what triggers the abort, but I don't think it's correct to trigger this when overall accuracy has improved.

My config:
https://github.com/ogsv/Marlin_SKRPRO/blob/2.0.x/Marlin/Configuration.h
https://github.com/ogsv/Marlin_SKRPRO/blob/2.0.x/Marlin/Configuration_adv.h

thinkyhead, 2020-04-08:
The implementation of `Z_STEPPER_ALIGN_KNOWN_STEPPER_POSITIONS` by @jufimu12 uses the cumulative mean error instead of checking each individual mean error. Perhaps this is more sensitive than it should be, or in fact is not the correct logical approach to determine decreasing accuracy.

I see that the check is, in-general, based on a percentage change, so for very small values like 0.2 a jump to 0.6 would look like a huge error.

It would be great if you and @jufimu12 could do some testing to see whether this is the case, what the correct logic should be, and whether these very tiny deviations matter to your print results and how much they could be ignored.

jufimu12, 2020-04-08:
@ogsv This is the bug I encountered and fixed. I think you are still having problems, because you are running Marlin version 2.0.4.3, but my fix was merged in version 2.0.5. If you update, it sould work. At least on my machine, it works fine now.

If you want more information on what exactly caused the bug:
[PR #17013 ](https://github.com/MarlinFirmware/Marlin/pull/17013#issue-381436521)
[The code which _now_ deals with analysing the error](https://github.com/MarlinFirmware/Marlin/blob/0518dec60d0931745efa2812fa388f33d68cfa29/Marlin/src/gcode/calibrate/G34_M422.cpp#L276)

> Perhaps this is more sensitive than it should be, or in fact is not the correct logical approach to determine decreasing accuracy.

I think this is less sensitive to false alarms, compared to the previous implementation. If one axis gets worse, but the others get better, it doesn't immediately abort.

> I see that the check is, in-general, based on a percentage change, so for very small values like 0.2 a jump to 0.6 would look like a huge error.

IIRC I even had that in mind for a quick fix, before I did it this way. I agree that that would be a good addition to the code.
Implementing it is fairly easy: [This line](https://github.com/MarlinFirmware/Marlin/blob/a1523d64199b69878d9208c335eaca293260077e/Marlin/src/gcode/calibrate/G34_M422.cpp#L292):
`if (last_z_align_level_indicator < z_align_level_indicator * 0.7f) {`
 could be changed to:
` if ((last_z_align_level_indicator < z_align_level_indicator * 0.7f) && ((Z_STEPPER_ALIGN_ACC * HARDCODED_CONSTANT) < z_align_level_indicator)) {`
while `HARDCODED_CONSTANT `could be about 5.0f. Alternatively, `(Z_STEPPER_ALIGN_ACC * HARDCODED_CONSTANT)` can be replaced by a new `#define` from `Configuration_adv.h`.

Another idea would be to detect the case in which the precision of the printer (kinematics or sensor) is at its limit and the values won't get better anymore anyway, so it could stop early, but not throw a error, just a warning.

> whether these very tiny deviations matter to your print results and how much they could be ignored

In my understanding, the user decides how much precision he needs by setting `Z_STEPPER_ALIGN_ACC`. Still, I haven't tested if the standard value of 20µm is appropriate. I did some tests regarding precision (repeatability) of Z probes some time ago. Standard inductive sensors reached <3µm deviation, while a 3D Touch was about an order of magnitude worse.

sjasonsmith, 2020-04-08:
I second @jufimu12's assertion that this looks like the behavior prior to their fix. Upgrading to a newer Marlin will almost certainly fix this issue.

ogsv, 2020-04-10:
My bad, I thought I had synced my fork to 2.0.5.2, but evidently I hadn't.
Patched my G34_M422.cpp to the latest version, and the addition of
`&& ((Z_STEPPER_ALIGN_ACC * 5.0f) < z_align_level_indicator))` works a treat! No false negatives in roughly 50 iterations tested so far.
Thanks a lot for the feedback guys, and sorry that I was outdated to start with...

closed 2020-04-10 by ogsv