MarlinFirmware/Marlin #17434: marlin power loss recovery doesn't work on pause print

narendok, 2020-04-07:
marlin power loss recovery doesn't work on pause print
When I pause my print & when sudden power cut off, after selecting recovery option, it doesn't show resume printing?
could someone pls check it.

thinkyhead, 2020-04-08:
In future issues will be automatically closed if the template is not filled out. We require information to determine whether there is a bug or just a confused user.

- What version of Marlin?
- Are you sure you turned on Power Loss Recovery (`M413 S1`) before your print?
- Have you done several tests with extra logging enabled?

boelle, 2020-06-20:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

narendok, 2020-06-20:
sorry, I couldn't update but i guess, something was my fault ,

thanks

btw 1 question, we build cheapest affordable iot printer "Slog3d"

do u think any copyright issue using your code, according to open-source
license, we've  read
but i just confirming if anything special ,  i need to keep in mind

-- 
Thanks & Regard:
*Narendra Verma*

Founder & CTO,
RoboSlog <http://www.roboslog.in>
[image: Facebook icon] <https://www.facebook.com/RoboSlog>  [image:
LinkedIn icon] <https://www.linkedin.com/company/RoboSlog>  [image: Twitter
icon] <https://twitter.com/RoboSlog>  [image: Youtbue icon]
<https://www.youtube.com/RoboSlog>  [image: Instagram icon]
<https://www.instagram.com/RoboSlog>
+91 8851719915 <8851719915>

boelle, 2020-06-20:
@thinkyhead will have to answer license issues

closed 2020-06-20 by boelle