MarlinFirmware/Marlin #17435: TMC2130 v3.0 form BigTreeTech not working with Marlin 2.0.5.3 and RAMPS 1.4

TomRous, 2020-04-07:
As described, it cannot connect to my Arduino Mega 2560 clone board with RAMP shield v1.4. I tried also older version of Marlin. I also use 2004 LCD with SD card reader.

**Problem**
Drivers cannot comunnicate with MCU with SPI. When sending M122 it shows address 0xFF:FF:FF etc.
**What I tried?**
Marlin 1.x
Marlin 2.x
Test jumper wires with multimeter- they are OK.
Check bottom of drivers- two micro jumpers connected correctly, SPI jumper not connected.
Software SPI as shown on [this Chris´ video]( https://www.youtube.com/watch?v=kMICfCa6bHk)

[Driver´s Github](https://github.com/bigtreetech/BIGTREETECH-TMC2130-V3.0)

furry-cheetah, 2020-04-19:
I have the same setup, working properly. Delta, mega2560+ramps2.4, bltouch, full graphic display, sd, bigtreetech v3 tmc2130 spi drives. I remove jumpers on biard and cut 4 legs at bottom side (to ramps side) where connected spi. And it works properly. Sd card works too fine. 
[image]

furry-cheetah, 2020-04-19:
[image]

furry-cheetah, 2020-04-19:
In configuration.h you need uncomment:
```
#define X_DRIVER_TYPE TMC2130
#define Y_DRIVER_TYPE TMC2130
#define Z_DRIVER_TYPE TMC2130
#define E0_DRIVER_TYPE TMC2130
```
in configuration_adv.h:

```
  #define X_CS_PIN          63
  #define Y_CS_PIN          40
  #define Z_CS_PIN          42
  #define E0_CS_PIN         65

#define STEALTHCHOP_XY
  #define STEALTHCHOP_Z
  #define STEALTHCHOP_E

  #define MONITOR_DRIVER_STATUS

  #if ENABLED(MONITOR_DRIVER_STATUS)
    #define CURRENT_STEP_DOWN     50  // [mA]
    #define REPORT_CURRENT_CHANGE
    #define STOP_ON_ERROR
  #endif

#define TMC_DEBUG
```
This is aux2 pins (yellow cables on photo, chip select on spi bus), you may use a different ones, any free pin (except aux-1 D0 and D1 pins - its a hardware uart0 to communicate with mcu and uploading firmware) i show my setup.

3 pins (green, blue and red you need connect to hardware spi (aux-3 50,51,52) exactly to this pins. I have a graphic display with sd and just solder 3 addition pins on header.

boelle, 2020-06-20:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-20 by boelle