MarlinFirmware/Marlin #17438: [Regression] Compile failure in stepper_dac.cpp due to "Use PSTR for common labels" 5e9a10dbaeaa885a670af0bd7ef3e7c068f5a7b8 March 1 commit

fiveangle, 2020-04-08:
Prior to "Use PSTR for common labels" https://github.com/MarlinFirmware/Marlin/commit/5e9a10dbaeaa885a670af0bd7ef3e7c068f5a7b8 March 1 commit, compile succeeds.

Subsequent to 5e9a10dbaeaa885a670af0bd7ef3e7c068f5a7b8, compile fails.  Rebasing on latest BF2 079644d01244a347c1535fac4beab61e5c1f853c compile still fails identically.

Configs based on latest BF2
https://github.com/fiveangle/Marlin/blob/bf2_dac_pstr_change/Marlin/Configuration.h
https://github.com/fiveangle/Marlin/blob/bf2_dac_pstr_change/Marlin/Configuration_adv.h

<details>

<summary>

Compile failure log...

</summary>

[...]
avr-g++ -o .pio/build/at90usb1286_dfu/src/src/feature/digipot/digipot_mcp4451.cpp.o -c -fno-exceptions -felide-constructors -std=gnu++11 -fsingle-precision-constant -fmax-errors=5 -g -fmerge-all-constants -Os -Wall -ffunction-sections -fdata-sections -mmcu=at90usb1286 -DPLATFORMIO=40302 -DTEENSY2PP -D__MARLIN_FIRMWARE__ -DUSB_SERIAL -DARDUINO=10805 -DTEENSYDUINO=145 -DF_CPU=16000000L -DLAYOUT_US_ENGLISH -IMarlin -I.pio/libdeps/at90usb1286_dfu/TMC26XStepper -I.pio/libdeps/at90usb1286_dfu/SlowSoftI2CMaster -I.pio/libdeps/at90usb1286_dfu/SailfishRGB_LED -I.pio/libdeps/at90usb1286_dfu/SailfishLCD -I.pio/libdeps/at90usb1286_dfu/Arduino-L6470/src -I.pio/libdeps/at90usb1286_dfu/LiquidTWI2 -I.pio/libdeps/at90usb1286_dfu/Adafruit_MAX31865 -I.pio/libdeps/at90usb1286_dfu/U8glib-HAL_ID1932/src -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/SPI "-I.pio/libdeps/at90usb1286_dfu/Adafruit NeoPixel_ID28" -I.pio/libdeps/at90usb1286_dfu/LiquidCrystal_ID136 -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/Wire -I/home/speedster/.platformio/packages/framework-arduinoteensy/libraries/Wire/utility -I/home/speedster/.platformio/packages/framework-arduinoteensy/cores/teensy -I/home/speedster/.platformio/packages/framework-arduinoteensy/cores Marlin/src/feature/digipot/digipot_mcp4451.cpp
In file included from Marlin/src/feature/dac/../../inc/MarlinConfig.h:46:0,
                 from Marlin/src/feature/dac/stepper_dac.cpp:27:
Marlin/src/feature/dac/stepper_dac.cpp: In function 'void dac_print_values()':
Marlin/src/feature/dac/stepper_dac.cpp:92:5: error: 'SP_X_LBL' was not declared in this scope
     SP_X_LBL, dac_perc(X_AXIS), PSTR(" ("), dac_amps(X_AXIS), PSTR(")")
     ^
Marlin/src/feature/dac/../../inc/../core/serial.h:135:49: note: in definition of macro '_SEP_2_P'
 #define _SEP_2_P(PRE,V)     serial_echopair_PGM(PRE,V)
                                                 ^
Marlin/src/feature/dac/../../inc/../core/serial.h:192:30: note: in expansion of macro '_SELP_17_P'
 #define __SELP_N_P(N,V...)   _SELP_##N##_P(V)
                              ^
Marlin/src/feature/dac/../../inc/../core/serial.h:193:30: note: in expansion of macro '__SELP_N_P'
 #define _SELP_N_P(N,V...)    __SELP_N_P(N,V)
                              ^
Marlin/src/feature/dac/../../inc/../core/serial.h:219:35: note: in expansion of macro '_SELP_N_P'
 #define SERIAL_ECHOLNPAIR_P(V...) _SELP_N_P(NUM_ARGS(V),V)
                                   ^
Marlin/src/feature/dac/stepper_dac.cpp:91:3: note: in expansion of macro 'SERIAL_ECHOLNPAIR_P'
   SERIAL_ECHOLNPAIR_P(
   ^
Marlin/src/feature/dac/stepper_dac.cpp:93:5: error: expected ')' before 'SP_Y_LBL'
     SP_Y_LBL, dac_perc(Y_AXIS), PSTR(" ("), dac_amps(Y_AXIS), PSTR(")")
     ^
Marlin/src/feature/dac/../../inc/../core/serial.h:135:49: note: in definition of macro '_SEP_2_P'
 #define _SEP_2_P(PRE,V)     serial_echopair_PGM(PRE,V)
                                                 ^
Marlin/src/feature/dac/../../inc/../core/serial.h:208:49: note: in expansion of macro '_SELP_13_P'
 #define _SELP_15_P(a,b,V...) do{ _SEP_2_P(a,b); _SELP_13_P(V); }while(0)
                                                 ^
Marlin/src/feature/dac/../../inc/../core/serial.h:210:49: note: in expansion of macro '_SELP_15_P'
 #define _SELP_17_P(a,b,V...) do{ _SEP_2_P(a,b); _SELP_15_P(V); }while(0)
                                                 ^
Marlin/src/feature/dac/../../inc/../core/serial.h:192:30: note: in expansion of macro '_SELP_17_P'
 #define __SELP_N_P(N,V...)   _SELP_##N##_P(V)
                              ^
Marlin/src/feature/dac/../../inc/../core/serial.h:193:30: note: in expansion of macro '__SELP_N_P'
 #define _SELP_N_P(N,V...)    __SELP_N_P(N,V)
                              ^
Marlin/src/feature/dac/../../inc/../core/serial.h:219:35: note: in expansion of macro '_SELP_N_P'
 #define SERIAL_ECHOLNPAIR_P(V...) _SELP_N_P(NUM_ARGS(V),V)
                                   ^
Marlin/src/feature/dac/stepper_dac.cpp:91:3: note: in expansion of macro 'SERIAL_ECHOLNPAIR_P'
   SERIAL_ECHOLNPAIR_P(
   ^
Marlin/src/feature/dac/stepper_dac.cpp:94:5: error: expected ')' before 'SP_Z_LBL'
     SP_Z_LBL, dac_perc(Z_AXIS), PSTR(" ("), dac_amps(Z_AXIS), PSTR(")")
     ^
Marlin/src/feature/dac/../../inc/../core/serial.h:135:49: note: in definition of macro '_SEP_2_P'
 #define _SEP_2_P(PRE,V)     serial_echopair_PGM(PRE,V)
                                                 ^
Marlin/src/feature/dac/../../inc/../core/serial.h:204:49: note: in expansion of macro '_SELP_9_P'
[... 37 lines cut ...]
                                                 ^
Marlin/src/feature/dac/../../inc/../core/serial.h:206:49: note: in expansion of macro '_SELP_11_P'
 #define _SELP_13_P(a,b,V...) do{ _SEP_2_P(a,b); _SELP_11_P(V); }while(0)
                                                 ^
Marlin/src/feature/dac/../../inc/../core/serial.h:208:49: note: in expansion of macro '_SELP_13_P'
 #define _SELP_15_P(a,b,V...) do{ _SEP_2_P(a,b); _SELP_13_P(V); }while(0)
                                                 ^
Marlin/src/feature/dac/../../inc/../core/serial.h:210:49: note: in expansion of macro '_SELP_15_P'
 #define _SELP_17_P(a,b,V...) do{ _SEP_2_P(a,b); _SELP_15_P(V); }while(0)
                                                 ^
Marlin/src/feature/dac/../../inc/../core/serial.h:192:30: note: in expansion of macro '_SELP_17_P'
 #define __SELP_N_P(N,V...)   _SELP_##N##_P(V)
                              ^
Marlin/src/feature/dac/../../inc/../core/serial.h:193:30: note: in expansion of macro '__SELP_N_P'
 #define _SELP_N_P(N,V...)    __SELP_N_P(N,V)
                              ^
Marlin/src/feature/dac/../../inc/../core/serial.h:219:35: note: in expansion of macro '_SELP_N_P'
 #define SERIAL_ECHOLNPAIR_P(V...) _SELP_N_P(NUM_ARGS(V),V)
                                   ^
Marlin/src/feature/dac/stepper_dac.cpp:91:3: note: in expansion of macro 'SERIAL_ECHOLNPAIR_P'
   SERIAL_ECHOLNPAIR_P(
   ^
*** [.pio/build/at90usb1286_dfu/src/src/feature/dac/stepper_dac.cpp.o] Error 1

</details>

thinkyhead, 2020-04-08:
Thanks for the note. Now patched!

closed 2020-04-08 by thinkyhead

drphil3d, 2020-04-25:
I'm still having this issue with the latest 2.0.5.3 as well on 2.0.5.2

`
In file included from Marlin\src\feature\dac\../../inc/MarlinConfig.h:45:0,

                 from Marlin\src\feature\dac\stepper_dac.cpp:27:

Marlin\src\feature\dac\stepper_dac.cpp: In function 'void dac_print_values()':

Marlin\src\feature\dac\stepper_dac.cpp:92:5: error: 'SP_X_LBL' was not declared in this scope

     SP_X_LBL, dac_perc(X_AXIS), PSTR(" ("), dac_amps(X_AXIS), PSTR(")")

     ^

Marlin\src\feature\dac\../../inc/../core/serial.h:135:49: note: in definition of macro '_SEP_2_P'

 #define _SEP_2_P(PRE,V)     serial_echopair_PGM(PRE,V)

                                                 ^

Marlin\src\feature\dac\../../inc/../core/serial.h:192:30: note: in expansion of macro '_SELP_17_P'

 #define __SELP_N_P(N,V...)   _SELP_##N##_P(V)

                              ^

Marlin\src\feature\dac\../../inc/../core/serial.h:193:30: note: in expansion of macro '__SELP_N_P'

 #define _SELP_N_P(N,V...)    __SELP_N_P(N,V)

                              ^

Marlin\src\feature\dac\../../inc/../core/serial.h:219:35: note: in expansion of macro '_SELP_N_P'

 #define SERIAL_ECHOLNPAIR_P(V...) _SELP_N_P(NUM_ARGS(V),V)

                                   ^

Marlin\src\feature\dac\stepper_dac.cpp:91:3: note: in expansion of macro 'SERIAL_ECHOLNPAIR_P'

   SERIAL_ECHOLNPAIR_P(

   ^

Marlin\src\feature\dac\stepper_dac.cpp:93:5: error: expected ')' before 'SP_Y_LBL'

     SP_Y_LBL, dac_perc(Y_AXIS), PSTR(" ("), dac_amps(Y_AXIS), PSTR(")")

     ^

Marlin\src\feature\dac\../../inc/../core/serial.h:135:49: note: in definition of macro '_SEP_2_P'

 #define _SEP_2_P(PRE,V)     serial_echopair_PGM(PRE,V)

                                                 ^

Marlin\src\feature\dac\../../inc/../core/serial.h:208:49: note: in expansion of macro '_SELP_13_P'

 #define _SELP_15_P(a,b,V...) do{ _SEP_2_P(a,b); _SELP_13_P(V); }while(0)

                                                 ^

Marlin\src\feature\dac\../../inc/../core/serial.h:210:49: note: in expansion of macro '_SELP_15_P'

 #define _SELP_17_P(a,b,V...) do{ _SEP_2_P(a,b); _SELP_15_P(V); }while(0)

                                                 ^

Marlin\src\feature\dac\../../inc/../core/serial.h:192:30: note: in expansion of macro '_SELP_17_P'
[... 130 lines cut ...]

Marlin\src\feature\dac\../../inc/../core/serial.h:192:30: note: in expansion of macro '_SELP_17_P'

 #define __SELP_N_P(N,V...)   _SELP_##N##_P(V)

                              ^

Marlin\src\feature\dac\../../inc/../core/serial.h:193:30: note: in expansion of macro '__SELP_N_P'

 #define _SELP_N_P(N,V...)    __SELP_N_P(N,V)

                              ^

Marlin\src\feature\dac\../../inc/../core/serial.h:219:35: note: in expansion of macro '_SELP_N_P'

 #define SERIAL_ECHOLNPAIR_P(V...) _SELP_N_P(NUM_ARGS(V),V)

                                   ^

Marlin\src\feature\dac\stepper_dac.cpp:91:3: note: in expansion of macro 'SERIAL_ECHOLNPAIR_P'

   SERIAL_ECHOLNPAIR_P(

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4533911/config.zip)
`

thinkyhead, 2020-04-26:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

commit 3ea23712 (2020-04-08): Fix stepper DAC compile

Fixes #17438