MarlinFirmware/Marlin #17440: [BUG] M401/M402 does not operate BL-touch unless HS mode enabled

mel4321, 2020-04-08:
### Bug Description
M401/M402 does not lift/stow bl-touch, I have started on instalation of Marlin on my U30, I do not know the impact on printing itself.

### My Configurations
Example\Alfawise\U20-bltouch

### Steps to Reproduce
1. test BL-touch operation via LCD menu, should work
2. send M401 and/or M402 codes via serial port, bl-touch does not move

#### Additional Information
I had troubleshooted issue. Them menu functions directly operate on bltouch object, the M401/M402 operates on probe object. I walk thru function probe_specific_action in 
https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/module/probe.cpp
and it looks like the is unimplemented condition for BLTOUCH with disable high speed mode.

`#elif HAS_Z_SERVO_PROBE`
`    #if DISABLED(BLTOUCH)`
`      MOVE_SERVO(Z_PROBE_SERVO_NR, servo_angles[Z_PROBE_SERVO_NR][deploy ? 0 : 1]);`
`    #elif ENABLED(BLTOUCH_HS_MODE)`
`      // In HIGH SPEED MODE, use the normal retractable probe logic in this code`
`      // i.e. no intermediate STOWs and DEPLOYs in between individual probe actions`
`     if (deploy) bltouch.deploy(); else bltouch.stow();`
`    #endif`

If BLTOUCH is ENABLED, and BLTOUCH_HS_MODE disabled, nothing will happen here.
I do not understand the intention of developer, for me the comment says same code is used for high speed and legacy. Probe object is some kind of generic object used in many places, I do not understand why this possible bug impact on other operations like bed leveling and has not been discovered earlier.

WDuck, 2020-04-09:
So after enabling BLTOUCH_HS_MODE M401 and M402 do work but on G28 it still doesn't deploy.

mel4321, 2020-04-09:
I had modified code in probe.cpp, changed 
`//    #elif ENABLED(BLTOUCH_HS_MODE)`
to
`    #elif 1`
The result is G28, G29, M401 and M402 can control the BLTouch. "M43 S" also works.
For M401/402 I am sure that it didn't work before this modification, for other g-codes I am not 100% sure. What I remember, the behaviour of G28 was strange, first it homed X and Y as ussual, then pull up Z and stopped. This could be triggered by this sw fault or some other issue with my bl-touch.

WDuck, 2020-04-09:
So after implementing the above mentioned adaption in probe.cpp M401, M402 and G30 were working for me.
G28 is now working after additionally changing line 586 in "Onditionals_LCD.h" to get "HOMING_Z_WITH_PROBE"
Old:
```
  #if Z_HOME_DIR < 0 && !HAS_CUSTOM_PROBE_PIN
    #define HOMING_Z_WITH_PROBE 1
```
New:
  ```
#if Z_HOME_DIR < 0 && HAS_CUSTOM_PROBE_PIN //"!" before HAS_CUSTOM_PROBE_PIN remooved
    #define HOMING_Z_WITH_PROBE 1
```

hills8, 2020-04-10:
> So after implementing the above mentioned adaption in probe.cpp M401, M402 and G30 were working for me.
> G28 is now working after additionally changing line 586 in "Conditionals_LCD.h" to get "HOMING_Z_WITH_PROBE"
> Old:
> 
> ```
>   #if Z_HOME_DIR < 0 && !HAS_CUSTOM_PROBE_PIN
>     #define HOMING_Z_WITH_PROBE 1
> ```
> 
> New:
> 
> ```
> #if Z_HOME_DIR < 0 && HAS_CUSTOM_PROBE_PIN //"!" before HAS_CUSTOM_PROBE_PIN remooved
>   #define HOMING_Z_WITH_PROBE 1
> ```

THX! that solved the problem for me too.

docmalc, 2020-04-10:
Did this issue all of a sudden start happening? 

It started last night for me but... 
I accidentally sent my Z a bit (a lot) to far up to where it really started pulling the sleeved bundle of cables (including bltouch) to my board really hard so I thought I pulled a wire from the board. 
But I'd rather do a couple Marlin tweaks and flash than to chase the wires to the board. 

> > So after implementing the above mentioned adaption in probe.cpp M401, M402 and G30 were working for me.
> > G28 is now working after additionally changing line 586 in "Conditionals_LCD.h" to get "HOMING_Z_WITH_PROBE"
> > Old:
> > ```
> >   #if Z_HOME_DIR < 0 && !HAS_CUSTOM_PROBE_PIN
> >     #define HOMING_Z_WITH_PROBE 1
> > ```
> > 
> > 
> > New:
> > ```
> > #if Z_HOME_DIR < 0 && HAS_CUSTOM_PROBE_PIN //"!" before HAS_CUSTOM_PROBE_PIN remooved
> >   #define HOMING_Z_WITH_PROBE 1
> > ```
> 
> THX! that solved the problem for me too.
> 
> Info: file is located in: Marlin/src/inc/Conditionals_LCD.h
> 
> this change was included with commit: [fe4c69b](https://github.com/MarlinFirmware/Marlin/commit/fe4c69b64abc74bc1ee1eb16d3d3771e3afbf25d)

lqbombjack, 2020-04-12:
Those two fixes work for me using a 3D Touch (BL-Touch clone)

I can confirm the M401/M402 commands started working for me once the BLTOUCH_HS_MODE option was enabled.  Prior to that I could only command the 3D Touch using M280 servo commands.

G28 Homing was Z Homing without deploying the probe and I would have to reset the board to prevent the nozzle crashing into the bed.  Inverting the HAS_CUSTOM_PROBE_PIN by removal of the "!" operator allowed the G28 command to deploy the 3D Touch sensor for homing and all works as expected.

Thanks to everyone for their issue comments.

thinkyhead, 2020-04-12:
A patch was applied in #14352 last June but the change was subsequently reverted pending further discussion. To summarize, the patch is to do this:
```diff
+  #elif ENABLED(BLTOUCH)
+
+    deploy ? bltouch.deploy() : bltouch.stow();
+
   #elif HAS_Z_SERVO_PROBE
 
+    MOVE_SERVO(Z_PROBE_SERVO_NR, servo_angles[Z_PROBE_SERVO_NR][deploy ? 0 : 1]);
-    #if DISABLED(BLTOUCH)
-      MOVE_SERVO(Z_PROBE_SERVO_NR, servo_angles[Z_PROBE_SERVO_NR][deploy ? 0 : 1]);
-    #elif ENABLED(BLTOUCH_HS_MODE)
-      // In HIGH SPEED MODE, use the normal retractable probe logic in this code
-      // i.e. no intermediate STOWs and DEPLOYs in between individual probe actions
-      if (deploy) bltouch.deploy(); else bltouch.stow();
-    #endif
 
   #elif EITHER(TOUCH_MI_PROBE, Z_PROBE_ALLEN_KEY)
```

thinkyhead, 2020-04-12:
Duplicate of #14328

closed 2020-04-12 by thinkyhead