MarlinFirmware/Marlin #17441: [FR] (feature request title)

jgoo9410, 2020-04-08:
I work for a company called 3dindustry and we have in development a 3d printer with 4 carriages and 4 extruders (with 4 hot-ends)

Our prototype is almost operational and we have managed to implement almost all the features we desire by some light modification in marlin. One feature we are trying to implement but have run aground on, is duplicating the E movements. The printer carriages can be setup in several configurations; 1-1-1-1 (quad printing), 1-1-1 (tri printing), 1-1 (dual printing), 1 (single printing), 11 (single carriage dual material), 11-11 etc etc.. 

One way we have attempted to implement the extruder control is by using the mixing extruder functionality. By altering the mix i.e. mix = 1,1,1,1 we would expect all 4 to extrude the same amount. If we were printing dual material dual carriage (11-11) we would set mix = 1,0,1,0 for material A and mix = 0,1,0,1 when printing material B. All good on that front. The issue arrises the having multiple extruders defined and mixing extruder defined, we get an error in the sanity check. Commenting out the errors in the sanity check doesn't resolve the issue. This of course means that we cannot control the 4 hot-end temperatures while having mixing enabled.

My request is that the two aren't mutually exclusive. Or if you know of a workaround that would suit the current setup? I know there is other firmware which allows this functionality but based on the plethora of useful features embedded in marlin we would like to move forward with it as our core firmware.

thinkyhead, 2020-04-08:
You should implement what you want directly rather than try to piggyback on Marlin's existing features. Most of what you want can be gleaned by looking at how `DUAL_X_CARRIAGE` modifies behavior and then doing more of that. The mixing extruder directly apportions steps from a single E axis to several steppers, providing the basic model for that trick. If you want to apportion steps in a more complicated way you will need to include that in your implementation.

jgoo9410, 2020-04-08:
Thanks for the info. I have had a look at the code surrounding `DUAL_X_CARRIAGE` In stepper.cpp but haven’t been able to find the segment that duplicates the E moves. Could you give me a little more to go on?

jgoo9410, 2020-04-08:
I managed to find the section responsible for the e-positioning. Full disclosure, I’m an EE and my cpp knowledge is more of a working one, so some of your code is a bit over my head. 

One thing I am certain of however, is that you aren’t going to like my implementation, so apologies in advance! 

```
#if DISABLED(MIXING_EXTRUDER)
  #define E_APPLY_STEP(v,Q) E_STEP_WRITE(stepper_extruder, v); E_STEP_WRITE(1, v);
#endif
```

So as above, I have just passed 1 to E_STEP_WRITE in place of stepper_extruder in order that E1 makes the same move as E0. I will have some mode dependant statement here to achieve the correct movements based on the current configuration of the gantry. 

That solves my movement problem.

```
if (motor_direction(E_AXIS)) {
        REV_E_DIR(stepper_extruder);
        REV_E_DIR(1);
        count_direction.e = -1;
      }
      else {
        NORM_E_DIR(stepper_extruder);
        NORM_E_DIR(1);
        count_direction.e = 1;
      }
```

Much the same situation in the direction section. Again not very pretty, but functional. 
I am not sure if this will throw the stepper counting out of sync, but as far as I’m aware it all get reset on a G28 move which eliminates any issues that might cause for our particular setup. 

Let me know what you think if you have the time.

Roxy-3D, 2020-04-09:
@jgoo9410   I would be happy to make the machine work as you want it.    I did a similar deal with Formbot to add the Symetric Duplication mode and the IDEX Mode menus to their  IDEX machine.   

Right now, a group of us have a project going to put color mixing extruders onto an IDEX machine.   

If you are interested in some help let me know...

jgoo9410, 2020-04-09:
Hi @Roxy-3D. Thank you for the offer, id be very grateful if you could help me implement the desired functionality. 

Your colour mixing project sounds interesting. I had a go at making one of my deltas a colour mixing printer a couple of years ago. A very expensive project in the end. I was using the diamond hot-end, but found that it really wasn't reliable enough to use. Dismantling it was a complete pain when it inevitably clogged too! What hardware are you using for your project?

As for the printer in question, here is a little info:
- It is single x-axis single y-axis with 4 hot-ends and 4 extruders.
- All of the hot-ends are on the x axis. 
- The carriages can be engaged and disengaged with the X belt through a separate piece of hardware, at the beginning of each print. Inactive carriage are parked at X min (we don't need any functionality for this feature as its controlled by our front end module). 
-The printer is designed to be as flexible as possible, allowing numerous different extruder combos for duplicate printing and multi material printing. I have tabulated the possible configurations here; https://imgur.com/a/jPqHZjQ

I have created an M command, `M444` to be specific, that does a bit of setup at the beginning of a print. If the setup is 4 separate carriages the command sent is `M444 C4`, perhaps a variable I could set the mode to in there would be a sensible implementation? Just an idea, if you have a different implementation in mind, ignore me.

Roxy-3D, 2020-04-09:
> Hi @Roxy-3D. Thank you for the offer, id be very grateful if you could help me implement the desired functionality.
> ...
> What hardware are you using for your project?

It is very difficult to write (and debug) code without a hardware platform to test against.
Typically a business that wants special code written provides an actual unit to facilitate the development.    So in the case of the code you want done for your company's machine the development platform would be a Production Level Prototype of what ever you plan on shipping.

What our group is going to use for the color mixing IDEX machine is a custom designed board with 12 sockets for step sticks.   3 for each extruder, 2 for the X-Axis to allow both carriages to move. and then 2 more for Y & Z.  (And 2 spares).     We have not identified a printer to put the electronics onto yet.

https://cdn.discordapp.com/attachments/690290520488017973/696849590111961159/20200406_173835.jpg

InsanityAutomation, 2020-04-09:
Weve actually been working on something expanding on this with the board we've been developing as well.

Basically this would need extruder duplication extending dual x carriage with dual y carriage to allow enabling each carriage to operate in synced mode or mirrored mode, then enable / disable each extruder.

What youre describing is the simple extruder duplication modes which should already exist similar to the functionality of Stacker by the sound of it. Take a look at multi nozzle duplication and M605. This feature was written blind as we did not have a machine to test on when it was requested. This was written by @thinkyhead and myself about a year ago.

[image]

jgoo9410, 2020-04-09:
Thank you both for your comments. I understand the benefits of having a unit to test on and in the moths to come, provided manufacturing access has been restored, I would be happy to send you a pre production unit @Roxy-3D. Currently we are in the position where me must validate our design before further investment can be expended. Which is why we need to see these features implemented before the project can move any further. We are pretty much there with the electronics and the hardware. If you feel you cannot implement the features without the unit in-front of you, I understand and I will continue along the vein I had been.

We have also designed our own electronics platform, which Is something I do have spare units of if you were at all interested. The setup consists of 3 modules. The motherboard contains a compute module(running linux), for running remote services onboard, with a custom front end on a 7" touch display, built in WIFI, ethernet and USB support. The first daughter board hosts 7 TMC 2130's, external Z control, the atmega2560 and 5 6A heater lines, all within a 85*80mm footprint https://imgur.com/gallery/EQjSw4n (scribbled on to cover my name). The second currently exists as an uno shield for testing but there is an order out for the complete unit with the uC embedded, this board handles all of the Z movements and performs true bed levelling. 

I don't know what your situation in your group is regarding electronics development, but I would be happy to assist an any electronics related matters, should you need.

Roxy-3D, 2020-04-09:
> Thank you both for your comments. I understand the benefits of having a unit to test on and in the moths to come, provided manufacturing access has been restored, I would be happy to send you a pre [...]

If your company does decide to produce the machine, please keep me in mind.   I'll get the full feature set working nicely and everything would be fully tested.    But I do need a machine to code against.

jgoo9410, 2020-04-09:
Okay I will do, thanks anyway. Just out of curiosity, does the marlin development have a location or are the developers scattered all over the place?

InsanityAutomation, 2020-04-09:
Roxy and myself are both on discord, so its an easy place to discuss in a more direct fashion. 

https://discord.gg/e7Y4mU

Developers are all over. Many in the US (Thinkyhead in Texas, myself near Detroit) and others all around the world.

Heres the board weve been working on to support both mixing idex and quad carriage.

[image]

Im a bit curious on the usage of an atmega2560 still in a new design. Any reason you arent looking at an LPC1768/9?

Also given the application, you can easily make use of G34 in order to perform either tri (stable) or quad (beta) stepper gantry alignment. You mentioned true leveling, and thats what this function performs.

Roxy-3D, 2020-04-09:
The Marlin developers are scattered across the world.   Probably, the biggest clusters of Marlin Developers are in the USA and Europe.    You can direct message me on the Marlin Discord server.  This is the link to get to it:   https://discord.gg/n5NJ59y      

We can swap email addresses there.  I am 'Roxy' on the Discord server.   I look forward to seeing more details about the project!

jgoo9410, 2020-04-09:
All good questions. I have been toying with the idea of moving to a 32 bit controller. I am vary familiar with the ATMEGA 2560 which is why I used it. It costs too much vs other controllers now which is what’s inspiring a change over. If I can port it easily enough I will do it. As for true bed levelling, it’s more of a pin availability issue than anything else.

Why have you used pin headers instead of embedding the stepper drivers? Have you considered embedding a compute module of some description so you can implement high level features such as print monitoring etc? 

Based on my carriage mechanism I decided to implement my own G-Code. It works and I’m happy with it so I’ll probably leave it alone. 

Il contact you both on discord and share some of the product details.

Roxy-3D, 2020-04-09:
> Why have you used pin headers instead of embedding the stepper drivers? Have you considered embedding a compute module of some description so you can implement high level features such as print monitoring etc?

Probably, the best answer is flexibility and cost.   Even with a 3 color mixing extruder on both carriages, that is 10 stepper motors.   The 12 stepper motor design is because we wanted some room for expansion.    And given everybody has their own favorite stepper driver, having sockets lets us use what ever an individual is comfortable with using.

All of the unused pins have been brought out to an expansion header so other features (like true bed leveling) can still be configured.

The board has a fairly powerful 32-bit processor on it.    I have already moved the Free-RTOS operating system under Marlin in the past.   At some point, we will resurrect that work on the 32-bit platforms and have Real Time Multi Tasking support in place.   The 32-bit processors have enough speed and memory, most likely we can do any other tasks that are desired on the main processor and not need an auxiliary processor.

InsanityAutomation, 2020-04-09:
Also given a board of this size, fully populating all drivers is quite costly. Complicate that with its intended use currently for development of advanced features, where developers are 10x more likely to blow stuff up and one occurrence is alot of $$. We also need to be able to move across different drivers for testing various things constantly. We are aware of the thermal constraints vs sunk into a copper pad on the PCB for cooling, and have no concern working active cooling around it.

The pin constraints go away with a 208 pin package like the LPC 4078 in use here...

At least take a look at M605, it really sounds like its exactly what youre doing. I dont see much of a difference in function vs a Stacker S4

```
 /**
   * M605: Set multi-nozzle duplication mode
   *
   *  S2       - Enable duplication mode
   *  P[mask]  - Bit-mask of nozzles to include in the duplication set.
   *             A value of 0 disables duplication.
   *  E[index] - Last nozzle index to include in the duplication set.
   *             A value of 0 disables duplication.
   */
```

jgoo9410, 2020-04-09:
Are you planning on retailing the board? 

Is that from marlin 2.0?
In the 2.0 I have, I need to enable dual X carriage and have max endstops etc, which I don’t have. Could I bypass the dual carriage and enable duplication? 

I’m 90% of the way to being convinced that the LPC is the way forward. Time will tell if I can be bothered reworking the board. If the lockdown goes on any longer, I might do it to kill some time. Truth be told, I have spent more time trying to implement duplicating than it would have taken me to rework the board.

Roxy-3D, 2020-04-09:
The board was designed to handle Marlin v2.0.x and it was done to give the flexibility of controlling what ever was needed.   We talked to a few machine makers about doing a color mixing IDEX machine and the big problem was we didn't have hardware we could point at that could handle the job.

Well...   We just now have hardware that can handle the job.   The group is ready to start bringing up the board and porting Marlin to it.   We don't have the board mounted to a particular hardware platform yet.   But if we can find a hardware maker willing to work with us...   We would make that machine the 'Reference Platform' for color mixing IDEX machines.

Independent of all that...   Yes...   You can buy the board.   On Marlin Discord server, you can send a message to Gee.   He is the hardware designer that did the board for us.

InsanityAutomation, 2020-04-10:
What you want is multi nozzle duplication, which is below dual x carriage.

jgoo9410, 2020-04-10:
I just found it.... F
Il try compiling it in a bit. 

As for suitable hardware @Roxy-3D and @InsanityAutomation. I will send you through some specifics later today on discord and if the machine meets the spec that you are looking for, then we can talk about the possibility getting one to you.  

Have you got a colour mixing nozzle yet? When I looked a few years ago, the only available option was the diamond hot end.

Roxy-3D, 2020-04-10:
> Have you got a colour mixing nozzle yet? When I looked a few years ago, the only available option was the diamond hot end.

Not yet...   But we could do this in phases.   We could start with a normal IDEX machine.   If there is room to change the nozzles and mount more stepper motors and filament spools, we would be making progress towards the goal.

jgoo9410, 2020-04-10:
All of that is very much a possibility on our machine. The machine just over 1M in length and .5m in width with a build area to match. The bigger issue in my mind is the nozzle. I had a hell of a time with the diamond hotend and I worked for a company here in the U.K. that attempted to make their own, with little success. If you asked me, I would consider the possibility of commissioning one from a well known hotend manufacturer. It would obviously depend on your budget for the project.

InsanityAutomation, 2020-04-11:
> Hi @Roxy-3D. Thank you for the offer, id be very grateful if you could help me implement the desired functionality.
>
> Your colour mixing project sounds interesting. I had a go at making one of my [...]

Im trying to make sure I fully grasp this concept here. Is a single stepper / belt controlling all 4 extruder heads, or are they on independent X axis as well as E axis? The first 2 points lean towards the single stepper, but the 3rd item creates some question here.

If the second, why an external controller? It seems rather straightforward to extend IDEX to quad X and then enable 4 material switching with those heads as well as the duplicate mode, and gain mirrored control with heads.

Roxy-3D, 2020-04-11:
I took it to mean they were similar to the generic Parking Extruders or Magnetic Parking Extruders.
They can be parked to get them out of the way.    But re-engaged later when needed.

jgoo9410, 2020-04-11:
I have videoed the prototype I have but was unable to add you as a friend on discord. I managed to add @InsanityAutomation so you could maybe add me the through them. 

We aren’t particularly interested in mirrored printing, I can’t see much benefit to being able to print mirrored parts. What we are trying to achieve is tandem printing. 

M605 is a good start but there doesn’t seem to be a way to use 2 carriages with 2 materials I.e. duplicating T1 and T2 moves. I’m sure it wouldn’t take much modification of M605 to obtain the desired result which might be a good avenue to go down.

Roxy-3D, 2020-04-11:
> I have videoed the prototype I have but was unable to add you as a friend on discord. I managed to add...

@jgoo9410   What user name did you use on Discord?  I'll add it, but I didn't find jgoo9410 there.