MarlinFirmware/Marlin #17444: [BUG] Disconnecting USB sometimes causes Archim to reboot

marciot, 2020-04-08:
### Bug Description

Under a very specific set of circumstances, disconnecting the USB will cause the Archim board to crash some time in the future.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4452106/config.zip)

### Steps to Reproduce

1. Connect to Archim via the serial port using PuTTY (or some other host software)
2. Issue "M155 S1"
3. Pull the USB cable (without disconnecting host)
4. Wait for a minute or two.

**Expected behavior:**

The board should continue to work.

**Actual behavior:**

The board may spontaneously reboot after a couple dozen seconds. This does not *always* happen, but it happens frequently enough to be noticeable.

It's worth mentioning that if you quit PuTTY or disconnect prior to pulling the USB cable, then the crash will not happen. It seems to be related to forcefully breaking a connection and data being written to the serial port after the connection is broken (hence the M155 S1)

thinkyhead, 2020-04-08:
Since the host will be disconnected, see if you can log to the SD card or simply to the screen. What you want to narrow down is the exact point in the code flow where the crash is occurring.

marciot, 2020-04-08:
I just tried enabling a log but the log file is completely empty. Not even sure if the log to SD works with the USB flashdrive code. This is the first I've learned of this feature.

marciot, 2020-04-08:
In case the USB hardware was getting in the way, I swapped out my USB module with an SD module and tried new firmware with SD_SUPPORT. I was able to reproduce the crash, but the log file is also empty.

Incidentally, I notice the updates from M155 S1 aren't getting logged. Is this intended? What is and what isn't logged?

boelle, 2020-06-20:
@marciot is issue still there?

closed 2020-08-04 by github-actions[bot]