MarlinFirmware/Marlin #17446: Compile error when PID_Debug defined

jimhiggs, 2020-04-09:
I get an error when building latest bugfix 2.0:
`.pio/build/BIGTREE_SKR_PRO/src/src/module/temperature.cpp.o: In function `Temperature::get_pid_output_hotend(unsigned char)':
H:\Users\Jim\Documents\GitHub\Marlin/Marlin\src\module/temperature.cpp:863: undefined reference to `PID_Debug_Flag'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\BIGTREE_SKR_PRO\firmware.elf] Error 1`

I am not a C programmer but it seems PID_Debug_Flag in temperature.cpp is because var is declared but not defined anywhere? If I change Line 826
from  extern bool PID_Debug_Flag; to extern bool PID_Debug_Flag = 0;
it compiles.

sjasonsmith, 2020-04-13:
You'll need to post your configuration files as required by the issue template for anybody to try to reproduce this.

closed 2020-04-27 by thinkyhead

jelred, 2020-05-25:
Issue is letter casing conflict of the variable where it's referenced versus defined. It is defined in M303.cpp at line 47 as pid_debug_flag = 0 but is all lower case. In the temperature.cpp file variable is externally referenced at line 834 but as all upper case which causes the "declared but not defined" error during compile. Changing the variable in temperature to all lower case to match how it's defined corrects this allowing it to successfully compile.