MarlinFirmware/Marlin #17447: [BUG] MKS_Robin No Menu after boot logo

dandantsui, 2020-04-09:
### Bug Description

No info screen or display after boot screen.

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4454202/Marlin.zip)

### Steps to Reproduce

1. Downgrade back to 2.0.4.4
2. Try latest branch of 2.0.5.3

**Expected behavior:** Info menu screen and lcd buttons to navigate menu

**Actual behavior:** blank screen after bootlogo

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
[https://imgur.com/a/A8EdyFJ](url)

Miraculix200, 2020-04-09:
Try commenting out #define EEPROM_SETTINGS   and compile it again. It's probably this bug: https://github.com/MarlinFirmware/Marlin/issues/17376

dandantsui, 2020-04-12:
@Miraculix200 thanks for the help! 

I actually just added `#define SDCARD_EEPROM_EMULATION` into my pins file and it worked fine!

closed 2020-04-12 by thinkyhead