MarlinFirmware/Marlin #17448: [BUG]

riki9191, 2020-04-09:
marlin 1.1.x and 1.1.x bugfix do not compile using anet A8 configuration file

### My Configurations
[configuration_file.zip](https://github.com/MarlinFirmware/Marlin/files/4454972/configuration_file.zip)

### Steps to Reproduce

 1. use anet A8 configuration file to compile
2. set anet v1.0 (optiboot) as board
  3. compile using arduino ide

**Expected behavior:** 
I experct arduino ide compile without problem
**Actual behavior:**
I receve the following issue:
"exit status 1
non-constant condition for static assertion"

and the code stop compile in the following line

"static_assert(digitalPinToPCICR(Z_MIN_PIN) != NULL, "Z_MIN_PIN is not interrupt-capable")"
#### Additional Information
[image]

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ellensp, 2020-04-09:
Disable #define ENDSTOP_INTERRUPTS_FEATURE

riki9191, 2020-04-09:
yes, but before the last update of 1.1.x and using marlin 2.0.5 this feature is enable and arduino ide compile without a problem

riki9191, 2020-04-09:
so, I try several version of marlin (from 1.1.6 to 1.1.9 and buxfix 1.1.x) and with the last version of arduino ide (1.8.12) I am not able to compile using the Anet A8 configuration file.
I suspect that there is some problem with the last version of arduino ide and marlin 1.1.x.
if I try to compile any version of marlin after 2.0 there is no problem.

ellensp, 2020-04-09:
Marlin 1.1.x is no longer under development.  This is a issue with Arduino IDE, It works fine on platformio. Marlin found a way around it in 2.0  so there is nothing to do here.
Marlin replaced digitalPinToPCICR with digitalPinHasPCICR in newer versions to bypass the issue.  see https://github.com/MarlinFirmware/Marlin/issues/9632

riki9191, 2020-04-09:
ok, thank you. I have also find confirm in internet now. also is possible to downgrade the avr version.
i close the issue.

closed 2020-04-09 by riki9191