MarlinFirmware/Marlin #17449: [BUG] (Compile error Due)

annhan, 2020-04-09:
Compile error when Spindle or laser defined with arduino DUE:
```Marlin\src\HAL\DUE\../../inc/Conditionals_adv.h:128:37: error: pasting "_CUTTER_DISP_" and "(" does not give a valid preprocessing token```

swilkens, 2020-04-09:
Please post your configuration files as a .zip file, as the template suggests.

ellensp, 2020-04-09:
This would seem to be a #define conflict over PWM
In .platformio/packages/framework-arduino-sam/system/CMSIS/Device/ATMEL/sam3xa/include/sam3x8e.h line 466 is 
```
#define PWM        ((Pwm    *)0x40094000U) /**< \brief (PWM       ) Base Address */
```
So when marlin attempts to use PWM has been replaced with the above. The marlin code is 

```
  #define _CUTTER_DISP(V)      _CAT(_CUTTER_DISP_, V)
  #define CUTTER_DISPLAY_IS(V) (_CUTTER_DISP(CUTTER_POWER_DISPLAY) == _CUTTER_DISP(V))
```
The last line is called first. it calls the previous line and attempts to join the token `_CUTTER_DISP_` with the variable CUTTER_POWER_DISPLAY which should be 
"PWM" but PWM is defined so is replaced with "((Pwm    *)0x40094000U)"
The result is the token `_CUTTER_DISP_((Pwm    *)0x40094000U)`  which is not in the list

```
  #define _CUTTER_DISP_PWM     1
  #define _CUTTER_DISP_PERCENT 2
  #define _CUTTER_DISP_RPM     3
```
Resulting in the error "does not give a valid preprocessing token"
As it is expecting a token name _CUTTER_DISP_PWM or _CUTTER_DISP_PERCENT or _CUTTER_DISP_RPM.

Commenting out the #define in sam3x8e.h gets past this error, but then bombs out with other framework code errors that require this #define.

I don't know how to proceed from here, so documenting what I found for those that are more experienced than I.

annhan, 2020-04-09:
> Please post your configuration files as a .zip file, as the template suggests.

it's default config, just enable Spindle.

> This would seem to be a #define conflict over PWM
> In .platformio/packages/framework-arduino-sam/system/CMSIS/Device/ATMEL/sam3xa/include/sam3x8e.h line 466 is
>
> ```
> #define PWM        ((Pwm [...]

Change PWM to PWM1 

in Configuration_adv.h
------------------------
```
  /**
   * Speed / Power can be set ('M3 S') and displayed in terms of:
   *  - PWM     (S0 - S255)
   *  - PERCENT (S0 - S100)
   *  - RPM     (S0 - S50000)  Best for use with a spindle
   */
  #define CUTTER_POWER_DISPLAY PWM1
```

in Conditionals_adv.h
------------------------
```
#if EITHER(SPINDLE_FEATURE, LASER_FEATURE)
  #define HAS_CUTTER 1
  #define _CUTTER_DISP_PWM1     1
```

in SanityCheck.h
--------------------
```
#if HAS_CUTTER
  #ifndef CUTTER_POWER_DISPLAY
    #error "CUTTER_POWER_DISPLAY is required with a spindle or laser. Please update your Configuration_adv.h."
  #elif !CUTTER_DISPLAY_IS(PWM1) && !CUTTER_DISPLAY_IS(PERCENT) && !CUTTER_DISPLAY_IS(RPM)
    #error "CUTTER_POWER_DISPLAY must be PWM, PERCENT, or RPM. Please update your Configuration_adv.h."
  #endif
```

ellensp, 2020-04-10:
Introduced here https://github.com/MarlinFirmware/Marlin/pull/15335

commit e222ba4c (2020-04-22): Prevent DUE / Spindle conflict

Fixes #17449
closed 2020-04-22 by thinkyhead