MarlinFirmware/Marlin #17452: [BUG] SKR 1.3 not connecting to PC

Dendrowen, 2020-04-09:
### Bug Description

After flashing my printer with the newest firmware version (branch 2.0.x, not the bugfix) my printer won't connect to a PC through USB.

### My Configurations

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4457345/Marlin.zip)

### Steps to Reproduce

1. Load these configuration files in marlin
2. build code with platformIO
3. copy firmware.bin to sd card
4. insert sd card in printer
5. boot and connect printer

**Expected behavior:** 
A new device being found and being able to connect through pronterface

**Actual behavior:** 
Nothing. Some lights on the printer come on. I do not have a screen.

#### Additional Information

I tried adjusting the baudrate and selecting different serial ports.

ellensp, 2020-04-10:
With this controller if the firmware is not running, it is not visible as a USB device.
1) take out the supplied complete and utter piece of shit 128mb sd card take a hammer to it. (it corrupts firmware and causes so much grief) 
2) relabel a new sd card "REARM" (recommend <= 32GB cards) 
3) plug the sd card into you computer, using its sd slot or a usb/sd adapter
4) compile and upload the firmware (it will copy it to the SD card)
5) safely eject the sd card and put it in the controller. 
6) power on the controller and cross your fingers it was only firmware.

Dendrowen, 2020-04-10:
I already installed a new SD card in the past. The firmware also gets installed because it disappears from the SD card after the first boot. (only a FIRMWARE.CUR left)

ellensp, 2020-04-10:
Used your config files, generated new firmware.bin using current bugfix2.0.x. It worked as expected. 
I have attached my firmware.bin (in a zip file) so you can try it, in case its something wrong with your compiler  
[firmware.zip](https://github.com/MarlinFirmware/Marlin/files/4460269/firmware.zip)
If this does not work, you have dead hardware.

Dendrowen, 2020-04-10:
It works. Thanks!

closed 2020-04-10 by Dendrowen